cjcliffe/CubicSDR #584: CubicSDR Refresh Loading time is too long on OSX

cn0xroot, 2017-10-24:
Hi , CubicSDR is a nice project.
but when i use it on osx,I Executive Refresh ，it need 3-5 minutes.

os:10.12.6
cpu:1.2G
mem:8G

Is there anybody has the same problem？

Toontje, 2017-12-05:
I don't know which Refresh you are referring to. I hope it's the Device Refresh on the Device Selection screen. 
It doesn't take this long on my system, just a few seconds, but what i do see is that my SDRPlay RSP1 is NEVER detected on launch of CubicSDR and ALWAYS on the first refresh. It's not a big deal, but maybe something the team can look in to.

SDRplay, 2017-12-05:
@Toontje this is a known issue on the Mac version of the API - it's on the list to be looked at. It's an API issue, not a SoapySDR/SoapySDRPlay/CubicSDR issue.

cn0xroot, 2017-12-05:
maybe @SDRplay is right.thanks @Toontje and @cjcliffe

SDRplay, 2017-12-05:
To be clear of what the issue is with the API. The RSP won't appear the first time SoapySDR requests a device list. Subsequent requests are fine. This will happen each time the RSP is reconnected to the machine or the machine is rebooted. A refresh will fix the problem and it takes seconds. There is nothing related to this that takes minutes - that, to me, sounds like something is timing out. Do you have SoapyRemote installed?

cn0xroot, 2017-12-05:
I figured out the time ，About fifty seconds 

```
port installed |grep Soap

  SoapySDR @0.6.0_0 (active)
```
just install soapySDR with macport,and build SoapyRTLSDR、soapy_power from source

SDRplay, 2017-12-05:
Try
SoapySDRUtil --info
to see what options are installed.

cn0xroot, 2017-12-05:
```
SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.0-release
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libHackRFSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libRedPitaya.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libmiriSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libosmosdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/librfspaceSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/librtlsdrSupport.so
Loading modules... done
Available factories...hackrf, lime, miri, null, osmosdr, redpitaya, remote, rfspace, rtlsdr,
```

SDRplay, 2017-12-05:
So on a refresh, CubicSDR will be requesting all of these to return device lists. That's probably where your time delay is coming from.

Also, FYI - libmirisdr is a reverse engineered partially working library that neither Mirics nor SDRplay recommend or support.

No SDRplay library, so it can't be the issue I was talking about.

cn0xroot, 2017-12-05:
emm, i use the bladerf and build SoapyBladeRF from https://github.com/pothosware/SoapyBladeRF

i has try to refresh only need 3 seconds.  it work fine now thanks!

closed 2017-12-26 by cn0xroot