cjcliffe/CubicSDR #586: CubicSDR + Arch Linux + SDRplay RSP2 not working

Terminalator, 2017-11-03:
I posted in some detail on the SDRplay forums about a problem I am having with CubicSDR and was directed over here. To save me re-posting everything again (as there was quite a lot), I hope it's OK to post the link to my issue.

http://www.sdrplay.com/community/viewtopic.php?f=7&t=2605

In summary, CubicSDR appears on the screen, I select the SDRplay device but nothing else happens. I see no waterfall, no varying spectrum etc. However all the menus are as expected, it just seems like no information is being displayed by CubicSDR.

vsonnier, 2017-11-03:
Please provide the CubicSDR log which is missing from your forum post. Thank you.

Terminalator, 2017-11-03:
Thanks for the reply and apologies for leaving out the log file. I hope this is what you need; it's the terminal output when I start CubicSDR:
```[john@jupiter64 ~]$ cubicsdr

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: Yes
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: Yes
Using glxSwapIntervalEXT.

Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono27_0.png', parsed 255 characters.
SDR enumerator starting.
SoapySDR init..
	API Version: v0.6.0
	ABI Version: v0.6
	Install root: /usr
	Loading modules... 
	Available factories...null, remote, sdrplay
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1707043D20
Make device 0
  mir_sdr_api_version=2.100000
  mir_sdr_hw_version=2

Reporting enumeration complete.
SDR enumerator done.
SDR thread starting.
device init()
[INFO] Using format CF32.
Device Stream MTU: 65536
starting readLoop()

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

Terminating SDR thread..
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
readLoop() ended.
device deinit()
SDR thread done.
Terminating SDR post-processing thread..
Terminating All Demodulators..
Terminating Visual Processor threads..
Garbage collected demodulator instance '100.877'... 
Application termination complete.```

starvald, 2017-11-03:
interestingly my RSP2 wasn't working this morning (on arch too) but just repulled soapysdr-git and soapysdrplay-git and it's now working.

Terminalator, 2017-11-03:
When you say it wasn't working, was it the same symptoms as mine? I'll try
another installation this evening.

starvald, 2017-11-03:
kinda different, Soapy wasn't seeing my rsp2 at all but the dump1090 sdrplay compatible version did.
so sdrplay libraries seemed to be working.
https://github.com/SDRplay/dump1090_sdrplus

vsonnier, 2017-11-03:
@Terminalator Thanks. Funny thing, this log is perfectly normal, as if the device was working normally by streaming data. So the only way to have nothing to see is to have zeroed samples continously...

Since you mentioned  @loughkb instructions, I suppose you used these scripts: https://github.com/loughkb/SDRplaylinux  (also follow the blog post) to compile CubicSDR ?

Terminalator, 2017-11-03:
I didn't use those scripts actually; I installed some of the packages from
the Arch repositories and compiled others from the AUR. I used Kevin's
scripts when I installed CubicSDR etc on my Ubuntu laptop.

Do you have any idea what could be causing this or better still, how to fix
it?

Thanks for taking time to reply.

vsonnier, 2017-11-03:
@Terminalator I have no idea.. Maybe re-installing the API driver ? Some blog answers mentioned 'blacklisting' kernel modules, but as SDRplay Support said you seem to communicate with your device just fine...  
Maybe it is just an OpenGL display problem and the rest of the system is running fine? Indeed I saw in the log the following trace: `Garbage collected demodulator instance '100.877'... ` 
which says that you created a demod instance at 100.877Mhz. Did you hear the FM station then ?
 
Please attach  some screenshots directly in the issue, at least one showing the "SDR devices" initial dialog, then another showing the main display.

Terminalator, 2017-11-03:
@vsonnier Thanks for the reply. Before I try to recompile the wxGTK packages, I will post some screenshots. The log output appears immediately when I click "start" on the CubicSDR device selection window.

The 100.877 frequency was a random click I made in the window - there was no signal / display present all. The module blacklisting applies only to the SDRplay RSP1 afaik.

[image]
[image]
[image]

Terminalator, 2017-11-03:
Well this just gets weirder....

I just did a pacman -Syu to update my system. A new kernel was installed 4.13.11-1-ARCH #1 SMP PREEMPT

Now it works! Well it works but appears (I assume) to lack OpenGL acceleration as the interface is quite slow compared to my laptop. I'll look into re-compiling wxGTK (if that is the actual problem). 
Any suggestions on a "laggy" interface welcome :)

Anyway, I'd like to thank everyone for the advice and help. I can only **assume** that there was a kernel issue which has now been fixed. I'll also update the thread on the SDRplay forums.

@vsonnier  @starvald  Thanks for your time and help.

vsonnier, 2017-11-16:
@Terminalator No news, so good news I suppose. I can close this issue now, I suppose 👍

closed 2017-11-16 by vsonnier