cjcliffe/CubicSDR #588: Mouse issues with Cubic

Mojoron, 2017-11-17:
The last two updates have caused mouse/curser issues. When clicking on waterfall to change frequencies, the VFO does not change even using action keys. Scrolling is not there any longer either. I'm using an SDRplay, first generation, and cubicsdr 0.2.2 with the new SDR driver HPI/HW v 2.11 MacBook Air.

dl2ymr, 2017-12-05:
Here the same issue on my iMac with SDRplay first generation and cubicsdr 0.2.2.

jcrippen, 2018-01-03:
I had some mouse problems with 0.2.2 with macOS 10.12.6. But I switched to the master branch from GitHub using [homebrew-sdr from dholm](https://github.com/dholm/homebrew-sdr/blob/master/cubicsdr.rb) with the `--HEAD` option. Things work fine now; I also noticed some significant redraw speed increases. If you have Homebrew installed you might try this to see if your problems are already fixed for the next release.

vsonnier, 2018-01-22:
Binaries v0.23 for MacOS have been released. Please @Mojoron , @dl2ymr @jcrippen tell me how it runs for you.

Mojoron, 2018-01-23:
Now my Soapysdr doesn’t work and I’m not a programer to be able to do terminal work. Any work around for a new soapysdr for cubic?

mojoron

vsonnier, 2018-01-23:
@Mojoron You don't have to build anything yourself, here https://github.com/cjcliffe/CubicSDR/releases/download/0.2.3/CubicSDR-0.2.3-Darwin.dmg is a standalone version for MacOS that gathers both CubicSDR and its lastest SoapySDR modules for communicating with the hardware.
For short, only install and run and it should be good.

vsonnier, 2018-02-02:
@Mojoron @dl2ymr @jcrippen Greetings all ! The Mac builds have been recently updated, so could you try
https://github.com/cjcliffe/CubicSDR/releases/download/0.2.3/CubicSDR-0.2.3-Darwin.dmg
and tell us if you still have issues or not, so we can close this issue. (or not).

dl2ymr, 2018-02-04:
Hi all,

thanks for your work, but the mouse issue is still here. Nothing changed.

Greetings and 73
Michael, DL2YMR

TheButterZone, 2018-04-07:
Same issue, OS, v0.2.3.

cjcliffe, 2018-05-08:
@dl2ymr @Mojoron @jcrippen -- Hi all, I still haven't reproduced the original issue here but I'm thinking with 0.2.4 now using wxWidgets 3.1.1 (which is used for mouse event handling and interaction) it might be worth trying again -- https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.4

Please let me know if anything has gotten better or worse; thanks!

dl2ymr, 2018-05-08:
Hi,
can't start CubicSDR 0.2.4 because the SDRPlay is not found.

After downgrading to 0.2.3 everything is working fine. The Mouse Issue is gone and the SDRPlay is found without any problems.

Greetings and 73
Michael, DL2YMR

vsonnier, 2018-05-27:
@dl2ymr Hello ! SDRPlay has just released an updated API/DRIVER for Mac that should solve the SDRs detection problem (#649), could you give it a try with CubicSDR v0.24 ?

dl2ymr, 2018-05-27:
Hi Vincent,
tnx for your information.
Now it works!!!
Thank you guys for your work.
73 de
Michael, DL2YMR

vsonnier, 2018-05-27:
@dl2ymr Great ! Have fun with CubicSDR ! 
Let's close this issue then.

closed 2018-05-27 by vsonnier