cjcliffe/CubicSDR #589: wiki:build-static-wxwidgets on ubuntu need add libgtk2.0-dev

cn0xroot, 2017-11-23:
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux#build-static-wxwidgets
configure fail:

`The development files for GTK+ were not found. For GTK+ 2, please ensure that pkg-config is in the path and that gtk+-2.0.pc is installed. For GTK+ 1.2 please check that gtk-config is in the path, and that the version is 1.2.3 or above. Also check that the libraries returned by 'pkg-config gtk+-2.0 --libs' or 'gtk-config --libs' are in the LD_LIBRARY_PATH or equivalent.`

sudo apt-get install libgtk2.0-dev

vsonnier, 2017-11-23:
@cn0xroot Could you please retry with the following options instead:
`./configure  --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --prefix=...etc`
which is used in @loughkb (https://github.com/loughkb/SDRplaylinux) excellent scripts to build CubicSDR and SDRPlay module under Linux.

So far it worked on Ubuntu 16.10 x64 (under WSL under Windows 10, no less) using `libgtk-3-dev` as the Wiki said.

@loughkb script main difference with the Wiki version lies in the options `--disable-webkit --disable-webview --disable-webviewwebkit` options instead of `--disable-webkit --disable-webview`

cn0xroot, 2017-11-24:
```
wxWidgets-3.1.0$ ./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --prefix=...etc
configure: error: expected an absolute directory name for --prefix: ...etc

```

I try it on ubuntu 16.04 , after install libgtk2.0-dev it work fine.

vsonnier, 2017-11-24:
Did you run `sudo apt-get install libgtk-3-dev libpulse-dev` before configuring vxWidgets ? 
I tried to `sudo apt-get remove libgtk2.0-dev` first, then `sudo apt-get install libgtk-3-dev` after, and it worked fine. ("pure" GTK+3 dev files, no GTK2.0 dev)

Of course, it works with installing  libgtk2.0-dev, however wxWidgets will detect GTK+2 first and use it instead of the GTK+3 libs and then CubicSDR will have... an ancient look-and-feel.

cn0xroot, 2017-11-24:
Thanks for your quickly relpay,i'll install libgtk-3-dev.

vsonnier, 2017-11-24:
Please tell me if it works for you, in this case I'll update the Wiki with @loughkb command line and close this issue.

cn0xroot, 2017-11-25:
Sure, it work fine now.

closed 2017-11-25 by cn0xroot

Sudarssan-N, 2021-08-14:
> sudo apt-get install libgtk2.0-dev

it works Thank you sir.