cjcliffe/CubicSDR #590: Top waterfall and main waterfall keep going out of sync

Toontje, 2017-11-25:
[image]
As you can see in the main waterfall at 13.986MHz there is nothing to be heard where as in the top waterfall there are CW signals seen. When tuning to those signals in the top waterfall the signals are indeed not there, so the main waterfall is the source of truth.
I don't see the purpose of the top waterfall. I think if you would take this away you would have a perfect area for controls which are now all hidden under keys or in various areas of the screen (Modes left, memories left, mute and solo to the right, delta lock in the middle, etc.)

vsonnier, 2017-11-25:
Hello @Toontje, you actually experienced #295, the issue will be treated some day in #338.

Toontje, 2017-11-25:
So what about getting rid of the modem waterfall and improve the main waterfall pan and zoom functions?

ve1jot, 2017-11-25:
I'm for that, also, it might lighten the graphics load?

lantren, 2017-11-28:
maybe as an option , for those with GPU problems but the separate modem zoom is too useful . It's glitchy yes. but it should be fixed, not discarded.

Toontje, 2017-11-28:
@lantren Can you give me the use case you are using it for? Just for my understanding because i just see it as a repetition of the main waterfall. There is nothing i can do with the modem waterfall that i cannot do with the main waterfall. Unless i miss something, hence my question.

lantren, 2017-11-28:
@Toontje . Sure. I find it most useful when monitoring multiple frequencies. Just today was attempting to locate a missing remote temperature sensor. I was monitoring 6 frequencies @ 433 mhz. I like not having to zoom anything around when a signal is detected on one of the frequencies since the modem view will display which ever frequency breaks squelch. I also like being able to fine tune in the modem view without the need to change my water fall view since i like to visually monitor larger area of the spectrum. Basically, i doesn't DO anything different , i just find having two useful.

Toontje, 2017-12-05:
@lantren Understood. I guess like this we all have our own use cases. I use it as panadapter for my HAM radio hobby. So all i do is search for signals within a specific frequency range and let CubicSDR tune my radio to it. I would love to drop the modem waterfall and have protocol decoders in CubicSDR. But then again, this has already been done in many other applications and we don't need to repeat the same thing over again. What i desperately need is a CW modem. For the rest i think CubicSDR is perfect as it is.

L1049G, 2018-02-10:
I have this problem also; main and top waterfalls getting out of sync. I've tried to find a work-around
without success so far. I like the top/modem waterfall, useful for example, for fine tuning USB/LSB
and centring DRM signals etc. I would like to see it kept.

vsonnier, 2018-05-19:
Hello @Toontje and all,
@cjcliffe Has recently made an improvement on the modem waterfall processing using the PFBCH2 channalizer, so that the 'ghost signals' as described in #338 should be gone, at the price of more CPU consumption.
So I've exposed the improvement by changing the "Reduce CPU Usage" toggle in the `Settings` menu into a "CPU usage" submenu with 3 different entries:
* High : Normal with the PFBCH2 on,
* Normal : using the usual PFBCH that may create ghost signals,
* Low : matches the previous "Reduce CPU usage"

High _should_ be the preferred setting if your machine is powerful enough which should be the case in 2018.  

Note that #338 and #295 covers 2 different issues:
*  a) Main waterfall showing 'signals' that appear/disappear when zooming,
*  b) A particular modem showing _true_ (so to speak) 'ghost signals' meaning they can be BOTH listened and seen on the modem waterfall but are not there at all.

The described improvement covers b), but a) is a separate issue which needs yet another filtering process.

Toontje, 2018-07-29:
> To be clear a modem waterfall is not a 'zoom of sorts' of the main waterfall because they are generated in a very different way.

Do you have a ELIM5 description of what the modem waterfall is for? Because indeed tend to see it as a "zoomed" part of the main waterfall.

vsonnier, 2018-07-29:
@Toontje The modem waterfall allows precise centering/moving and sizing of a particular demod.
It is also the image of the processed signal that is feeding the audio part.
For performance reasons, a demod is not fed with the entire bandwidth (ex 8MHz for a SDRPlay) Although that would work, it is not used that way because:
-  That would soon saturate the CPU in case multiple demods are present,
-  It is somewhat a waste to use a 8MHz-large processing to demodulate a single 200Khz max (FM stereo) located somewhere in this 8MHz zone, and doing the same thing for each demod.

That is where the Polyphase Filterbank Channalizer  kicks in. This a DSP method to cut the 8MHz bandwith into N independent processing streams (a.ka. channels), spread out in frequency: (see #295 for details)

[image]

In CubicSDR, the N is chosen such as the bandwidth is split into 500Khz-wide channels.
Then each demod is associated with the best enclosing channel corresponding to its frequency, so is fed with 500Khz worth of bandwidth, not the whole 8Mhz. Much better from the CPU perspective.

Problem is, there is a catch: DSP processing is no magic, and if a demod is located close to a 500Khz boundary, ghost/mirrored images appear, which translates into "ghost" signals.

The advantage of PFBCH2 over the PFBCH one is that each channel overlaps its neigbour and it's bandwidth is "virtually"  2x bigger, so that a demod is __never__ close to the boundary, and so don't create false images. Cons: needs 2x more processing per-demod too.

To answer your question:
- The main waterfall is made of FFT of the whole bandwidth
- Each demod waterfall is made of FFT of a particular PFCH channel, which is also used for audio processing.

cjcliffe, 2018-08-04:
@Toontje the CPU usage options will be in the 0.2.5 build later this month; I'll see about adding the ability to hide the modem waterfall under the display options as well -- there's a compile flag for it already, but no runtime controls.

@vsonnier gave a good description; and yes technically it is a zoom we're just saying the signal is handled in a way that allows it to get ghosted visually since it runs from a lower bandwidth PFB channel compared to the full-bandwidth waterfall.

closed 2026-11-14 by cjcliffe