cjcliffe/CubicSDR #591: CTCSS/Tone Squelch Support?

jinkside, 2017-12-14:
It would be really nice if the squelch system supported CTCSS, also known as _tone decoding_ or _PL tones_. I imagine a dialog with a combobox or list that would have the common tones (103.5Hz, etc.).

ifuchs, 2018-02-23:
I second that! SDR# has a plugin for CTCSS decoding and squelch but that of course won't run on OSX.

ifuchs, 2018-03-01:
Perhaps there is a Windows executable that will run under Wine and do the CTCSS/PL squelch? DSDPlus works well with CubicSDR in this way.

jinkside, 2018-03-01:
Possible. Piping audio streams around arbitrarily is not something Windows handles very well.

ifuchs, 2018-03-01:
CubicSDR sends the audio to Soundflower and DSDplus gets audio from Soundflower. Now I just have to find (or write) a program that does the CTCSS decoding and mutes the audio based on the code.

jinkside, 2018-03-01:
That's great for Mac users, then, but I'm not aware of an equivalent for Windows users. I have Virtual Audio Cable or something like it, but each audio stream ends up as a microphone and a speaker, so it's a little messy.

ifuchs, 2018-03-01:
Sorry, I was looking for a Mac solution since there is a solution for Windows (SDR# + the CTCSS/DCS plugin)

erichahn, 2018-05-01:
Any support for ctcss/dcs inside Cubic SDR? I love being able to monitor multiple channels at once in cubic and how easy it is.