cjcliffe/CubicSDR #592: Is there a premade bookmarks file with bands, shotwave radios stations, commercial broadcast, and such ? Also a discription of what I am building / testing with CubicSDR and SDRPlay

jerrynsr, 2017-12-15:
It would be great if we can all maintain a bookmarks.xml file that we can share.  Also a bookmarks with the bands,  channels, commercial broadcast stations, shortwave, etc.. all saved in a premade bookmark.

This app is great!  I am building a remote outdoor SDR receiver with multiple antennas attached to my SDR-Play.   I have SoapySDRServer running on a Kangaroo running Ubuntu.   Right now I only have a Dipole and HiZ antenna connected, the kangaroo is in a my work area running connected to my  SDRPlay running SoapySDRServer connected via 5Ghz Wifi.   So far I am getting great results. the goal is to get this setup outside in a weather proof box connected to a mast outside powered by a solar panel / marine battery

cjcliffe, 2018-01-03:
Possibly add support for importing/exporting folders to facilitate something like this?

Toontje, 2018-02-18:
You can always create a new bookmarks page here: https://github.com/cjcliffe/CubicSDR/wiki/_new and share yours.