cjcliffe/CubicSDR #593: Bookmark ranges zoomlevel not working correctly

Toontje, 2017-12-26:
[image]
In this image you can see that i have created a range from 7MHz-7.04MHz. When i select Go to Range the waterfall display shows me 7.25MHz-7.29MHz.
Can you make CubicSDR make a good effort to select the right range and the best fitting zoom level for that range in the waterfall when i click Go to Range?

vsonnier, 2018-01-09:
@Toontje Hello ! I think I've found the problem. Could you please test the last master to see if its good for you ? Thanks !

Toontje, 2018-01-11:
I'd have to set up my build environment again, but yes, i will test it and let you know what i find.

cjcliffe, 2018-01-15:
@Toontje 0.2.3 build is available on the releases page which now includes this fix.

vsonnier, 2018-01-21:
Hello @Toontje ! Could you confirm your problem is either fixed or not with v0.23 version ? Thanks !

vsonnier, 2018-02-02:
Well, no news is good news I suppose, so I'm closing this issue.

closed 2018-02-02 by vsonnier

Toontje, 2018-02-05:
Sorry for not responding. No news means no time in my case...

commit 0f72a0f6 (2018-01-09): Fix #593 : Make the middle of the range always the central frequency