cjcliffe/CubicSDR #595: Support Module for RTLSDR Build Problem

James-Brown, 2018-01-01:
I can follow the Wiki page to build for Windows (using Win 10 64 bit and Visual Studio 2017) down to where I start the SoapyRTLSDR Support Module build. Then I get CMake errors:

CMake Error: The source "C:/MSVCDev/SoapyRTLSDR/CMakeLists.txt" does not match the source "C:/MSVCDev/CubicSDR/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory."

Is there an update for this process? I wan't to be able to build support for HackRF so if there is a process for this that would be a great help.

vsonnier, 2018-01-03:
I have not built it in a while, but I did on Visual Studio 2015 before using both [libusb](https://github.com/libusb/libusb) and [librtlsdr](https://github.com/librtlsdr/librtlsdr) from Github, and configuring / generating solution files with the CMake-GUI. I guess we should update the Wiki accordingly.
Since I know work with Visual Studio 2017 as well, I take this issue.

James-Brown, 2018-01-03:
I kept a log of differences (very few) while I was attempting the build. 
You can see where I got to on my web site:
http://www.seti.net/engineering/engineering.php

James Brown

From: Vincent Sonnier 
Sent: Tuesday, January 2, 2018 21:36
To: cjcliffe/CubicSDR 
Cc: James Brown ; Author 
Subject: Re: [cjcliffe/CubicSDR] Support Module for RTLSDR Build Problem (#595)

I have not built it in a while, but I did on Visual Studio 2015 before using both libusb and librtlsdr from Github, and configuring / generating solution files with the CMake-GUI. I guess we should update the Wiki accordingly.
Since I know work with Visual Studio 2017 as well, I take this issue.

vsonnier, 2018-05-19:
@James-Brown  Hello!   At last, I've updated  the CubicSDR build instructions for Windows, together with a detailed section for SoapyRTLSDR which should be OK now. Closing this issue then.

closed 2018-05-19 by vsonnier

James-Brown, 2018-05-19:
Thanks. I’ll give it a go and report back...

From: Vincent Sonnier 
Sent: Saturday, May 19, 2018 07:04
To: cjcliffe/CubicSDR 
Cc: James Brown ; Mention 
Subject: Re: [cjcliffe/CubicSDR] Support Module for RTLSDR Build Problem (#595)

@James-Brown Hello! At last, I've updated the CubicSDR build instructions for Windows, together with a detailed section for SoapyRTLSDR which should be OK now. Closing this issue then.