cjcliffe/CubicSDR #600: audio_recording branch with no recording function in Arch Linux

cryoguy, 2018-01-13:
I git-cloned and compiled the audio_recording branch in my Arch linux system.  I do not see any audio recording function, such as path setting under File menu etc; there is no difference to the master branch.  Any thoughts or suggestions?

starvald, 2018-01-13:
hmm arch user here too, i compiled from aur but changed the source line to
`source=(cubicsdr-git::"git+https://github.com/cjcliffe/cubicsdr.git#branch=audio_recording")`

cjcliffe, 2018-01-15:
@cryoguy the recording start / stop button will also appear in the left-side bookmark list when selecting an active demodulator.

parkerlreed, 2018-01-15:
Also just to note, it's in master now. :) (So the PKGBUILD fix isn't needed anymore)

vsonnier, 2018-01-22:
Binaries v0.23 with this feature are now released for every pltaform, so this can be closed.

closed 2018-01-22 by vsonnier