cjcliffe/CubicSDR #601: 0.2.3 OSX Crash on exit

cjcliffe, 2018-01-15:
`
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
Assertion failed: (e == 0), function ~recursive_mutex, file /BuildRoot/Library/Caches/com.apple.xbs/Sources/libcxx/libcxx-120.1/src/mutex.cpp, line 82.
`

vsonnier, 2018-01-15:
@cjcliffe I can't help you much here, because I don't have a Mac. 
Still, I noticed an Apple-specific call to `AudioThread::deviceCleanup()` on exit. What is the purpose of this call ?

cjcliffe, 2018-01-15:
@vsonnier I figured you wouldn't be able to test but might be able to weigh in on recursive_mutex issues.  Will post here as I uncover stuff later.

vsonnier, 2018-01-15:
Found a strange thing in `void AudioSinkThread::run()`, the `inputRef` variable is never initialized, but tested anyway.
I suppose the idea was to make a ` AudioThreadInputPtr AudioSinkThread::inputRef` member where the last

vsonnier, 2018-01-16:
In the commit d8ac9559fe281c43b4207af8daae9378a2de6021  on master I've done some adjustments on `BookmarkMgr` where `recursive_mutex` are used: 
- some unprotected calls were fixed, 
- some others didn't need locks at all so I've removed them, 
- Changed some calls to return copies instead of references which is better in multithread.
 
I don't know if it is going to affect the present problem, though. 

The 2 others classes using `recursive_mutex` are the DomodulatorInstance and Mgr, and it really seems to be needed there.

vsonnier, 2018-01-21:
I tried to rework the `AudioThread` by understanding the class better, putting a lot of comments as well.
When attempting to cleanup the `deviceController` list by `deviceCleanup()` at termination, (as we should) the call hang here on Windows because apparently the matching thread is seemingly already joined and deallocated....
So I guess it is to be applied only on MacOS as you did.

cjcliffe, 2018-01-22:
@vsonnier bookmarkmgr https://github.com/cjcliffe/CubicSDR/commit/d8ac9559fe281c43b4207af8daae9378a2de6021 updates seemed fine here on the macbook; but it wasn't crashing here on exit either way -- just my work machine appears to do that.

Latest update at https://github.com/cjcliffe/CubicSDR/commit/36224defd7ffc229cc6edf5e123d6aeed2794d9e now appears to hang on exit on the macbook and requires a forced quit if any demod was created.

vsonnier, 2018-01-22:
@cjcliffe  Ah Progress ! When More is Less ! 
In commit https://github.com/cjcliffe/CubicSDR/commit/1279c34ed26c189138233b43322b2c27121c71e0 I rollbacked to a `deviceCleanup()` doing only a terminate, not `join()` that hang both our machines.

Still I think the problem really lies in the `deviceController` list, where it seems the `AudioThread` thread is apparently already terminated. Indeed, we never goes through this part of `run()`:
https://github.com/cjcliffe/CubicSDR/blob/1279c34ed26c189138233b43322b2c27121c71e0/src/audio/AudioThread.cpp#L530

that concerns controller threads.

Maybe the problem lies on this line:
https://github.com/cjcliffe/CubicSDR/blob/1279c34ed26c189138233b43322b2c27121c71e0/src/audio/AudioThread.cpp#L418

where a controller thread is added in its own list of bounded threads.

vsonnier, 2018-01-22:
Right, I think I've found the problem preventing the `deviceController` to be join properly, because of deadlocks. 
I think everything looks fine after the last commit https://github.com/cjcliffe/CubicSDR/commit/f2de1dd987ad1c63f37fe515ba450994b98b512c. Tell me how it behaves for you.

cjcliffe, 2018-01-22:
@vsonnier latest commit seems to be working fine on the macbook; will let you know how it goes on the iMac

cjcliffe, 2018-01-22:
@vsonnier looks good on the iMac too; no more crashes on exit or hangups with the latest commit  -- thanks!  I wasn't able to figure out how to properly trace a thread issue like this from XCode.

Going to update the current OSX release with the bugfixed version.

vsonnier, 2018-01-23:
Well, there was no magic involed in finding these deadlocks, just a matter of putting breakpoints going step by step, and the help that `join()` on the `deviceController` threads was never completing. 
So by interrupting the execution in debugger freezing the whole threads I eventually found a blocking point at some `std::lock_guard`.  On Windows version it has a data member flag `locked` showing true.
Then walking on the frozen threads I found the other frozen `std::lock_guard`... Then it was obvious :)

commit f2de1dd9 (2018-01-22): Fix #601 ?: Removed deadlock that prevent Audio controller threads to die.
closed 2018-01-23 by vsonnier