cjcliffe/CubicSDR #602: HackRF and  Windows 10

Alex-111, 2018-01-19:
The manual AMP slider  does not save its position. It always switches back to 0...

(LNA and VGA works as expected....)

vsonnier, 2018-01-19:
Hello @Alex-111. The problem likely lies in the `SoapyHackRF` module, for which I've created the issue: 
https://github.com/pothosware/SoapyHackRF/issues/18

vsonnier, 2018-01-23:
@Alex-111 Hello. The issue should have been fixed in https://github.com/pothosware/SoapyHackRF master branch. 
Either you can re-build it yourself, or you can wait to try the next iteration of the [PothosSDR win64 distribution](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D) i.e post January 11 build that is. 
Generally @guruofquality release one every month.

vsonnier, 2018-02-07:
FYI, a new PothosSDR win64 dsitribution is out: http://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2018.02.05-vc14-x64.exe

vsonnier, 2018-02-21:
@Alex-111 Ping ! Have you tried the recent PothosSDR distribution ? Please tell us if it solved your problem, thanks !

Alex-111, 2018-02-21:
Sorry for not replying:

Yes now it works as expected! Thanks for your support!

closed 2018-02-21 by Alex-111