cjcliffe/CubicSDR #603: Compiling wxWidgets on High Sierra 10.13.2 - fails

blrichwine, 2018-01-21:
Anyone know how to compile wxWidgets on High Sierra? I'm using a fully updated osx build and the latest MacPort with updates. I tried both 3.1.0 and 3.0.3. Both fail with the same error:

./src/osx/core/utilsexc_base.cpp:74:13: error: use of undeclared identifier 'verify_noerr'
            verify_noerr(Gestalt(gestaltSystemVersion, &sUMASystemVersion));
            ^
1 error generated.

Thanks for any help

EtherealT, 2018-01-30:
I'm also having the exact problem

righthalfplane, 2018-02-02:
I was able to compile wxWidgets 3.0.3 on High Sierra by first defining a cpp flag -

CPPFLAGS=-D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=1 

and then

./configure --prefix=/usr/local/wxWidgets --with-macosx-version-min=10.9

This built what I needed

righthalfplane, 2018-02-03:
wxWidgets-3.1.0 on High Sierra required one additional option to get it to install using tcsh

setenv CPPFLAGS -D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=1

./configure --prefix=/usr/local/wxWidgets --with-macosx-version-min=10.9 --disable-mediactrl

make -j 8

sudo make install

KaungZawHtet, 2018-03-03:
How and where to define CPP_Flag

righthalfplane, 2018-03-04:
With tcsh you set the environment variable CPPFLAGS with -

setenv CPPFLAGS -D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=1

With sh you set the environment variable CPPFLAGS with -

CPPFLAGS=-D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=1; export CPPFLAGS

mohamad325, 2018-03-04:
Thanks for mailing

Mohamad asadi.

cjcliffe, 2018-05-08:
Can anyone confirm this is still an issue with wxWidgets-3.1.1?  

I've been reluctant to install High Sierra here due to the multitude of issues I've read online plus the two macs at work that both failed horribly during the upgrade and had to be reformatted, reinstalled and then migrated from TC backups.

vsonnier, 2022-01-15:
Too old issue, with outdated wxWidgets and Cubic. Users were able to build for Mac since then.

closed 2022-01-15 by vsonnier