cjcliffe/CubicSDR #606: Rig Control Drop-down missing in Linux Mint 18.1 Cinnamon 32bit

K6ETA, 2018-01-27:
The Rig Control menu is not appearing after manually installing the 0.2.3 TAR - the rest of the program is working flawlessly with SDRPlay. Great software - THANK YOU.

vsonnier, 2018-01-28:
@K6ETA You need to build CubicSDR with HAMLIB enabled, i.e having I think `hamlib-utils` and `libhamlib-dev` packages and such installed, and building with CMake adding `-DUSE_HAMLIB=1`.
Alternatively, why not using the ApplImage ?

vsonnier, 2018-02-25:
@K6ETA Ping ! Any news on your issue ?

K6ETA, 2018-02-25:
Oh, hi Vincent,

Yes I recompiled the software using the mentioned commands and bingo! Rig control appeared and works perfectly with my IC-7300! Well, almost perfectly at least. Occasionally it doesn't fully sync the frequency, but it works 95% of the time!

I must say, CubicSDR is a fine bit of coding! Many thanks!

Steve K6ETA

------ Original Message ------

From: Vincent Sonnier
To: cjcliffe/CubicSDR
Cc: K6ETA, Mention
Sent: February 25, 2018 at 12:50 AM
Subject: Re: [cjcliffe/CubicSDR] Rig Control Drop-down missing in Linux Mint 18.1 Cinnamon 32bit (#606)

@K6ETA Ping ! Any news on your issue ?

------ Original Message ------

From: Vincent Sonnier
To: cjcliffe/CubicSDR
Cc: K6ETA, Mention
Sent: February 25, 2018 at 12:50 AM
18.1 Cinnamon 32bit (#606)

vsonnier, 2018-02-25:
@K6ETA Thats great ! Have fun with CubicSDR !

closed 2018-02-25 by vsonnier