cjcliffe/CubicSDR #607: Cubic 0.2.3 crash on Win 10 and Mac OSX when starting Soapy Remote device over Wifi

rrobinett, 2018-01-28:
I have attached an SDRPlay 2 to a Raspberry Pi and  Cubic can find and run the RP2 when the Pi is connected to my LAN with a wired ethernet connection.  
However when the Pi is attached to the LAN via Wifi Cubic will not find it in a search, probably because the Wifi connection does not allow broadcast or multicast traffic.
When I add the Pi's Wifi IP address Cubic will find the RP2 and list its status.
However Cubic crashes when told to start the RP2 it just added.
The Soapy server on the Pi logs that Cubic is interacting with it, and the Soapy server terminates cleanly when the remote Cubic crashes, so there is plenty of valid communication between the Cubic client and the Soapy server.
Since both the Mac and Win10 versions of Cubic crash in the same way, that suggests there is an unhandled communication condition common to both OS environments.

My goal is to battery power the RP2 and Pi and fly them under a tethered balloon,  so having a wifi connection is essential.

vsonnier, 2018-01-28:
@rrobinett Right, this looks shockingly similar to #496. On my side, (also Win 10 and SDR Play 2) I've tried several times SoapyRemote on a `localhost` connection, without problems.
 
I suppose you installed on both sides the latest driver and Pi image (API v2.11) so lets try with the SoapyRemote settings used on the Win 10 machine: either list them, of better put a screenshot of the SDR devices dialog here. 
Indeed there is a lot of different settings you can put, and it is very easy to use an invalid combination.

rrobinett, 2018-01-28:
Thanks for your interest.

On the Pi I installed the most recent RPI 3 V0.2 image of 26 Nov 2017 which
I believe includes the V2.11 driver this Soapy version:

pi@raspberrypi:~ $ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-gba9ad2d2
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so
Loading modules... done
Available factories...null, remote, sdrplay,
pi@raspberrypi:~ $
​

SDRPlay includes Mac Cubic 0.2.2 on their download site, but that version
doesn't automatically or manually connect to the wired or wireless Pi.
Win10 Cubic 0.2.2 does connect automatically and manually to the wired
Pi+RP2 and crashes in the same way as Cubic 0.2.3.  Cubic 0.2.3 connects
manually and automatically on both Mac and Win10 and when connected via
Wifi they crash in the same way on both the   Mac and Win10.  Following are
screen shots from Win10 Cubic v0.2.3:

When connected by wired ethernet, Cubic automatically finds the Pi+RP2
during startup and displays this info about it:

[image: Inline image 1]

When connected by wired ethernet, Cubic automatically finds the Pi+RP2
during startup and displays this info about it:

[image: Inline image 2]

The 'Not Found ()' entry under the Remote is suspicious and the only
visible difference between the two.

I am a system SW developer, but not familiar with either the Mac or Windows
IDEs.  However I am eager to help debug this problem given any mentoring
you can supply, since I can find no other combination of HW and SW to
achieve the goal of this project:

SDRPlay 2 => Raspberry Pi (battery powered) <------ Wifi -----> SDR =>
WSJT-x

thanks,

Rob

vsonnier, 2018-01-28:
@rrobinett Please respond using the Github Web interface, I can' see your screenshots otherwise.

rrobinett, 2018-01-28:
Sorry, I'm a newbie.  Here are my three screenshots:

[image]
[image]
[image]

vsonnier, 2018-01-28:
@rrobinett the `Remote Window` and `Remote Priority` are visibly senseless, use something like  256*1024 and 1 respectively. Maybe lower `Remote MTU` as well, and try again.

rrobinett, 2018-01-28:
Of course you are correct that those values are invalid, but the values are the same in the working wired ethernet case.
Anyway, how do I change those values?  I tried to change MTU to 1400 in the GUI, but that didn't stick when I switched to another input device and then came back to the Soapy device

vsonnier, 2018-01-28:
@rrobinett First select on the left pane the right, device, the on values listed on the right, you click on the one you want to change, type a new value, `Enter` then you click elsewhere to be sure the value stays. Then `Start`.

rrobinett, 2018-01-28:
The new values don't stick. 
Once I click on another Device and come back they have returned to their bad values.

vsonnier, 2018-01-28:
@rrobinett Don't click on another Device, but on another field __of the same device.__ (to loose focus in GUI speak)

rrobinett, 2018-01-28:
OK, now Cubic doesn't crash, but it doesn't run either.  It just hangs

vsonnier, 2018-01-28:
@rrobinett Please give us both the logs of CubicSDR and SoapyServer simultaneous execution.

rrobinett, 2018-01-28:
I am away from home for several hours.  Where do I find those files?
-- 
Rob Robinett
rob@robinett.us

vsonnier, 2018-01-29:
@rrobinett You don"t find them, you make them:) For SoapySDRServer I maint only its traces you get on the  terminal.
For CubicSDR on the Windows 10 machine, its a bit more complicated: 
-  Open Start Menu => type `cmd.exe` (it will be found automatically, called command line or something), then you have your terminal.
-  Navigate to your CubicSDR install : `dir` as `ls`, `cd` as ... `cd` ...etc.
- You won't have traces directly even if you run it in the terminal unless you pipe them in a file, so run CubicSDR by `CubicSDR.exe > run.log`.

A simpler version, if you have Git for Windows installed: Open a Git Bash at CubicSDR location, and then you have your usual Xterm with stdout/stderr and such.

rrobinett, 2018-01-29:
On Windows
​ I didn't have permissions to write the log file to the CubicSDR
directory, but I could write it from \User\admin and it does record what I
expect is useful information to the attached run.log file​:

C:\Users\admin>
C:\Users\admin>"C:\Program Files\CubicSDR\CubicSDR.exe" > run.log

I get this output on the Pi.  The Segmentation Fault occurs after I press
'Start' on the Cubic dialog box:

pi@raspberrypi:~ $ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-gba9ad2d2
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so
Loading modules... done
Available factories...null, remote, sdrplay,
pi@raspberrypi:~ $ SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

8ea45451-4b32-150e-8567-788d007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server...
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.157.45]:55994)
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.157.45]:55995)
SoapyServerListener::accept([::ffff:192.168.157.45]:55996)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.157.45]:55998)
SoapyServerListener::accept([::ffff:192.168.157.45]:55999)
SoapyServerListener::accept([::ffff:192.168.157.45]:56000)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.157.45]:56001)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.157.45]:56002)
SoapyServerListener::accept([::ffff:192.168.157.45]:56003)
SoapyServerListener::accept([::ffff:192.168.157.45]:56004)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.157.45]:56005)
Segmentation fault
pi@raspberrypi:~ $

rrobinett, 2018-01-29:
Here is the run.log file:
[run.log](https://github.com/cjcliffe/CubicSDR/files/1673821/run.log)

vsonnier, 2018-01-29:
@rrobinett Thanks for the input.
So basically the SoapySDRServer receives a kind of command at `Start` that make it crash, the rest is CubicSDR normal behaviour given the circumstances.

The only way now it to try to find a reproducible configuration and debug the SoapySDRServer side  itself:
- What are the precise stream settings you used for CubicSDR ? (screenshot please)
- Make sure to run the SoapySDRServer at a given ip:port address visible for CubicSDR, for instance `SoapySDRServer --bind=192.168.157.45:61234`
- On Cubic side, do not use the 'Local Net' discovery, always use a manual add like in `Add` => `SoapyRemote` => 192.168.157.45:61234

guruofquality, 2018-01-29:
> SoapySDRServer --bind
> etc...
> Segmentation fault

Reminds me of this https://github.com/pothosware/SoapyRemote/issues/26 If I have understood correctly, its not necessarily something in SoapySDRServer network stuff itself, but something in the SDRPlay wrapper or mirics libs thats crashing, and we need to know what specifically. It would help a lot to

1) Rebuild libs on the pi with -DCMAKE_BUILD_TYPE=Debug
2) run SoapySDRServer with gdb and get a backtrace
3) add/or any prints to the SoapySDRPlay functions so we can see which function is bombing out

vsonnier, 2018-01-29:
@rrobinett @guruofquality Also in this use case, the Ethernet connection seems fine. So what is the differences ? different MTU ? longer calls for Wifi ?...etc.

guruofquality, 2018-01-29:
Well, I just want to see the backtrace or to know where its crashing. If it turns out that some call or bug is stomping on memory, everything feels like some kind of heisenbug. So little differences in prints and delays can exacerbate or even hide the bug. So anyway, thats just speculation until we know where it failed.

rrobinett, 2018-01-29:
I have manually specified the interface and port on both SoapyServer and Cubic.  Here is the only settings page I can access before the crash:

[image]
........................
It has been 20 years since I've used gdb, so I am very rusty on this, but the SoapyServer is dying in memcmp() and the current frame info doesn't seem very useful to me.
....................
pi@raspberrypi:/usr/local/bin $ gdb --args SoapySDRServer --bind=192.168.157.143:61234
GNU gdb (Raspbian 7.7.1+dfsg-5+rpi1) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from SoapySDRServer...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/local/bin/SoapySDRServer --bind=192.168.157.143:61234
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

63190b4d-6778-150e-8567-06eb007f0101
Launching the server... tcp://192.168.157.143:61234
Server bound to 192.168.157.143:61234
Launching discovery server... 
[New Thread 0x76c2c050 (LWP 1774)]
[New Thread 0x7642c050 (LWP 1775)]
Press Ctrl+C to stop the server
SoapyServerListener::accept(192.168.157.45:60496)
[New Thread 0x758ff050 (LWP 1776)]
.
.
.
[Thread 0x72b4f050 (LWP 1924) exited]
[New Thread 0x72b4f050 (LWP 1925)]
[New Thread 0x758ff050 (LWP 1926)]
[New Thread 0x70f3c050 (LWP 1927)]
[New Thread 0x7073c050 (LWP 1928)]
[Thread 0x7073c050 (LWP 1928) exited]
[New Thread 0x6fdff050 (LWP 1929)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x70f3c050 (LWP 1927)]
0x76fbc3a0 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
(gdb) Quit
(gdb) bt
#0  0x76fbc3a0 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
#1  0x3ff00000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) info frame
Stack level 0, frame at 0x70f3b998:
 pc = 0x76fbc3a0 in memcmp; saved pc = 0x3ff00000
 called by frame at 0x70f3b998
 Arglist at 0x70f3b988, args: 
 Locals at 0x70f3b988, Previous frame's sp is 0x70f3b998
 Saved registers:
  r0 at 0x70f3b990, lr at 0x70f3b994
(gdb)

rrobinett, 2018-01-29:
Forgive me if I missed some steps here, but I've not debugged code in 20 years, so the github/cmake environment is completely new to me.
 
I downloaded SoapyRemote from git-hub, unzipped it and ran 'cmake -DCMAKE_BUILD_TYPE=Debug', then ran the SoapyServer in ./server under gdb.
The output log does then seem to include printouts of the C++ function calls, but bt doesn't show me any more than before.  Where should I go next?
=======================
.......
SoapyServerListener::accept(192.168.157.45:61019)
[New Thread 0x758ff050 (LWP 3344)]
SoapyServerListener::accept(192.168.157.45:61020)
[New Thread 0x72b4f050 (LWP 3345)]
[New Thread 0x750ff050 (LWP 3346)]
SoapyServerListener::accept(192.168.157.45:61021)
[New Thread 0x717ff050 (LWP 3357)]
[Thread 0x717ff050 (LWP 3357) exited]
[Thread 0x750ff050 (LWP 3346) exited]
[Thread 0x758ff050 (LWP 3344) exited]
[Thread 0x72b4f050 (LWP 3345) exited]
SoapyServerListener::accept(192.168.157.45:61022)
[New Thread 0x750ff050 (LWP 3358)]
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.157.45:61023)
[New Thread 0x717ff050 (LWP 3359)]
[New Thread 0x72b4f050 (LWP 3360)]
[Thread 0x72b4f050 (LWP 3360) exited]
[New Thread 0x72b4f050 (LWP 3361)]
[Thread 0x72b4f050 (LWP 3361) exited]
[New Thread 0x72b4f050 (LWP 3362)]
[New Thread 0x758ff050 (LWP 3363)]
[New Thread 0x70f3c050 (LWP 3364)]
[New Thread 0x7073c050 (LWP 3365)]
[Thread 0x7073c050 (LWP 3365) exited]
[New Thread 0x6fdff050 (LWP 3366)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x70f3c050 (LWP 3364)]
0x76fbc3a0 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
(gdb) bt
#0  0x76fbc3a0 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
#1  0x00000000 in ?? ()
(gdb) frame
#0  0x76fbc3a0 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
(gdb)

vsonnier, 2018-02-02:
@rrobinett Hello ! On Youtube, Kevin Loughin (@loughkb) explains how he made some scripts to automate compiling of CubicSDR, including SoapySDR modules, under Linux. 
On this video in particular, he says those scripts were used to compile on RasberyPI without modification: [Ham Radio - SDRPlay running with CubicSDR on a raspberry Pi 3](https://www.youtube.com/watch?v=dx_iEpcZsDs).
He put those scripts on Github [here](https://github.com/loughkb/SDRplaylinux). 

So inspired by those scripts I would say you have to do the following to re-compile and effectively install SoapySDRremote:
````bash
git clone https://github.com/pothosware/SoapyRemote.git
cd SoapyRemote
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug
make
make install
````

rrobinett, 2018-02-07:
Hi,  
Today I finally have had time to start debugging this.  Since both Cubic and the SoapyServer on the Pi crash at the same time, it seems that the Cubic problem is likely in the Soapy library common to SoapyServer and SoapyClient.  I see a number of posts in Soapy from users describing segfaults like mine, so it seems most productive for me to debug the SoapyServer on the Pi and then come back here and recompile Cubic with any fixes which seem relevant to client mode.  Does that make sense to you?

vsonnier, 2018-02-07:
@rrobinett Indeed, for me Cubic don't crash at all, but send some kind of malformed info to the SoapyServer that make it (the Server) crash. 
So before attempting to trace it back to the client, a good place to start is to run the SoapyServer in debug and hope the traceback will us a clue of what was the malformed packet was. 
I think there are loggers in Soapy, but I don't know how to activate them.

rrobinett, 2018-02-07:
As suggested by @guruofquality, I have compiled the Pi SoapyServer with  '-DCMAKE_BUILD_TYPE=Debug' and it does printout some additional function call information, but nothing useful in tracing the crash.  I've tried adding additional printouts, but I don't understand the structure of the program well enough to add printouts in the thread functions which seem to be where that crash occurs.  So I'm in need of a little mentoring from @guruofquality or others on the Soapy forum.

rrobinett, 2018-02-08:
I have uncovered the stimulus for the Soapy segfault as due to too much IQ data for the speed of the LAN.   I have reported it in the Soapy forum, so I think this issue can be closed as far as Cubic is concerned, although Soapy could be improved to report that condition to Cubic.  Here is a copy of my Soapy post:

I have discovered that the segfault is associated with higher bitrates. When I turn the sample rate down to 1 Mbps or less the system starts running stably.
iperf measures about 35 Mbps across the wifi link and at 1M sample rate Win120 reports 25+ Mbps of network traffic, so Soapy seems to break when packet loss occurs
I can imagine that even in wired configurations the higher sample rates will stimulate a segfault, but for my WSPR application I can run at 1Mbps or less to get around these problems.
However it would be good if this error condition were reported by Soapy rather than segfaulting.

vsonnier, 2018-03-21:
@rrobinett Hello again ! The SoapySDRRemote issue https://github.com/pothosware/SoapyRemote/issues/26 was just closed, but it is unclear if you reached your goal or not ? For instance, what maximum sample rate did you achieved ?

rrobinett, 2018-03-21:
> SoapySDRRemote issue pothosware/SoapyRemote#26

I have just started testing this morning.
Aafter following the clean build instructions I can run SoapyServer on the
Pi but it prints out a lot of debug messages.  In spite of that, it does
server 250 ksps streams across the LAN to CubicSDR running on a PC.  I
think it would be most useful to suppress the verbose messages on
SoapyServer before trying to stress the communications channel.  How do I
do that?​

vsonnier, 2018-03-21:
Well, if you built the usual configuration, like that:
````
git clone https://github.com/pothosware/SoapySDR.git
git clone https://github.com/pothosware/SoapySDRPlay.git
git clone https://github.com/pothosware/SoapyRemote.git
cd SoapySDR
mkdir build
cd build
cmake ..
make -j4
make install
ldconfig
cd ..
cd ..
# Now we build the SDRPlay module for Soapy
cd SoapySDRPlay
mkdir build
cd build
cmake ..
make
make install
cd ..
cd ..
# And we build SoapyRemote
cd SoapyRemote
mkdir build
cd build
#cmake ..
cmake .. -DCMAKE_BUILD_TYPE=Release
make
make install
cd ..
cd ..
````
You should get the Release kind of things, What are those traces on SoapyServer side ?

rrobinett, 2018-03-21:
​Yes I followed those steps exactly, although I needed to run 'sudo make
install' and 'sudo ldconfig'
​
When I run 'SoapySDRServer --bind &> t.log' I get this kind of verbose
output in t.log:

....
setToggleStates: initialising sampNum=0x00017a02, gainSetting=0x224,
FsToggle=0, RfToggle=1, GrToggle=1
setToggleStates: initialising Fs=2000000.000, Rf=98800000.000, Gr=60
mir_sdr_GpioExtWrite: Addr=19 Data=0xa5
mir_sdr_2500_Readback001Trim: readback = 0xe442a0f
mir_sdr_2500_Bypass001Trim: reg13=0x1ed reg14=0x7102ae
Finished handler!
Calling handler...
SoapyClientHandler::hendleOnce() call =#102
Finished handler!
Calling handler...
SoapyClientHandler::hendleOnce() call =#1402
Finished handler!
Calling handler...
SoapyClientHandler::hendleOnce() call =#2
mir_sdr_usb_ReleaseDeviceIdx 0
Finished handler!
Calling handler...
SoapyClientHandler::hendleOnce() call =#3
Finished handler!
SoapyServerListener::handlerLoop() ENDING after #5
Calling handler...
SoapyClientHandler::hendleOnce() call =#22
Finished handler!
Calling handler...
SoapyClientHandler::hendleOnce() call =#3
Finished handler!
SoapyServerListener::handlerLoop() ENDING after #15
SoapyServerListener::close()
SoapyServerListener::close()
....

If you want, I am available today for a Google Hangouts / Teamview / Mac
Screenshare session.
Thanks

vsonnier, 2018-03-21:
Did you destroyed the `SoapySDR`, `SoapySDRPlay` and `SoapyRemote` directories and clone again from sratch ? 
The traces suggest this is still the experimental changes and not the clean source versions from the respective `master`  branches.

rrobinett, 2018-03-21:
I certainly thought I had cleaned my environment and rebuilt from scratch,
but in response to your post I went and did that once again and voila
SoapySDRServer now runs with minimal printouts
It also now runs stably at 2.048 Msps which generates 70 Mbps of ethernet
traffic, close to the limit of the Pi 3b.
So for me the segfault problems are fixed.
I am impressed by the efficiency of SoapySDRServer, at 2.048 Msps it uses
less than 10% of the Pi CPU.
thanks for the support from you and Josh.

vsonnier, 2018-03-21:
Great ! Now I suppose you'll have to test it over Wifi, which was your objective isn't it ?

rrobinett, 2018-03-21:
Yes my final test will be over wifi, but the Pi + SDRPlay is at a remote
location on a hilltop 45 minutes from my home and is active use.  I
hesitate to disable the eth0 interface because I can't get to that site for
10 days.
Also, I have on order 2 of the new Pi 3b+ which promise improved ethernet
and wifi performance.
In the mean time I thought I could try to stress the new code with the Pi
3b + RTL-SDR here in my office, but it appears that I don't have the Soapy
rtl-sdr driver

vsonnier, 2018-06-07:
Time to close this conversation I think because the core issue has been solved, so let's continue on #639 if issue araise again.

closed 2018-06-07 by vsonnier