cjcliffe/CubicSDR #608: Bug with sample rate selection handling?

Magalex-2x14, 2018-01-31:
Noticed a suspicious thing - when choosing a 4MHz sampling rate, I still see about 2MHz of spectre on the waterfall (not noticeably wider, than witn 2MHz sampling)...

Tell me, who is technically more literate, this is some kind of bug (maybe not with CubicSDR itself, but with receiver drivers), or am I missing something in the principles of SDR operation? 

OS X 10.13.3, SDRplay RSP2 pro, CubicSDR v0.2.3 (installed from prebuild package from Releases).
Tried to restart the program, reconnect the receiver - nothing helped.

Sorry for bad English.
[image]
[image]

SDRplay, 2018-01-31:
The IF bandwidths are 200kHz, 300kHz, 600kHz, 1.536MHz, 5 MHz, 6 MHz, 7 MHz, 8 MHz

With a 4 MHz sample rate, a IF bandwidth of 1.536 MHz is selected, You can see the filter band edges. If you go up to 5 MHz, where a 5 MHz IF bandwidth can be used, it will look flatter.

Magalex-2x14, 2018-01-31:
Oh, thanks... Did not pay attention to it while reading the datasheet ...

Since this opportunity has turned up, I want to express my gratitude to all those involved in the CubicSDR project for the nice user friendly software and to the @SDRplay team for the great product and broad support 👍

closed 2018-01-31 by Magalex-2x14