cjcliffe/CubicSDR #609: CubicSDR 0.2.3 does not work with RFSpace NetSDR on High Sierra

righthalfplane, 2018-02-02:
NetSDR worked with the first version for CubicSDR that I tried using Yosemite, but it now fails on High Sierra, Sierra and El Captain. I tracked to problem down to the "sdr discover" routines - they now fail to find the NetSDR device and give an error. It looks like the "sdr discover" routines were derived from those in the old versions of CuteSDR. The discover routines in CuteSDR were rewritten early last year and work well with Sierra and High Sierra. Other programs search for the device and fail, but when I give them the network information - they use it and everything works fine. With CubicSDR I can find no way to get it to skip discover step and just use the network information given.

righthalfplane, 2018-02-11:
I rewrote the "discover_netsdr" routine so that it works on Sierra.  I rebuild CubicSDR 0.2.3 using my new routine and it now works well on MacOS Sierra with the NetSDR and Sdrplay.

vsonnier, 2018-02-11:
@righthalfplane Out of curiosity, what SoapySDR module is used to access RF Space devices ?

righthalfplane, 2018-02-11:
In SoapyOsmo (git clone https://github.com/pothosware/SoapyOsmo.git) the file "rfspace_source_c.cc" has the code that accesses the RF Space devices. I rewrote the discover_netsdr routine.

vsonnier, 2018-02-12:
@righthalfplane Thank you for your effort ! You should also create a pull request, or at least an issue on SoapyOsmo Github so that everybody could benefit of your good work.

guruofquality, 2018-02-12:
I can take the pull request, but keep in mind SoapyOsmo basically has a copy of gr-osmosdr which it wraps up various modules without pulling in gnuradio: https://github.com/pothosware/SoapyOsmo/wiki

Point being, since its just containing a copy of gr-osmosdr, so any worthwhile patches should go upstream to gr-osmosdr or they might get lost when updating at a later date...

or if I missed some upstream update from gr-osmosdr, let me know so I can update SoapyOsmo.

righthalfplane, 2018-02-13:
I have pointed out the problem on the "osmocom-sdr mailing" list - they seem to be working on some of gr-osmosdr problems - Is there a better place to try?

jjschwarz, 2018-08-28:
This seems to still be an issue in the new 0.2.5 release.  CuteSDR connects fine, but CubicSDR just gives a "not found" why I connect via IP to the NetSDR.  Any thoughts on additional troubleshooting or a temporary patch?