cjcliffe/CubicSDR #610: Installation via homebrew succeeds but CSDR crashes on launch

stuzog, 2018-02-02:
```
bash$ brew install cubicsdr --HEAD
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (caskroom/cask).

==> Installing cubicsdr from dholm/sdr
==> Cloning https://github.com/cjcliffe/CubicSDR.git
Updating /Users/Stuart/Library/Caches/Homebrew/cubicsdr--git
==> Checking out branch master
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INS
==> make
Warning: Calling <<-EOS.undent is deprecated!
Use <<~EOS instead.
/usr/local/Homebrew/Library/Taps/dholm/homebrew-sdr/cubicsdr.rb:30:in `block in install'
Please report this to the dholm/sdr tap!

Warning: dholm/sdr/cubicsdr dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/cubicsdr/HEAD-7037189: 29 files, 2.5MB, built in 5 minutes 21 seconds
bash$ CubicSDR
Loading:: configuration file '/Users/Stuart/Library/Application Support/CubicSDR/config.xml'
/usr/local/bin/CubicSDR: line 5: 17198 Illegal instruction: 4  ./CubicSDR
bash$ 
```
However, running CubicSDR from terminal results in an instant crash (report submitted.

```
Model: iMac7,1, BootROM IM71.007A.B03, 2 processors, Intel Core 2 Duo, 2.4 GHz, 6 GB, SMC 1.21f4
Graphics: ATI Radeon HD 2600 Pro, ATI,RadeonHD2600, PCIe, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR2 SDRAM, 667 MHz, 0xCE00000000000000, 0x4D342037305435323637415A332D43463720
Memory Module: BANK 1/DIMM1, 2 GB, DDR2 SDRAM, 667 MHz, 0x7F4F000000000000, 0x54533235364D535136345636552020202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x88), Broadcom BCM43xx 1.0 (5.10.131.39.16)
Bluetooth: Version 4.4.6f2, 3 services, 26 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: CT500BX100SSD1, 500.11 GB
Parallel ATA Device: OPTIARC DVD RW AD-5630A
USB Device: USB 2.0 Bus
USB Device: USB2.0 Hub Controller
USB Device: Keyboard Hub
USB Device: Apple Optical USB Mouse
USB Device: Apple Keyboard
USB Device: hp LaserJet 1320 series
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: USB Bus
USB Device: USB Bus
USB Device: USB Bus
USB Device: USB Bus
USB Device: Bluetooth HCI
USB Device: USB Bus
USB Device: IR Receiver
FireWire Device: unknown_device, Unknown
Thunderbolt Bus: 

```

cjcliffe, 2018-03-14:
@stuzog I've seen this issue before; though unfortunately I believe it's related to the binaries being produced by the compiler using CPU instructions that are not supported by the Core2Duo line of processors. 

I'm unfortunately unable to test or fix this scenario properly since I don't have any Core2 OSX systems available for testing.

stuzog, 2018-03-14:
Yes, I think you’re correct, unfortunately. This particular computer is an early Core2 Duo and it seems to have some idiosyncrasies of its own. Never mind, these old systems will slowly fade away.
Stuart

closed 2018-05-07 by cjcliffe