cjcliffe/CubicSDR #611: Debian stable / undefined symbol

NeptuneTheMystic, 2018-02-04:
When attempting to run CubicSDR-0.2.3-x86_64.AppImage on Debian stable (Stretch), I get the error:

./CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_buffer_set_cluster_level

dimka-rs, 2018-02-16:
Same here:
```
$  ~/Downloads/CubicSDR-0.2.3-x86_64.AppImage 
/home/dimka/Downloads/CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_buffer_set_cluster_level

$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=17.04
DISTRIB_CODENAME=zesty
DISTRIB_DESCRIPTION="Ubuntu 17.04"
```

khaytsus, 2018-02-18:
Same here on Fedora 25, 26, and 27.  0.2.1 and earlier work fine.

righthalfplane, 2018-02-28:
The problem with the missing "hb_buffer_set_cluster_level" is very strange. On my ubuntu, 

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"

I do not have that problem. A search of  my libpangoft2 shows no such external -

dir@dir-pc:~$ nm /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.a | grep hb_buffer_set
                 U hb_buffer_set_direction
                 U hb_buffer_set_flags
                 U hb_buffer_set_language
                 U hb_buffer_set_script
                 U hb_buffer_set_direction
                 U hb_buffer_set_unicode_funcs

Looking at libpangoft2 -

 ls -l /usr/lib/x86_64-linux-gnu/libpangoft2*
-rw-r--r-- 1 root root 140040 Oct 23  2015 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.a
lrwxrwxrwx 1 root root     27 Oct 23  2015 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so -> libpangoft2-1.0.so.0.3800.1
lrwxrwxrwx 1 root root     27 Feb 11  2016 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 -> libpangoft2-1.0.so.0.3800.1
-rw-r--r-- 1 root root  90000 Oct 23  2015 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0.3800.1

I find that "libpangoft2-1.0.so.0" points to libpangoft2-1.0.so.0.3800.1. 

Try the commands - 

ls -l /usr/lib/x86_64-linux-gnu/libpangoft2*

and

nm /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.a | grep hb_buffer_set

and see what they show. It this point it looks like there may be two versions of libpangoft2-1.0

NeptuneTheMystic, 2018-02-28:
```
$ cat /etc/debian_version 
9.3

$ ls -l /usr/lib/x86_64-linux-gnu/libpangoft2*
-rw-r--r-- 1 root root 141736 Apr  8  2017 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.a
lrwxrwxrwx 1 root root     27 Apr  8  2017 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so -> libpangoft2-1.0.so.0.4000.5
lrwxrwxrwx 1 root root     27 Apr  8  2017 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 -> libpangoft2-1.0.so.0.4000.5
-rw-r--r-- 1 root root  87904 Apr  8  2017 /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0.4000.5

$ nm /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.a | grep hb_buffer_set
                 U hb_buffer_set_cluster_level
                 U hb_buffer_set_direction
                 U hb_buffer_set_flags
                 U hb_buffer_set_language
                 U hb_buffer_set_script
                 U hb_buffer_set_direction
                 U hb_buffer_set_unicode_funcs

```

righthalfplane, 2018-03-01:
I looked at the source of the libpangoft2 library (It is part of libpango1.0-0).

you find the statement -

#if HB_VERSION_ATLEAST(1,0,3)
  hb_buffer_set_cluster_level (hb_buffer, HB_BUFFER_CLUSTER_LEVEL_MONOTONE_CHARACTERS);
#endif

in the notes you see -

    Build: Require HarfBuzz 0.9.30

    Pango uses the HB_VERSION_ATLEAST macro, but it was only
    defined in HarfBuzz 0.9.30. Thus, compiling with an older
    version fails.

The error that is showing up is what you would expect to see if you built a program on a system with a libharfbuzz of version (1,0,3)  or greater and tried to run it on a system with libharfbuzz less than 1.0.3.  

I am using "CubicSDR-0.2.3-x86_64.AppImage" from "https://github.com/cjcliffe/CubicSDR/releases" and I do not see the problem. Running the ldd command in fact shows it to be using only a few system libraries with everything else statically loaded.

dir@dir-pc:~/Desktop/SdrPlay-Linux$ ldd -v CubicSDR-0.2.3-x86_64.AppImage
	linux-vdso.so.1 =>  (0x00007fff8fbfd000)
	libfuse.so.2 => /lib/x86_64-linux-gnu/libfuse.so.2 (0x00007fdf7c043000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fdf7be26000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fdf7bb15000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fdf7b8fb000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdf7b531000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fdf7b32d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fdf7c27f000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fdf7b0bd000)

	Version information:
	./CubicSDR-0.2.3-x86_64.AppImage:
		libfuse.so.2 (FUSE_2.5) => /lib/x86_64-linux-gnu/libfuse.so.2
		libpthread.so.0 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libpthread.so.0
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libfuse.so.2:
		libdl.so.2 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libdl.so.2
		libpthread.so.0 (GLIBC_2.3.2) => /lib/x86_64-linux-gnu/libpthread.so.0
		libpthread.so.0 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libpthread.so.0
		libc.so.6 (GLIBC_2.7) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.17) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libpthread.so.0:
		ld-linux-x86-64.so.2 (GLIBC_2.2.5) => /lib64/ld-linux-x86-64.so.2
		ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.2) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_PRIVATE) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libglib-2.0.so.0:
		libpthread.so.0 (GLIBC_2.12) => /lib/x86_64-linux-gnu/libpthread.so.0
		libpthread.so.0 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libpthread.so.0
		libc.so.6 (GLIBC_2.7) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.9) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.3) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.8) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.10) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.15) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.17) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libz.so.1:
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libc.so.6:
		ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
		ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
	/lib/x86_64-linux-gnu/libdl.so.2:
		ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
		libc.so.6 (GLIBC_PRIVATE) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libpcre.so.3:
		libpthread.so.0 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libpthread.so.0
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6

NeptuneTheMystic, 2018-03-01:
```
$ ls -l CubicSDR-0.2.3-x86_64.AppImage 
-rwxr--r-- 1 (me) (me) 14680064 Feb  4 09:13 CubicSDR-0.2.3-x86_64.AppImage

$ dpkg -l|grep harf
ii  libharfbuzz-dev:amd64                         1.4.2-1                                     amd64        Development files for OpenType text shaping engine
ii  libharfbuzz-gobject0:amd64                    1.4.2-1                                     amd64        OpenType text shaping engine ICU backend (GObject library)
ii  libharfbuzz-icu0:amd64                        1.4.2-1                                     amd64        OpenType text shaping engine ICU backend
ii  libharfbuzz0b:amd64                           1.4.2-1                                     amd64        OpenType text shaping engine (shared library)

```

righthalfplane, 2018-03-04:
I blacklisted some kernel modules to get CubicSDR-0.2.3-x86_64.AppImage partly working (vsonnier suggested this) -

edit/create in /etc/modprobe.d a file called blacklist.conf, which contains those three lines:

blacklist sdr_msi3101
blacklist msi001
blacklist msi2500
blacklist dvb_usb_rtl28xxu 

I added to last one to get my RTL stick working on Ubuntu 16.04 . Perhaps, this will also work on your system.

mohamad325, 2018-03-04:
Ok

Mohamad asadi.

tylla, 2018-03-07:
@righthalfplane I have the same symptom on Debian 9.3:
```
$ CubicSDR-0.2.3-x86_64.AppImage
CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-
gnu/libpangoft2-1.0.so.0: undefined symbol: hb_buffer_set_cluster_level
```
What I have noticed is that running as you asked:
```
$ nm /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.a |grep hb_buffer_set
                 U hb_buffer_set_cluster_level
                 U hb_buffer_set_direction
                 U hb_buffer_set_flags
                 U hb_buffer_set_language
                 U hb_buffer_set_script
                 U hb_buffer_set_direction
                 U hb_buffer_set_unicode_funcs
```

but the error message references `/usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0` (**.so** instead of **.a**) which is a stripped ELF binary, no wonder that there is no hb_buffer_set_cluster_level symbol in it.
Btw: libpangoft2-1.0.a is only available on the system if libpango1.0-dev was installed, which I did now, but it did not solved my problem.

My `ldd -v CubicSDR-0.2.3-x86_64.AppImage` looks almost exactly the same (there are some ordering differences, does this count as real difference?!).

As about the harfbuzz versions:
```
$ dpkg -l|grep harfbuzz
ii  libharfbuzz-bin                           1.4.2-1                                     amd64        OpenType text shaping engine (utility)
ii  libharfbuzz-dev:amd64                     1.4.2-1                                     amd64        Development files for OpenType text shaping engine
ii  libharfbuzz-gobject0:amd64                1.4.2-1                                     amd64        OpenType text shaping engine ICU backend (GObject library)
ii  libharfbuzz-icu0:amd64                    1.4.2-1                                     amd64        OpenType text shaping engine ICU backend
ii  libharfbuzz0b:amd64                       1.4.2-1                                     amd64        OpenType text shaping engine (shared library)
```$ dpkg -l|grep harfbuzz
ii  libharfbuzz-bin                           1.4.2-1                                     amd64        OpenType text shaping engine (utility)
ii  libharfbuzz-dev:amd64                     1.4.2-1                                     amd64        Development files for OpenType text shaping engine
ii  libharfbuzz-gobject0:amd64                1.4.2-1                                     amd64        OpenType text shaping engine ICU backend (GObject library)
ii  libharfbuzz-icu0:amd64                    1.4.2-1                                     amd64        OpenType text shaping engine ICU backend
ii  libharfbuzz0b:amd64                       1.4.2-1                                     amd64        OpenType text shaping engine (shared library)
```
Now that I read again @khaytsus's comment I realized that maybe really the .so vs .a difference is the key to this error. It tries to load a library which has no symbols.
Someone with some familiarity in the code could take a look at this, I'm totally lost in it.

HeroCC, 2018-03-07:
It looks like the last full release with a working appimage was (at least for me) 0.2.1-alpha-bookmarks. Later releases either didn't have an AppImage or didn't work (up to 0.2.3 Release Builds).

ivanenev, 2018-03-10:
Same error Ubuntu 17.10 

./CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_font_funcs_set_variation_glyph_func

cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=17.10
DISTRIB_CODENAME=artful
DISTRIB_DESCRIPTION="Ubuntu 17.10"

righthalfplane, 2018-03-14:
On Ubuntu 17.10, I worked around the problem that CubicSDR-0.2.3-x86_64.AppImage has with the undefined symbol to get it load and run. The partial history file included below shows most of the steps. I created a desktop folder call "share-source" - it has three folders in it, pango1.0-1.40.12, harfbuzz-1.4.2, and share. I set the CPPFLAGS to -fpic and rebuild the two libraries with the "--enable-static" option. I then copied the .o files into the share folder and created a new shared library called libshared.so. The next step was to replace the libpangoft2 library with the new one - this could have ill effects on other programs - so you may not want to try this - I did this just to see if it would get CubicSDR-0.2.3-x86_64.AppImage to run and it did. The new library has both harfbuzz and pango libraries in it so there is no undefined symbol. Copying libpangoft2-1.0.so.0.4000.12-save back to libpangoft2-1.0.so.0.4000.12 will restore the original library.

     1	6:54	apt-get build-dep libpango1.0-0
    11	7:02	apt-get source libpango1.0-0
    17	7:13	apt-get source harfbuzz
    36	7:38	cd pango1.0-1.40.12
    40	7:45	setenv CPPFLAGS -fpic
    43	7:46	./configure --enable-static
    44	7:46	make -j 4
    57	7:52	cd harfbuzz-1.4.2
    62	7:53	./configure --enable-static
    64	7:54	make -j 4
    68	7:56	mkdir share
    69	7:56	cd share
    72	7:58	cp ../harfbuzz-1.4.2/src/*.o .
    74	7:59	rm main-main.o test_buffer_serialize-test-buffer-serialize.o test_size_params-test-size-params.o test-test.o test_would_substitute-test-would-substitute.o
    79	8:00	cp ../pango1.0-1.40.12/pango/.libs/*.o .
    88	8:05	cp ../harfbuzz-1.4.2/src/hb-ucdn/*.o .
    92	8:06	gcc -shared -o libshared.so *.o
   102	8:17	pushd /usr/lib/x86_64-linux-gnu/
   105	8:18	sudo mv libpangoft2-1.0.so.0.4000.12 libpangoft2-1.0.so.0.4000.12-save
   106	8:19	sudo cp /home/dir/Desktop/share-source/share/libshared.so libpangoft2-1.0.so.0.4000.12

cjcliffe, 2018-03-22:
Going to try making a few AppImage builds using alternate VMs; I'll post the builds here when they're ready to test.

cjcliffe, 2018-03-23:
Hi All -- Here's an updated AppImage build using vagrant (instead of my usual ubuntu vm) from latest master branch:

[CubicSDR-0.2.3-x86_64.AppImage.zip](https://github.com/cjcliffe/CubicSDR/files/1840052/CubicSDR-0.2.3-x86_64.AppImage.zip)

Let me know if that fixes any issues and hopefully I can make that the standard build procedure.

NeptuneTheMystic, 2018-03-23:
Thank you, Charles.

Unfortunately I'm still getting:

```

./CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_buffer_set_cluster_level

dpkg -l|grep pangoft2
ii  libpangoft2-1.0-0:amd64                       1.40.5-1                                    amd64        Layout and rendering of internationalized text

```

cjcliffe, 2018-03-23:
Hmm, at https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues#harfbuzz I see they say:

> Bundling HarfBuzz seems to cause symbol lookup error: /usr/lib64/libpangoft2-1.0.so.0: undefined symbol: hb_buffer_set_cluster_level (suggesting that the version of HarfBuzz that is being bundled is too old for the lib64/libpangoft2-1.0.so.0 on the system; which means that we need to bundle HarfBuzz then we also must bundle libpangoft2-1.0.so.0). Not bundling HarfBuzz seems to cause symbol lookup error: /lib64/libharfbuzz.so.0: undefined symbol: FT_Get_Var_Blend_Coordinates on Solus 3, indicating that the HarfBuzz from the system needs a newer FreeType than what we are bundling
>
> HarfBuzz should be handled by the distribution and they should provide a clearly defined interface for applications to develop against. Build systems should use older versions than what is available in the oldest still-supported available distributions. Applications written against earlier versions of these should be guaranteed to still run with later versions of these.

I think that's suggesting I just have to do some additional work to make sure it bundles libharfbuzz and libpangoft2.  Will give it a try soon.

corq, 2018-03-27:
Still getting 

`./CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_font_funcs_set_variation_glyph_func`

Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful
Gnome 3.26.2

cjcliffe, 2018-04-03:
Hey all,

Here's another build to try; this time built using Vagrant /w Debian Jessie image:

[CubicSDR-0.2.3-x86_64.AppImage-Deb8.zip](https://github.com/cjcliffe/CubicSDR/files/1869898/CubicSDR-0.2.3-x86_64.AppImage-Deb8.zip)

I've set up some test VMs as well and It's working here on Debian 8, Ubuntu 15, 16, 17 and Skywave Live CD.

I tried several other Vagrant build VMs including CentOS 6/7, Debian Wheezy and other versions of Ubuntu without much luck -- either they had issues with wxWidgets 3.1.1, hamlib or dependencies  that were out of date.  Debian Jessie only required adding 'fuse' to the existing ubuntu 14.04 bootstrap and otherwise built without issues.

Thanks to everyone for the feedback and testing so far.

mcarden, 2018-04-03:
That new appimage gets me past the Undefined Symbol error on Fedora 26, and a cubicsdr GUI is generated. There doesn't look to be support for the SDRplay RSP1A though.

NeptuneTheMystic, 2018-04-03:
Success here with Debian 9.4 and a custom 4.13.11 kernel!

jhunolt, 2018-04-03:
New appimage is successful for me as well on Ubuntu 17.10 4.14.0 Plasma with RTL-SDRv3, thanks.

cjcliffe, 2018-04-03:
@mcarden Here's essentially the same build but I've remembered to include the SDRPlay support this time:

[CubicSDR-0.2.3-x86_64.AppImage-Deb8-with-SDRPlay.zip](https://github.com/cjcliffe/CubicSDR/files/1873761/CubicSDR-0.2.3-x86_64.AppImage-Deb8-with-SDRPlay.zip)

mcarden, 2018-04-04:
It's still not seeing the RSP1A and I see this error in the console:
"[ERROR] SoapySDR::loadModule(.//////////lib/SoapySDR/modules0.6/libsdrPlaySupport.so)
  dlopen() failed: libmirsdrapi-rsp.so.2.11: cannot open shared object file: No such file or directory"

It then goes on to say:
"Available factories...airspy, audio, bladerf, hackrf, null, redpitaya, remote, rtlsdr"

...and indeed an rtl-sdr does work just fine.

mcarden, 2018-04-04:
I have the API installed but no idea if it's the right one - I didn't think I had a choice. I exported LD_LIBRARY_PATH and /usr/local/lib does contain libmirsdrapi-rsp.so.2.11 and symlinks to the shorter names.
Looks like I need to dig deeper. Thanks.

SDRplay, 2018-04-04:
what OS and processor are you using?

mcarden, 2018-04-04:
OS: Fedora 26
Kernel: x86_64 Linux 4.15.12-201.fc26.x86_64
CPU: Intel Core i7-6600U @ 4x 3.4GHz

SDRplay, 2018-04-04:
Can you show the result of `ls -l /usr/local/lib/libmirsdr*`

mcarden, 2018-04-04:
$ ls -l /usr/local/lib/libmirsdr*
lrwxrwxrwx. 1 root root     36 Apr  3 17:20 /usr/local/lib/libmirsdrapi-rsp.so -> /usr/local/lib/libmirsdrapi-rsp.so.2
lrwxrwxrwx. 1 root root     39 Apr  3 17:20 /usr/local/lib/libmirsdrapi-rsp.so.2 -> /usr/local/lib/libmirsdrapi-rsp.so.2.11
-rw-r--r--. 1 root root 326008 Apr  3 17:20 /usr/local/lib/libmirsdrapi-rsp.so.2.11

SDRplay, 2018-04-04:
very odd that it says no file found. Have you run `sudo ldconfig`

Have you build SoapySDRPlay from master after installing the API?

mcarden, 2018-04-04:
I have run sudo ldconfig.
I'm testing an appimage prepared by cjcliffe earlier (scroll up to see): CubicSDR-0.2.3-x86_64.AppImage-Deb8-with-SDRPlay.zip
This is all pretty much to avoid having to build everything from source to get the RSP1A to run on Fedora.

SDRplay, 2018-04-04:
was the API included in the package you got? Was the same API used in the build of the image?

mcarden, 2018-04-04:
I don't know the answer to either question.

SDRplay, 2018-04-04:
Sounds like something specific to Fedora - I've just downloaded the same AppImage and ran it on Ubuntu just fine. Unfortunately I don't have a Fedora install to look at this on. I will try to get some time to install it and have a look.

cjcliffe, 2018-04-04:
@SDRplay AppImage is compiled against the API for SDRPlay but libmirsdr is not included in the binary -- it's expected to be installed separately.

SDRplay, 2018-04-04:
@cjcliffe yep that's what I did on Ubuntu and works fine. No idea why that doesn't work on Fedora. Will take some time for me to get an install to look into it.

cjcliffe, 2018-04-04:
@SDRplay I'll give it a go later as well; my initial guess is that perhaps /usr/local/lib/ is not part of the standard lib path on Fedora.

righthalfplane, 2018-04-04:
x86_64.AppImage-Deb8-with-SDRPlay.zip works on ubuntu 16.04, with my RSP2 but not with my RSP1. libmirsdrapi-rsp.so.2.11 seems to only work with the RSP2.

SDRplay, 2018-04-04:
The API supports all RSPs. If you can't get the RSP1 to work with it, then it means you need to blacklist the following kernel modules... sdr_msi3101, msi2500 and msi001, then reboot.

mcarden, 2018-04-04:
@SDRplay @cjcliffe  I have it running on Fedora now. It seems /usr/local/lib isn't common on Fedora, so I created a file, /etc/ld.so.conf.d/soapy_x86_64.conf  containing just the line, /usr/local/lib  then again ran ldconfig.
Magic! CubicSDR starts, finds the RSP1A, and in moments I'm listening to awful local broadcast FM. :)
Thank you both for your patient assistance.

cjcliffe, 2018-04-04:
@mcarden excellent, I'm glad to see it's working for you; I think that's most of the AppImage issues covered now.

@SDRplay might be good to detect Fedora with lib installer and use a standard system path or a custom path with an /etc/ld.so.conf.d/ entry.

SDRplay, 2018-04-04:
yes good idea.

righthalfplane, 2018-04-05:
CubicSDR-0.2.3-x86_64.AppImage-Deb8-with-SDRPlay runs on Ubuntu 17.10, but I get no sound. I have a 7.1 sound card, but I listen on a pair of usb ear phones. I build the Ubuntu 17.10 version of CubicSDR and I can select from any of nine possible output devices one of which is the usb ear phones - the sound then works. CubicSDR-0.2.3-x86_64.AppImage-Deb8-with-SDRPlay offers only PulseAudio as an output device and it output no sound. I just a endless list of error messages.

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !

tylla, 2018-04-06:
Last AppImage works on Debian 8.10 (kernel 3.16.0-4-amd64). Thanks!

cjcliffe, 2018-05-07:
Thanks everyone for reports and testing!

0.2.4 is available on the release page which now uses the same Debian 8 build scripts as the version provided here so I'm going to close up this issue for now -- anyone should feel free to open a new one if there's 0.2.4 related issues.

@righthalfplane that sounds like it might be an issue unrelated to AppImage specifically.

Please try compiling it locally in Ubuntu 17 using the build instructions in the Wiki; alternatively 0.2.3 appears to be available through the standard packages in Ubuntu 18.   If the problem persists please open a new issue which should get more visibility to others who might know better than myself.

closed 2018-05-07 by cjcliffe