cjcliffe/CubicSDR #614: "SDR Devices" dialog only allows three digit entry for offset frequency

lutusp, 2018-02-07:
In version 0.2.3, when using the "Ham It Up" downconverter, one must enter a frequency offset of -125000 KHz (i.e. 125 MHz), but the "SDR Devices" dialog, with units of KHz, only accepts three digits, making this setting impossible.

One can recover by opening "Settings ... Frequency Offset", which does allow more digits, but this won't be likely to occur to many users.

vsonnier, 2018-02-07:
@lutusp  Thanks, it was a bug. I fixed it in https://github.com/cjcliffe/CubicSDR/commit/490f72ceebd1c87c762e4ff38f2169d301b13a66 in master.

closed 2018-02-07 by vsonnier

lutusp, 2018-02-07:
I should have been more detailed in my original bug report. The unmentioned added problem is that the frequency offset entered into the "SDR Devices" dialog is also used to correct any clock errors in the connected dongle. For example, using SDR# I can enter a frequency offset in Hertz, which allows me to completely correct clock errors and submit a required downconverter offset in one entry. So, for a particular dongle, used in an application where exact frequency correction is required, and after some experimentation, I might enter:

-124999080 Hz

The above example (1) sets a downconverter's required offset and (2) fully corrects a difference between a particular dongle's quartz oscillator and reality, expressed in Hertz. This is needed to allow such things as AIS reception, just one of several examples where parts-per-million corrections are required.

It's true there's an additional PPM correction entry (Settings ... Device PPM), but it would be nice to have a single entry that does both, as most of the other SDR clients do, because one might use different dongles for different purposes, each of which requires a different clock correction entry. As things stand, each of them requires two entries instead of one.

So I ask that the frequency entry in the "SDR Devices" dialog have units of Hertz, not kiloHertz. Thanks!

vsonnier, 2018-02-07:
@lutusp The frequency offset if just an offset in the user displayed frequency, nothing more, for use with downconverters and upconverters. I changed it from Hz precision to KHz at some point because some down/upconverters needed more than 2Ghz of shift, and the UI limitation prevent entering values > +-2**31.

This is not for correcting __real__ clock offsets at all. Indeed PPM corrections are there just for that, and represent the __reality__ of the device frequency offsets, because it is meaningless to be precise at Hz there due to the imperfections of the hardware and better expressed as a fraction of the current frequency, hence the PPM. 
The PPM setting is saved by device, so you can effectively use different PPM on different devices.

The other SDR clients have exactly the same concept of "displayed frequency offset" + "PPM correction", and in your example you were just using the frequency offset as a PPM correction, but it sounds wrong to me.

lutusp, 2018-02-07:
> ... and in your example you were just using the frequency offset as a PPM correction ...

Both, actually -- it sets both upconverter offset and correction at the PPM level. Both Gqrx and SDR# implement it that way. It just saves time for people who change devices regularly, as I do.

As things stand, once I saw the problem with 0.2.3, I immediately returned to 0.2.2, which allows an entry to within one kilohertz of the desired rate -- close enough but not as good as that found in Gqrx and SDR#, both of which allow entries that resolve about ten Hz.

> ... but it sounds wrong to me.

Fair enough, but consider that, regardless of how they're entered, ultimately the user's chosen frequency offset and correction are combined and submitted to the dongle, which is able to accept a large offset value that resolves ten Hertz, as shown here in a live test:

https://i.imgur.com/ncBycTh.png

In the above image, I'm listening to a heterodyne, and by spinning the mouse wheel I see and hear ten-Hertz frequency increments. It's essentially a perfect way (within the known limitations of the dongles) to produce a large offset to simultaneously meet the offset requirements of an upconverter and fix small clock errors -- at once. 

> I changed it from Hz precision to KHz at some point because some down/upconverters needed more than 2Ghz of shift, and the UI limitation prevent entering values > +-2**31. 

Only because the value being passed by "getPropertyValue().getInteger()" is an int (32 bits) -- this could be solved with "getPropertyValue().getLongLong()" (returns a signed 64 bit value). It is immediately placed in a long variable (64 bits), but too late to preserve an entered value with more resolution.  By contrast, an IEEE-754 double has a 53-bit mantissa, and that's what appears to be in use in these other programs. A 53-bit value that resolves ten Hertz can accept a largest value of 900 GHz. Just saying. :)

But barring the above suggested methods, please at least go back to the scheme that was present in 0.2.2, which allowed one KHz increments -- it's a compromise, but better than the present method that only allows ... wait for it ... one MHz steps.

Please don't take this as criticism -- I think CubicSDR is a terrific program with an ingenious interface, one that I much prefer over the alternatives ... with this one small problem.

commit 490f72ce (2018-02-07): Fix #614: Wrong Freq offset applied if set from SDR Devices