cjcliffe/CubicSDR #615: Running from a relative path

gvanem, 2018-02-14:
Running `CubicSDR.exe` from the command-line in the directory of `CubicSDR.exe` is no problem.

But running from the command-line from a directory relative. E.g. `..\bin\CubicSDR.exe` is a problem;
it fails to find all the `share/font/*.fnt` files and I get *no* sound!.

parkerlreed, 2018-02-14:
You would have to create a shortcut and set the working directory in the properties dialog. The Windows build is probably hardcoded to act like a portable install so it's always going to look for assets in the same folder you run it from.

gvanem, 2018-02-14:
A `GetModuleFileName (NULL, my_name. sizeof(my_name));` could fix it. But OK.
I don't follow you on the *portable install*. I have lot of portable apps here that can run from any directory.

vsonnier, 2018-02-14:
@gvanem  As the Wiki says for [Build Windows](https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows) you need to build with `BUILD_INSTALLER=1`. In this case, `fonts` are then searched relative to the executable. 
If not set, fonts are searched in `../share/cubicsdr/` whatever that means.

Since I built with BUILD_INSTALLER = 1, I tried to run using a relative path as you did and it works for me.

gvanem, 2018-02-14:
> Since I built with BUILD_INSTALLER = 1,  I tried to run using a relative path as you did and it worked for me. 

Which means `R-DES_FOLDER="../share/cubicsdr/"`.  The `RES_FOLDER` is a compile-time constant. Nothing to do with a run-time directory variable.
IMHO, it should be *<the-dir-of-CubicSDR.exe>/share/cubicsdr/*. How  hard could that be? Must be something in wxWigets for this..

vsonnier, 2018-02-15:
@gvanem Hello, I've comitted some changes to make the `fonts` search smarter:
- First search for : `RES_FOLDER/fonts`
- else search for `[Cubic exe path]\RES_FOLDER\fonts`,
- else search for `[Cubic exe path]\fonts`

I think it covers your issue now.

gvanem, 2018-02-15:
@vsonnier : thanks for your effort. That fixed it.

commit c1e60f31 (2018-02-15): Fix #615 Be smarter when looking for font resources