cjcliffe/CubicSDR #616: WAV file header writes incorrect chunk length

nsmith-, 2018-02-18:
Most wav readers don't seem to care if they reach `EOF` before the claimed chunk length, but apparently GNUradio does, and so I was having issues with the WAV source block.
The chunk header always claims the length is 16 bytes longer than it is.  The [offending line](https://github.com/cjcliffe/CubicSDR/blob/c1e60f31f511d0bcfbae83389619d707ffb73198/src/audio/AudioFileWAV.cpp#L109) should be changed to:
```c++
 write_word(outputFileStream, file_length - (dataChunkPos + 8));
```

Here is a python script that fixes existing wav files:
```python
import struct
import sys

fname = sys.argv[1]
f = open(fname, "r+b")
f.seek(4)
f_len = struct.unpack("<i", f.read(4))[0] + 8

f.seek(40)
data_len = struct.unpack("<i", f.read(4))[0]
hdr_len = 44
correct_data_len = f_len - hdr_len

if correct_data_len - data_len != 0:
    print "File '%s' data chunk size is %d bytes shorter than it should be, fixing" % (fname, data_len - correct_data_len)
    f.seek(40)
    f.write(struct.pack("<i", correct_data_len))
```

vsonnier, 2018-02-18:
@nsmith- Many thanks for your fix! I've just pushed it into the master.

nsmith-, 2018-02-18:
Thanks! Good idea to explicitly write 4 bytes, since `sizeof(size_t)` would be 8 on 64-bit platforms and trash the first sample.

vsonnier, 2018-02-19:
> Thanks! Good idea to explicitly write 4 bytes, since sizeof(size_t) would be 8 on 64-bit platforms and trash the first sample.
My thoughts exactly :)
Is is OK then on your side, so this issue can be closed ?

nsmith-, 2018-02-19:
Yes, I think this is resolved.

commit 92eb5109 (2018-02-18): Fix #616 wrong WAV size reported in header (thanks @nsmith-)
commit 7f732735 (2018-02-18): Fix #616 wrong WAV size reported in header (thanks @nsmith-)
closed 2018-02-19 by nsmith-