cjcliffe/CubicSDR #617: Suggestion: "Display" option to always show an IFGR and RFGR sliders.

Magalex-2x14, 2018-02-18:
Hello everybody!

How difficult is it to make an option with which you can pin an IFGR/RFGR level sliders even when the IF AGC is on? 
I'll explain - I work in a rather noisy environment, and I have to work a lot with the amplification / attenuation levels. On HF, this is especially useful, because for effective interference suppression I have to increase the gain of the active antenna while simultaneously increasing the attenuation in the SDRplay receiver and making sure that the IF/RF levels are within acceptable limits (thus I practically get rid of the interference received by the long cable and the input circuits).  I get a very satisfactory results leaving the IF AGC on, and working only with RFGR, but I have to often access the pulldown menu, and it's tiresome, because this is a lot of clicks...

In my opinion, such an option (always show an IFGR and RFGR sliders, disabled by default) will not complicate the interface, but will increase usability in certain conditions.

Sorry for bad English.

SDRplay, 2018-02-18:
I've discussed at least having the RFGR usable whilst AGC is enabled with @vsonnier which I believe can be done, but needs more information to/from SoapySDR to be able to do it. You certainly can't use IFGR whilst AGC is enabled, so we wouldn't want to give the impression that you can, which may be the case if it's displayed. However the RFGR is totally independent of the AGC functionality, so should be easily usable ideally.

Magalex-2x14, 2018-02-18:
Maybe I'm misunderstanding something, or my (and Google translator :) English is guilty of misunderstanding, but the RFGR setting works as it should while the IF AGC is enabled. But, unfortunately, user must use the drop-down menu instead of RFGR slider, since the both IFGR/RFGR sliders disappear when the AGC is activated. In addition, after switching off the AGC, the IFGR slider is in the position corresponding to the last one established by the AGC, so, it seems to me, it would be very convenient to be able to keep both sliders visible, and keep updating shaded (or maybe directly marked with "AGC" text) IFGR slider with the current value set by the AGC.
I'm not a programmer, but I understand something, and I'll try to dig into this as soon as I find enough time. But seems to me, that subject option is piece of cake for project authors :)

SDRplay, 2018-02-18:
I think we are agreeing. My understanding is that although the IF and RF gain is independent, when AGC is enabled, currently both sliders disappear. I don't think it's currently possible to have either one slider visible or to have both sliders visible but one greyed out. I think it needs changes to both SoapySDR and CubicSDR, but maybe the authors of those applications can comment.

vsonnier, 2018-02-20:
Hello @Magalex-2x14 @SDRplay. I thought a bit of a Display customization here which would only be on the CubicSDR side without changing the SoapySDR API at all:
- Add in Display menu a by-gain option to make it always visible in AGC mode,
- Add some kind of visual clue in the Gains area, ex a pulsing overlay displaying "AGC on" on top of the gain area somewhere
- This would not prevent user input at the same time. The set/get feedback loop should be already there in the code following the changes I made to synchromize gains display and menu settings.

SDRplay, 2018-02-20:
Please note that it's not possible to manually modify the IFGR whilst the AGC is enabled, so that shouldn't be possible, don't want to confuse anyone or give the impression that it's possible. The AGC should not cover the RFGR, again to not give the impression that the AGC affects the RFGR. The AGC is IF only. The RSP1A behaves in the same way as the RSP2. The RSP1 can also behave in the same way, except the RF gain is on/off as apposed to a range of values, but could be displayed in the same way. Other than that, sounds good :-)

vsonnier, 2018-02-20:
@SDRplay Thanks, but don't get too impatient because I don't know exactly when I have time to do this change, which could quite long to get right when UI is involved.

Magalex-2x14, 2018-02-21:
I dig a little in the code, and strictly forbidden to hide the controls, changing code a little. It seems to work as it should :) It can be seen now how the receiver changes the IFGR in real time and reacts to the RFGR changes :) Very cool and handy :) 
With the menu item, I quickly did not manage to figure it out - need to dig to wxWidgets, so far I am enclosing my changes in the form of screenshots, in case anyone finds it useful (i don't know, how to do it right with github).

[image]
[image]

Magalex-2x14, 2018-02-27:
A few observations from the results of my experiments with CubicSDR+RSP2 gain controls:

1. Gain sliders pinning works. And I really like this :) But sometimes (could not catch under what conditions), when IF AGC is on, IFGR slider loses sync with real GR value. Reenabling AGC at every RFGR change seems to solves the problem. Not sure, who is to blame, but maybe it must be done at the driver level?

2. Sometimes (quite rare), after disabling AGC, CubicSDR is trying to set strange IFGR value - -2.14748e+09 (overflowed?)

3. The code responsible for checking the difference between the old and new gain value (GainCanvas.cpp) does not work - CubicSDR sets the gain every time at each smallest step of the slider.

4. Finally, there is a nasty snag with the maximum RF gain reduction level - it depends on the current frequency range... Choosing a value that is outside the specified range sometimes results in CubicSDR hang. Table of dependencies from the official SDRPlay API docs under the spoiler:
<details/>

[image]

</details>

cjcliffe, 2018-03-14:
I think a solution here might be to introduce a new option to the SoapySDR interface -- an indication as to which gains are actually part of the AGC.  Right now it's assumed all gains are either AGC ON or AGC OFF instead of specifying "These Gains are part of AGC".  This would easily let us know for any device which gains are actually part of the AGC set so that unrelated gain values could still be shown and adjusted without issues.

vsonnier, 2018-03-17:
I agree with @cjcliffe. Given my past efforts, I'm finally no longer willing to invest time in another complicated workaround solution.
To be honest, I almost never use AGC myself on RSP2 which I feel less efficient than a manual gain, especially now when it's so simple to move the gain sliders so I'm lacking motivation here. Of course it could be "cool" to have a "dynamic slider" effect when AGC is on, but it would need quite an effort to make it right as @Magalex-2x14 discovered. 
It would probably only work on SDRPlay(s) too, given the state of SoapySDR modules.

Magalex-2x14, 2018-03-18:
OK! I agree with all the arguments. Let's close it?

closed 2018-03-20 by Magalex-2x14