cjcliffe/CubicSDR #619: Gain changes, strange behavior.

Magalex-2x14, 2018-02-24:
During the experiments with #617 I noticed a strange behavior:
1. AGC is off.
2. RF Gain slider changes do not lead to visible effects.
3. The RF gain values of the slider and in the pulldown menu is differ.
4. Subsequent RFGR slider changes can lead to visible effects or not, but the gain value in the menu lags behind the slider (by 1 or more sometimes).
5. Subsequent RF gain changes with pulldown menu always lead to slider syncronization, but no visible effects (often, but not always).

IFGR slider changes (AGC is off) after a noticed problem has no visible effects.
Stopping and then starting device after a noticed problem leads to endless messages in terminal (with freezed waterfall):

```SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
```

I thought at first that the reason for this was my crooked playful hands, but after redownloading the 0.2.3 from releases page and reinstalling SoapySDR libs from sdrplay.com downloads section I was convinced that the problem there is also present.

CubicSDR v0.2.3 release,  SDRplay RSP2pro

```
mini-2x14-2:~ Aleksey$ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-g285e72aa
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so
Loading modules... done
Available factories...null, sdrplay,
```
[image]

Magalex-2x14, 2018-02-24:
It seems that the problem (or part of a problem?) is relevant for frequencies only above about 400 MHz... Maybe i missed something from the docs. Seems like actual RFGR steps is relevant to current frequency band, that is not considered in the program or driver...

vsonnier, 2018-02-24:
@Magalex-2x14 Hello ! Due to rounding errors, sometimes the RFGR gain slider will de-synchronize from the menu value, but as you have seen a menu change always restore the consistency. I've already tried to fix it in all corner cases for __hours__ during my previous menu  <==> slider synchro work. 
Sometimes  you can't win against floating point, and I'm not going to invest more time on this particular glitch. 
If you want to see another problem I'm definitely not trying to fix anymore, try setting the Rf gain in the SDR settings initial dialog, and see what happens. (well, nothing, it has no effect)
It's a chicken-and-egg problem, amplified by the startup sequence. 

As for your experiments, let's stay focused and keep the discussion about it in #617, shall we ?

closed 2018-02-24 by vsonnier