cjcliffe/CubicSDR #620: segmentation fault at start

imesg, 2018-02-24:
Start CubicSDR in terminal and get and need some help figuring why.

Thanks,

gene@go3:~# CubicSDR

RtApiAlsa::getDeviceInfo: no supported sample rates found for device (hw:0,0).

Audio Device #0 
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 1
	Duplex channels: 0
	Native formats:
	Supported sample rates:

Audio Device #1 hw:HDA Intel PCH,0
	Default Output? No
	Default Input? No
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
	Supported sample rates:
		44100hz
		48000hz

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:1,3), Device or resource busy.

Audio Device #2 
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 0
	Duplex channels: 0
	Native formats:
	Supported sample rates:

Audio Device #3 hw:HDA Intel PCH,7
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 8
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
	Supported sample rates:
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #4 hw:HDA Intel PCH,8
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 8
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
	Supported sample rates:
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #5 hw:HDA Intel PCH,9
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 8
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
	Supported sample rates:
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #6 hw:HDA Intel PCH,10
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 8
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
	Supported sample rates:
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #7 default
	Default Output? No
	Default Input? No
	Input channels: 32
	Output channels: 32
	Duplex channels: 32
	Native formats:
		16-bit signed integer.
		24-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		4000hz
		5512hz
		8000hz
		9600hz
		11025hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Segmentation fault (core dumped)

vsonnier, 2018-02-25:
@imesg Looks like you are not supporting ALSA audio properly and that make CubicSDR crash. Try to compile CubicSDR by excluding this audio API and including others, i.e add on Cmake command line:
`-DUSE_AUDIO_PULSE=1 -DUSE_AUDIO_OSS=1 -DUSE_AUDIO_ALSA=0`  for instance.

vsonnier, 2018-03-09:
@imesg Ping ! Hello, did you make an attempt at recompiling Cubic with the suggested options ?

imesg, 2018-03-11:
Yes, I was already doing it that way. I  am using gentoo linux which installs every with an ebuild. Contained therein was this line:
-DUSE_AUDIO_PULSE=1 -DUSE_AUDIO_OSS=1 -DUSE_AUDIO_ALSA=0
the one you suggested. This however does not work. Fails with the original error. I am still trying to figure it out. I have tried different options with the audio stuff including rtaudio.

This is, however, installed by the portage package manager. I can manually compile and run it successfully. I am still trying to figure it out.

Thanks for your help.

vsonnier, 2018-03-25:
Hello @imesg ! It may be related to this RtAudio issue: https://github.com/thestk/rtaudio/pull/129 
It's a small change, so maybe you can give it a try. 
Beware though, from that commit https://github.com/cjcliffe/CubicSDR/commit/03eb22a5c78240763a6970d7a2f9bc394bf8ad44 CubicSDR tends to use the system RtAudio, not the `external`-provided one so you have to be sure to recompile with the right one.

vsonnier, 2019-03-31:
Stale issue, no feedback : time to close it.

closed 2019-03-31 by vsonnier

katp32, 2020-12-26:
Correct me if I'm wrong, but this issue was never resolved? On my system, CubicSDR crashes in the exact same way, also most likely due to ALSA. However, my system's ALSA configuration is perfectly functional (and in fact I've tried a variety of different configurations, all with the same result), and yet CubicSDR still crashes. It seems this is not an issue with the system's ALSA configuration, but rather, CubicSDR not supporting ALSA correctly...

pylover, 2021-12-24:
Exact segmentation fault in Ubuntu2.04 and master branch.

```bash
.
.
.

Audio Device #5 hw:HDA Intel PCH,9
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 8
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
	Supported sample rates:
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #6 hw:HDA Intel PCH,10
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 8
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
	Supported sample rates:
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Segmentation fault (core dumped)
```

kswit, 2024-04-02:
If under Wayland display server have hint:
`WAYLAND_DISPLAY= CubicSDR`