cjcliffe/CubicSDR #621: db scale on spectrum displays

Toontje, 2018-02-25:
Maybe i'm missing something and it's already there, but if not, would it be possible to show a dB scale on the spectrum displays?

Toontje, 2018-02-25:
What does this value refer to? It changes greatly whenever i change the spectrum averaging.
[image]

vsonnier, 2018-02-25:
@Toontje Hello ! As of today, only a floor / ceil values are displayed indeed. I would also like to have a way to display the dB value as a proper scale attached to the horizontal lines already drawn in the spectrum as well.

As for the scale range management CubicSDR continuously tries to adapt the ceil /floor according to the realtime min/max values of the visible FFT, more or less like:
`floor/ceil dB = 20 * log10(FFT floor/ceil)`. 
(This question is also related to #531)

CubicSDR tries to shows the user only the relevant information, by "auto-zooming". The same principle applies on the colors of the waterfall which are self-adapting to offer maximum contrast.
This is unique to Cubic, the others SDR soft relies on additional zoom controls so the user can manually adapt the display.
 
For me, it was the killer feature (with OpenGL realtime) display that bought me in for CubicSDR compared to the other SDR softs out there which feel clumsy by comparison. 

The drawback is that sometimes the constantly changing scale is annoying if you want to see the relative levels over a period of time. There are 2 workaround for that:
- Activate the peak (P) display so that the scale will self-stabilize because of peak saturation,
- On the spectrum region, hold Right-Click and drag up and down: the scale will change according to it and no longer self-adjust until you click again on the Spectrum region.

Toontje, 2018-02-25:
Hi @vsonnier!  Thanks for the explanation. Yeah, it's the auto-scaling and not knowing what power level i am looking at that trips me over sometimes. Thanks for the Right-Click. There is so much not (yet) documented. ;-)
One more question: If the signal doesn't change, why does the floor value change when i change the spectrum averaging?

vsonnier, 2018-02-26:
@Toontje I think it make sense:
the formula is roughly `min/max dB = 20 * log10 (  min/max(FFT_with_spectrum_averaging)))`
Forgot the  `log10` which is irrelevant here, it comes to find a min/max over `FFT_with_spectrum_averaging` samples.
Now `FFT_with_spectrum_averaging` takes the `FFT_samples` and mean them over a period of time, the smaller the spectrum averaging setting the longer the time `FFT_samples` are mean together. 

Intuitively, the longer you mean those FFT samples over time the more the min/max peak values in the sample batch are smoothed out, so to speak: the more you integrate samples over time, the more the individual min/max values disappear.

With this very un-mathematical explanation you can explain what we observe: the faster the spectrum is refreshed, the peak values (either minimum or maximum) have more chance to stand out, so the displayed dB range raises.

Toontje, 2018-02-26:
un-mathematical explanation accepted. ;-)

closed 2018-02-26 by Toontje