cjcliffe/CubicSDR #622: CubicSDR 0.2.3 hits and misses on ubuntu 16.04 LTS

righthalfplane, 2018-02-27:
I booted ubuntu 16.04 LTS on my old PC to try CubicSDR 0.2.3.

With a fresh build of CubicSDR-0.2.3.tar.gz, CubicSDR did not see my Sdrplay 1 (RSP1), but my SdrPlay 2 (RSP2) seemed to work perfectly.

With a fresh build of CubicSDR-0.2.3.tar.gz and a one line change in my revised Macintosh version of discover_netsdr, CubicSDR work perfectly with my NetSDR on ubuntu 16.04 LTS. The original version of discover_netsdr does not work on ubuntu causing CubicSDR to fail for the NetSDR.

With a fresh build of CubicSDR-0.2.3.tar.gz, CubicSDR had the same problem with the RTL stick as CubicSDR-0.2.3-x86_64.AppImage

With a fresh build of CubicSDR-0.2.3.tar.gz, CubicSDR did not work with my Hackrf One. The utilities that come with new hackrf library seem to work fine, but CubicSDR and SoapySDRUtil have problems. The Hackrf One is the only one of my devices that fail with CubicSDR-0.2.3 on my Macintosh. It does work with some versions of gqrx and gnuradio on the Macintosh and Linux.

With CubicSDR-0.2.3-x86_64.AppImage, my RTL stick failed  -
***********************************
Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
SDR thread starting.
device init()
[INFO] Using format CF32.
Device Stream MTU: 131072
starting readLoop()
SDRThread::readStream(): 2. SoapySDR read failed with code: -1
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
**********************************
With CubicSDR-0.2.3-x86_64.AppImage, Hackrf One failed with -
***********************************
SoapySDR init..
	API Version: v0.6.0
	ABI Version: v0.6
	Install root: ./////////
	Loading modules... 
	Available factories...airspy, audio, bladerf, hackrf, null, redpitaya, remote, rtlsdr
[WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
  setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = PulseAudio
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

  device = HackRF One
  driver = hackrf
  label = HackRF One #0 406464c823692c4b
  part_id = a000cb3c004e474e
  serial = 0000000000000000406464c823692c4b
  version = 2014.08.1
Make device 1
[INFO] Opening HackRF One #0 406464c823692c4b...
[INFO] Could not Open HackRF Device
Error making device: hackrf open failed
********************************************

righthalfplane, 2018-02-28:
Got the RTL stick to work - the linux loader incorrectly linked in two version of the same modules into CubicSDR - removing the extra set of modules made everybody happy.

vsonnier, 2018-02-28:
@righthalfplane Glad to here it ! So the remaining problems are Applimage ones ? In this case I suggest contributing to the other already existing open issues about it.

vsonnier, 2018-03-01:
@righthalfplane Right, so let's try not to mix too many problems and stick to the Ubuntu 16.04 one. You have a RSP2 working but your RSP1 is not, so I assume you have correctly installed the latest API/HW driver from www.sdrplay.com/downloads, i.e v2.11.

If you used a RSP1 before you probably know you have to blacklist kernel modules to make it work like this:
 edit/create /etc/modprobe.d called blacklist.conf, which contains three lines:
````
blacklist sdr_msi3101
blacklist msi001
blacklist msi2500
````
You can also try to rebuild the whole CubicSDR + related SoapySDR modules using the excellent scripts @loughkb made: https://github.com/loughkb/SDRplaylinux

I used them to compile on Ubuntu 16.04 hosted on Windows 10 without problems.

righthalfplane, 2018-03-01:
I used the RSP1 on MacOS Sierra, MacOS High Sierra and Windows 7. It has only been 2 weeks on Ubuntu. Yours is the first mention I have seen of the need to black list those modules - that got the RSP1 going, but it broke the stick. I had to blacklist dvb_usb_rtl28xxu to get the Stick going again. So at this point 4 out of my 5 SDRs work on Ubuntu and MacOS Sierra. RSP1,  RSP2, NetSDR, and the RTL stick all work with CubicSDR-0.2.3 . HackRF One no longer works with CubicSDR-0.2.3 - the first time that I tried CubicSDR, many months ago on some old version of MacOS, HackRF One did work. HackRF One works with gqrx and gnu radio-companion on MacOS Sierra and Ubuntu.

CubicSDR-0.2.3-x86_64.AppImage now works with the RTL stick, but fails with the HackRF One. It does not seem to have support for RSP1,  RSP2 or NetSDR built in. Is there more than one version of CubicSDR-0.2.3-x86_64.AppImage ? I do not see the errors that the other people are seeing and ldd shows no links to the library modules that they are getting errors about.

righthalfplane, 2018-03-02:
I had had RSP1 and RSP2 working fine with SDRuno. I installed CubicSDR on windows 7 and copied the driver across, but it did not work. I then make the mistake of running SDRplay_RSP_API-Windows-2.11.1 again - the driver install failed. SDRuno no longer runs. uninstalling and reinstalling  SDRplay_RSP_API-Windows-2.11.1  does not help. I forgot that I had done that before - I had to do a full system restore to get things going again. Is there anyway to get things going with out a full system restore ?

vsonnier, 2018-03-02:
@righthalfplane A prerequisite for Windows is to install a "Visual C++ 2015 (or 2017) redistributable", go Google for the latest version.
Now SDRUno installs the HW API/DRIVER together with the main program, so installing the HW/API Driver alone should have the same effect. 
The HW/API installer also says to only plug the device AFTER the complete installation, so take care of following this advice.

righthalfplane, 2018-03-03:
I resolved some of the problems with the drivers by installing - Microsoft Patch "Windows6.1-KB3033929-x64.msu" . I found the information  about patch on the www.sdrplay.com Forum.

[ Windows 7 patch](https://www.sdrplay.com/community/viewtopic.php?f=7&t=383&p=1472&hilit=3033929#p1472)

After installing that patch the drivers for RSP1 and  RSP2 installed and worked with SDRuno and SDRConsole. The PothosSDR-2018.02.05-vc14-x64 programs, CubicSdr, gqrx and SoapySDRUtil all seemed to run OK, but they would fail trying to start up RSP1, RSP2, Hackrf One or the RTL stick. SoapySDRUtil claimed that the modules were there (--info), but they did not work(--probe) with any of the devices. CubicSDR-0.2.3-win32 also ran, but would not find any of the devices. The install files had two version of the 64 bit mir_sdr_api.dll file - both worked with SDRuno and SDRConsole - neither of them worked with the CubiSDR stuff.

SDRplay, 2018-03-03:
Remember there is a difference between apps that run on 64bit systems and apps that are 64bit. If you don't have any luck with the 64bit API, try the 32bit API which will still work on a 64bit system. What's important, is what the application is looking for.

righthalfplane, 2018-03-03:
I got CubicSDR-0.2.0-win32 and CubicSDR-0.2.0-win64 to work by accident. The correct  files happen to be left behind from a previous install. It turns out that the mir_sdr_api.dll files that are installed by SDRplay_SDRConsole_Installer_2.3_b2760_2 are compatible with CubicSDR-0.2.0 but not with CubicSDR-0.2.3. The mir_sdr_api.dll files that are installed by SDRplay_SDRConsole_Installer_3.0_961 do not work with  CubicSDR-0.2.0 or  CubicSDR-0.2.3. The mir_sdr_api.dll files that are installed by SDRplay_HDSDR_Installer_2.76_3 do not work with  CubicSDR-0.2.0 or  CubicSDR-0.2.3. The mir_sdr_api.dll files that are installed by SDRplay_RSP_API-Windows-2.11.1 do not work with  CubicSDR-0.2.0 or  CubicSDR-0.2.3. The bottom line is I have not found and mir_sdr_api.dll file that work with CubicSDR-0.2.3 on Windows 7.

SDRplay, 2018-03-03:
I'm a bit confused. Are you copying the dull from the API installation directory or SDR-Console installation directory? Can you tell me the path and filesize of the dll that works please?

righthalfplane, 2018-03-04:
Each of these install programs create or write into the Folder "Program Files/SDRPlay". They create various versions of C:\Program Files\SDRplay\API\x64\mir_sdr_api.dll and C:\Program Files\SDRplay\API\x86\mir_sdr_api.dll files. The ones created by "SDRplay_SDRConsole_Installer_2.3_b2760_2" were found to work with CubicSDR-0.2.0-win32 and CubicSDR-0.2.0-win64. None of the others worked with either version of CubicSDR-0.2.3. The 64 bit version of mir_sdr_api.dll that worked with CubicSDR-0.2.0-win64 was 230KB. The 32 bit version of mir_sdr_api.dll that worked with CubicSDR-0.2.0-win32 was 234KB.

SDRplay, 2018-03-04:
So there are a number of things here. SDR Console doesn't support the RSP1A so the version of the API that is included in that installer is 2.10 (Sept 2017) - this is also the same API DLL that comes with CubicSDR 0.2.3 64bit (that's the one I've just tested). I've also just copied in the 64bit API DLL that does support the RSP1A which is 2.11 (Nov 2017) and that also works. So from a 64bit perspective, I've just done the download from the cubicsdr website and everything works. I'll now check the 32bit version as well.

by the way, you do have the C++ redist installed? Here's the link and you need the right version for your OS: https://www.microsoft.com/en-us/download/details.aspx?id=48145

SDRplay, 2018-03-04:
I just installed the 32bit CubicSDR 0.2.3, copied in the 2.11 API (Nov 2017) and it's working well with the RSP1A - so I can only think that it's the redist package that you need. Not sure what else it can be.

vsonnier, 2018-03-04:
@SDRplay @righthalfplane A thing I noticed when API v2.11 was released is it needed the Visual C++ __2013__ runtime, while the previous versions didn't need any particular version.
So here we are, assuming you are on a x64 system:
- SDRUno latest is 32 bit, and __probably__ installs the "Visual C++ 2013 redistributable 32 bit" as part of the installer,
-  For CubicSDR v0.23 x64 and the [PothosSDR win64 distribution](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D) themselves, you need to install the "Visual C++ 2015 Redistributable x64" to run,
-  But for the SDRPLay(s) to be recognized by both CubicSDR(s) installed above, you need to also install the "Visual C++ 2013 Redistrubutable x64".
- Install the API/HW Driver v2.11.1 last to be sure, and copy the x64 `mir_sdr_api.dll` into the Cubic v0.23 installation directory. The [PothosSDR win64 distribution](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D) packaged one have already the right one in its installation directory.

Bottom line:
- Reinstall API/HW Driver v2.11.1,
- Copy the x64 dll at it appropriate place, 
- Install "Visual C++ 2013 Redistributable x64"

@SDRPlay API v2.10 and before only depended on the system component `msvcrt.dll` either because you compiled with MSYS2/Cygwin or you linked the C++ runtime statically with Visual Studio (`/MT`)
Why changing to a  `Multithreaded DLL /MD` binary ?

SDRplay, 2018-03-04:
@vsonnier yep you're right the dependencies are different. I'll need to look into that. What's odd is that the 2013 C++ redist is included with the API installer, so why is @righthalfplane having a problem?

SDRuno isn't a good test in this experiment as it doesn't use or install the API, SDR-Console V2.3 won't use 2.11 either. So I would suggest getting it from here: https://www.sdrplay.com/windl.php just to be sure.

vsonnier, 2018-03-04:
@SDRplay You may have provided the 32 bit redist one, not the 64 bit one.

SDRplay, 2018-03-04:
Filename: "{tmp}\vcredist_x64.exe"; Parameters: "/passive /norestart"; Flags: waituntilterminated; Check: IsWin64
Filename: "{tmp}\vcredist_x86.exe"; Parameters: "/passive /norestart"; Flags: waituntilterminated

Source: "VS2013_Redist\vcredist_x64.exe"; DestDir: "{tmp}"; Flags: ignoreversion deleteafterinstall; Check: IsWin64
Source: "VS2013_Redist\vcredist_x86.exe"; DestDir: "{tmp}"; Flags: ignoreversion deleteafterinstall

yep, that's the installer

vsonnier, 2018-03-04:
@SDRplay @righthalfplane I bet on the multiple overwrites of the API done by installing the SDRConsole and/or HDSDR on top of a SDRUno install then.

SDRplay, 2018-03-04:
I wonder, just looking at that, it would install both on a 64bit machine - hmmm

SDRplay, 2018-03-04:
Any download from our website (except SDRuno) will install the API - if you install SDR-Console V2.3 AFTER anything else, then yes, you'll get the old API 2.10. All other installers will install the latest API 2.11

righthalfplane, 2018-03-04:
Yes, mine is a 64 bit machine and system. I finally got everything running. I first made sure that all of the Windows Updates were up to date. I then tried the various Cubic versions, but still only the 0.2.0 version worked. I then uninstall everything. I installed SDRplay_SDRConsole_Installer_2.3_b2760_2 and then PothosSDR-2018.02.05-vc14-x64. Everything I tried in PothosSDR now works CubicSDR, gqrx and SoapySDRUtil. I then installed CubicSDR v0.2.3 x64 and it worked without the mir_sdr_api.dll in its folder. I then installed CubicSDR v0.2.3 x32 and it did not work. I uninstall SDR Console 2.3 and installed SDR Console version 3. It worked and everything else but CubicSDR v0.2.3 x32. I then copied the  32 bit mir_sdr_api.dll into the CubicSDR v0.2.3 x32 folder and it now works. The Visual C++ 2013 and Visual C++ 2015 Redistributables got installed several times along the way.

vsonnier, 2018-03-05:
Closing this issue now, the problem was not related to CubicSDR but to various installation problems with external libraries.

closed 2018-03-05 by vsonnier