cjcliffe/CubicSDR #623: CubicSDR 0.2.3 crashes on strart (MAC)

bnzf, 2018-02-27:
I loaded the binary version of CubicSDR 0.2.3 for Mac from the SDRPlay website.
On start I see for an instant the interface window and then crash with report:
```

Process:         CubicSDR [9055]
Path:            /Volumes/VOLUME/*/CubicSDR.app/Contents/MacOS/CubicSDR
Identifier:      com.cubicproductions.cubicsdr
Version:         0.2.3 (0.2.3)
Code Type:       X86-64 (Native)
Parent Process:  launchd [242]
Responsible:     CubicSDR [9055]
User ID:         502

Date/Time:       2018-02-27 20:43:12.384 +0100
OS Version:      Mac OS X 10.9.5 (13F1911)
Report Version:  11
Anonymous UUID:  5363C294-CE8C-13CC-33D5-3B63BD266E49

Sleep/Wake UUID: ECA3BE68-BA9A-4EEE-85AB-5C9561E5CE13

Crashed Thread:  14

Exception Type:  EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000002, 0x0000000000000000

Dyld Error Message:
  Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /usr/local/lib/libmirsdrapi-rsp.so.2.11
  Expected in: /usr/lib/libSystem.B.dylib

Thread 0:: Dispatch queue: com.apple.main-thread
0   GLEngine                      	0x00007fff85824cee gleUpdateDeferredStateFull + 3816
1   GLEngine                      	0x00007fff858271a6 gleDoDrawDispatchCore + 252
2   GLEngine                      	0x00007fff857df919 glDrawArrays_IMM_Exec + 225
3   com.cubicproductions.cubicsdr 	0x0000000106286375 GLPanel::draw() + 165
4   com.cubicproductions.cubicsdr 	0x00000001062711d8 SpectrumCanvas::OnPaint(wxPaintEvent&) + 696
5   com.cubicproductions.cubicsdr 	0x00000001063c552f wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 431
6   com.cubicproductions.cubicsdr 	0x00000001063c6c7a wxEvtHandler::ProcessEventLocally(wxEvent&) + 90
7   com.cubicproductions.cubicsdr 	0x00000001063c6b77 wxEvtHandler::ProcessEvent(wxEvent&) + 183
8   com.cubicproductions.cubicsdr 	0x00000001063c71d0 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 16
9   com.cubicproductions.cubicsdr 	0x00000001063e189e wxWindow::MacDoRedraw(long) + 910
10  com.cubicproductions.cubicsdr 	0x0000000106464607 wxWidgetCocoaImpl::drawRect(void*, NSView*, void*) + 791
11  com.cubicproductions.cubicsdr 	0x0000000106463256 wxOSX_drawRect(NSView*, objc_selector*, CGRect) + 86
12  com.apple.AppKit              	0x00007fff85f1704f -[NSView _drawRect:clip:] + 3748
13  com.apple.AppKit              	0x00007fff85f158c4 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 1799
14  com.apple.AppKit              	0x00007fff85f13706 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 841
15  com.apple.AppKit              	0x00007fff85f14bc4 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 6151
16  com.apple.AppKit              	0x00007fff85f14bc4 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 6151
17  com.apple.AppKit              	0x00007fff85f14bc4 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 6151
18  com.apple.AppKit              	0x00007fff85f14bc4 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 6151
19  com.apple.AppKit              	0x00007fff85f12eb1 -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 314
20  com.apple.AppKit              	0x00007fff85f0fe9f -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 2828
21  com.apple.AppKit              	0x00007fff85eef2da -[NSView displayIfNeeded] + 1680
22  com.apple.AppKit              	0x00007fff85f5474e _handleWindowNeedsDisplayOrLayoutOrUpdateConstraints + 884
23  com.apple.AppKit              	0x00007fff8652a061 __83-[NSWindow _postWindowNeedsDisplayOrLayoutOrUpdateConstraintsUnlessPostingDisabled]_block_invoke1331 + 46
24  com.apple.CoreFoundation      	0x00007fff876e5d67 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
25  com.apple.CoreFoundation      	0x00007fff876e5cd7 __CFRunLoopDoObservers + 391
26  com.apple.CoreFoundation      	0x00007fff876d73b8 __CFRunLoopRun + 776
27  com.apple.CoreFoundation      	0x00007fff876d6e75 CFRunLoopRunSpecific + 309
28  com.apple.HIToolbox           	0x00007fff8c10da0d RunCurrentEventLoopInMode + 226
29  com.apple.HIToolbox           	0x00007fff8c10d685 ReceiveNextEventCommon + 173
30  com.apple.HIToolbox           	0x00007fff8c10d5bc _BlockUntilNextEventMatchingListInModeWithFilter + 65
31  com.apple.AppKit              	0x00007fff85db824e _DPSNextEvent + 1434
32  com.apple.AppKit              	0x00007fff85db789b -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 122
33  com.apple.AppKit              	0x00007fff85dab99c -[NSApplication run] + 553
34  com.cubicproductions.cubicsdr 	0x000000010644a66e wxGUIEventLoop::OSXDoRun() + 110
35  com.cubicproductions.cubicsdr 	0x000000010639fe24 wxCFEventLoop::DoRun() + 52
36  com.cubicproductions.cubicsdr 	0x00000001063180a1 wxEventLoopBase::Run() + 161
37  com.cubicproductions.cubicsdr 	0x00000001062e45ca wxAppConsoleBase::MainLoop() + 202
38  com.cubicproductions.cubicsdr 	0x00000001064121ca wxApp::OnRun() + 26
[... 392 lines cut ...]
CG backing stores                  18.0M
CG image                              4K
CG raster data                       96K
CG shared images                    188K
IOKit                              73.5M
IOKit (reserved)                      4K        reserved VM address space (unallocated)
Kernel Alloc Once                     8K
MALLOC                            165.9M
MALLOC (admin)                       32K
Memory Tag 242                       12K
STACK GUARD                        56.1M
Stack                              15.7M
VM_ALLOCATE                        77.4M
__DATA                             69.9M
__IMAGE                             528K
__LINKEDIT                         77.0M
__TEXT                            135.2M
__UNICODE                           544K
mapped file                        63.1M
shared memory                        68K
===========                      =======
TOTAL                             753.2M
TOTAL, minus reserved VM space    753.1M

```

vsonnier, 2018-02-28:
@bnzf Hello ! I changed the title to hightlight the OSX version, because it definitely works for other versions out there.

bnzf, 2018-02-28:
Hello. I had installed the driver before installing CubicSDR. I downloaded both yesterday from sdrplay.

bnzf, 2018-02-28:
I tried two other Mac, one with OS 10.13 there it runs fine. And one with OS 10.6.8 where it also crashes:
```

Process:         launchd [284]
Path:            /Applications/CubicSDR.app/Contents/MacOS/CubicSDR
Identifier:      com.cubicproductions.cubicsdr
Version:         ??? (???)
Code Type:       X86-64 (Native)
Parent Process:  launchd [176]

Date/Time:       2018-02-28 15:22:20.522 +0100
OS Version:      Mac OS X 10.6.8 (10K549)
Report Version:  6

Interval Since Last Report:          146582 sec
Crashes Since Last Report:           13
Per-App Crashes Since Last Report:   1
Anonymous UUID:                      7632B892-0CB7-4286-9BFA-FD563961D0D9

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0xfffffffffffffff8
Crashed Thread:  Unknown

Backtrace not available

Unknown thread crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000055  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x0000000000000000
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x00007fff5fc01028  rfl: 0x0000000000010203  cr2: 0xfffffffffffffff8

Binary images description not available

Model: MacPro3,1, BootROM MP31.006C.B05, 8 processors, Quad-Core Intel Xeon, 2.8 GHz, 18 GB, SMC 1.25f4
Graphics: ATI Radeon HD 5770, ATI Radeon HD 5770, PCIe, 1024 MB
Memory Module: global_name
Bluetooth: Version 2.4.5f3, 2 service, 12 devices, 1 incoming serial ports
Network Service: Ethernet 2, Ethernet, en1
PCI Card: ATI Radeon HD 5770, sppci_displaycontroller, Slot-1
PCI Card: ATI Radeon HD 5770, ATY,HoolockParent, Slot-1
Serial ATA Device: SAMSUNG HD642JJ, 596,17 GB
Serial ATA Device: Maxtor 6Y250M0, 233,76 GB
Serial ATA Device: WDC WD6400AAKS-41H2B0, 596,17 GB
Serial ATA Device: SAMSUNG HD103UI, 931,51 GB
Serial ATA Device: OCZ-VERTEX2, 55,9 GB
Serial ATA Device: OCZ-VERTEX2, 55,9 GB
Parallel ATA Device: OPTIARC DVD RW AD-7170A
USB Device: USB3.0 External M.2 SSD, 0x174c  (ASMedia Technology Inc.), 0x55aa, 0xfd100000 / 7
USB Device: USB2.0 Hub, 0x05e3  (Genesys Logic, Inc.), 0x0608, 0xfd300000 / 2
USB Device: HA2-A3, 0x10d5  (Uni Class Technology Co., Ltd.), 0x000d, 0xfd330000 / 4
USB Device: Hub in Apple Pro Keyboard, 0x05ac  (Apple Inc.), 0x1003, 0xfd310000 / 3
USB Device: Apple Optical USB Mouse, 0x05ac  (Apple Inc.), 0x0304, 0xfd311000 / 6
USB Device: Apple Pro Keyboard, 0x05ac  (Apple Inc.), 0x020c, 0xfd313000 / 5
USB Device: Bluetooth USB Host Controller, 0x05ac  (Apple Inc.), 0x8206, 0x5d200000 / 2
FireWire Device: built-in_hub, Up to 800 Mb/sec

```

vsonnier, 2018-02-28:
@bnzf The error 
````
Symbol not found: _OSAtomicIncrement32Barrier
Referenced from: /usr/local/lib/libmirsdrapi-rsp.so.2.11
````
suggests that the SDRPlay API driver needs a service that is not available to your version. It likely means your OSX version is too old to run the driver. 
The second error you had on Mac OS 10.6.8 suggests a similar problem.
If you want a confirmation on this issue, you have to ask help from @SDRplay  SDRPlay Support.

SDRplay, 2018-02-28:
That's not in the API specifically, it's most likely in libusb - you may need a later version - try 1.0.19 as 1.0.20 and 1.0.21 have known issues as far as the API is concerned. You can get libusb from http://libusb.info

bnzf, 2018-02-28:
If I try to build (./configure, make) libusb 1.0.19 I get an error
```

bookii:libusb-1.0.19 bnzf$ make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in libusb
  CC       libusb_1_0_la-core.lo
  CC       libusb_1_0_la-descriptor.lo
  CC       libusb_1_0_la-io.lo
  CC       libusb_1_0_la-strerror.lo
  CC       libusb_1_0_la-sync.lo
  CC       libusb_1_0_la-hotplug.lo
  CC       os/libusb_1_0_la-threads_posix.lo
  CC       os/libusb_1_0_la-darwin_usb.lo
os/darwin_usb.c:1281:27: warning: expression which evaluates to zero treated as
      a null pointer constant of type 'IOUSBInterfaceInterface550 **' (aka
      'struct IOUSBInterfaceStruct550 **') [-Wnon-literal-null-conversion]
  cInterface->interface = IO_OBJECT_NULL;
                          ^~~~~~~~~~~~~~
/System/Library/Frameworks/IOKit.framework/Headers/IOTypes.h:164:24: note: 
      expanded from macro 'IO_OBJECT_NULL'
#define IO_OBJECT_NULL  ((io_object_t) 0)
                        ^~~~~~~~~~~~~~~~~
1 warning generated.
  CC       os/libusb_1_0_la-poll_posix.lo
  CCLD     libusb-1.0.la
Making all in doc
make[2]: Nothing to be done for `all'.
make[2]: Nothing to be done for `all-am'.
bookii:libusb-1.0.19 bnzf$ 

```

SDRplay, 2018-02-28:
I upgraded Snow Leopard to Yosemite and didn't have any problems after that. Unfortunately I can't say why 1.0.19 won't build for you. You could try other builds. The issue with 1.0.20 is that I couldn't get it to work with sample rates > 3 MHz - I haven't yet had a chance to evaluate 1.0.21 and there is a release candidate for 1.0.22 if you build from github.

bnzf, 2018-02-28:
I installed libusb 1.0.21 (./bootstrap.sh ./configure, make, make install, reboot),  CubicSDR is still crashing.
I perhaps should mention that the SDRlay is not connected (I am waiting for delivery). I just wanted to explore the software (which seems to be no issue with os 10.13).

```
Process:         CubicSDR [387]
Path:            /Volumes/VOLUME/*/CubicSDR.app/Contents/MacOS/CubicSDR
Identifier:      com.cubicproductions.cubicsdr
Version:         0.2.3 (0.2.3)
Code Type:       X86-64 (Native)
Parent Process:  launchd [235]
Responsible:     CubicSDR [387]
User ID:         502

Date/Time:       2018-02-28 21:33:14.436 +0100
OS Version:      Mac OS X 10.9.5 (13F1911)
Report Version:  11
Anonymous UUID:  5363C294-CE8C-13CC-33D5-3B63BD266E49

Crashed Thread:  14

Exception Type:  EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000002, 0x0000000000000000

Dyld Error Message:
  Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /usr/local/lib/libmirsdrapi-rsp.so.2.11
  Expected in: /usr/lib/libSystem.B.dylib

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff940dca3a __semwait_signal + 10
1   libsystem_c.dylib             	0x00007fff885f4dcc nanosleep + 200
2   libc++.1.dylib                	0x00007fff88497b38 std::__1::this_thread::sleep_for(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> > const&) + 75
3   com.cubicproductions.cubicsdr 	0x00000001086e5116 AppFrame::OnIdle(wxIdleEvent&) + 7750
4   com.cubicproductions.cubicsdr 	0x00000001088d252f wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 431
5   com.cubicproductions.cubicsdr 	0x00000001088d3c7a wxEvtHandler::ProcessEventLocally(wxEvent&) + 90
6   com.cubicproductions.cubicsdr 	0x00000001088d3b77 wxEvtHandler::ProcessEvent(wxEvent&) + 183
7   com.cubicproductions.cubicsdr 	0x00000001088d41d0 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 16
8   com.cubicproductions.cubicsdr 	0x0000000108a3e828 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 72
9   com.cubicproductions.cubicsdr 	0x000000010897b246 wxAppBase::ProcessIdle() + 134
10  com.cubicproductions.cubicsdr 	0x000000010891f17a wxApp::ProcessIdle() + 26
11  com.cubicproductions.cubicsdr 	0x00000001088251dc wxEventLoopBase::ProcessIdle() + 28
12  com.cubicproductions.cubicsdr 	0x00000001088ac5dc wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 92
13  com.apple.CoreFoundation      	0x00007fff8b551d67 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
14  com.apple.CoreFoundation      	0x00007fff8b551cd7 __CFRunLoopDoObservers + 391
15  com.apple.CoreFoundation      	0x00007fff8b543493 __CFRunLoopRun + 995
16  com.apple.CoreFoundation      	0x00007fff8b542e75 CFRunLoopRunSpecific + 309
17  com.apple.HIToolbox           	0x00007fff8ff79a0d RunCurrentEventLoopInMode + 226
18  com.apple.HIToolbox           	0x00007fff8ff797b7 ReceiveNextEventCommon + 479
19  com.apple.HIToolbox           	0x00007fff8ff795bc _BlockUntilNextEventMatchingListInModeWithFilter + 65
20  com.apple.AppKit              	0x00007fff89c2424e _DPSNextEvent + 1434
21  com.apple.AppKit              	0x00007fff89c2389b -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 122
22  com.apple.AppKit              	0x00007fff89c1799c -[NSApplication run] + 553
23  com.cubicproductions.cubicsdr 	0x000000010895766e wxGUIEventLoop::OSXDoRun() + 110
24  com.cubicproductions.cubicsdr 	0x00000001088ace24 wxCFEventLoop::DoRun() + 52
25  com.cubicproductions.cubicsdr 	0x00000001088250a1 wxEventLoopBase::Run() + 161
26  com.cubicproductions.cubicsdr 	0x00000001087f15ca wxAppConsoleBase::MainLoop() + 202
27  com.cubicproductions.cubicsdr 	0x000000010891f1ca wxApp::OnRun() + 26
28  com.cubicproductions.cubicsdr 	0x000000010885a98e wxEntry(int&, wchar_t**) + 62
29  com.cubicproductions.cubicsdr 	0x00000001086d9a70 main + 48
30  libdyld.dylib                 	0x00007fff8a8c65fd start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff940dce6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8ac2af08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff8ac2dfb9 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff940dd662 kevent64 + 10
1   libdispatch.dylib             	0x00007fff87ac5421 _dispatch_mgr_invoke + 239
2   libdispatch.dylib             	0x00007fff87ac5136 _dispatch_mgr_thread + 52
[... 397 lines cut ...]
mapped file                        63.1M
shared memory                        68K
===========                      =======
TOTAL                             771.1M
TOTAL, minus reserved VM space    771.1M
 

Model: MacBookPro11,3, BootROM MBP112.0142.B00, 4 processors, Intel Core i7, 2.3 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x02FE, -
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x02FE, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (6.30.223.154.65)
Bluetooth: Version 5.2.7f4 15858, 3 services, 15 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en4
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: Kingston SHPM2280P2/960G, 960,2 GB
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 1, 5.5

```

bnzf, 2018-03-01:
CubicSDR 0.2.0 works fine with OS 10.9.5. I only see no posibility to acsess the SDRPlay, which meanwhile has arrived.

bnzf, 2018-03-01:
I installed CubicSDR 0.2.3 without the SDRPlay driver  on an other Mac with OS 10.9.5 and it runs fine. So it is indeed related to the driver

closed 2019-04-11 by cjcliffe