cjcliffe/CubicSDR #624: Offset on/off switching

WaltN3MTJ, 2018-03-01:
It would be nice being able to turn the offset on and off with a single click once the offset is entered (unless I'm missing something).

I didn't see this enhancement request as far back as 2015, I hope I haven't missed an earlier request.

vsonnier, 2018-03-02:
@WaltN3MTJ What do you mean by offset ?

WaltN3MTJ, 2018-03-02:
Sorry, **Settings|Frequency Offset** where you can correct tuning for an upconverter, etc.

cjcliffe, 2018-03-14:
@WaltN3MTJ Seems reasonable but I'm not sure where we'd put that in the current UI setup and make it consistent with other elements; a keyboard or 'disable offset' menu toggle that isn't saved might be a good interim solution.

WaltN3MTJ, 2018-03-14:
@cjcliffe That was what I was thinking. A thought would eventually be the keyboard shortcut bringing up the frequency offset menu with a simple checkbox in addition to the actual frequency offset entry.