cjcliffe/CubicSDR #625: Spyserver

ifuchs, 2018-03-02:
Is there currently a way to use CubicSDR with a Spyserver? Would like to run a RTL-SDR on RPi and use CubicSDR remotely.

guruofquality, 2018-03-02:
I would recommend to install SoapyRemote: https://github.com/pothosware/SoapyRemote/wiki

But in regards to the spyserver protocol specifically, any volunteers to make a SoapySpyClient or SoapySpyServer (I guess in this case client)? more info https://github.com/pothosware/SoapySDR/issues/156 :-)

ifuchs, 2018-03-02:
Thanks for the quick reply. I am somewhat new to SDR. So functionally, SoapyRemote is to CubicSDR and Gqrx what Spyserver is to SDR#?

mmgarland3, 2018-03-02:
Yes, except the SoapySDR library is open and supports dozens of SDRs, and
numerous applications use the SoapySDR client driver.

SDR# and Spyserver are closed implementation owned by Airspy and actively
lock-out support for other products.

closed 2018-03-11 by vsonnier