cjcliffe/CubicSDR #627: Keyboard/Mouse Improvements (March 2018)

a-raccoon, 2018-03-06:
* **Numpad Change Frequency**
Allow numeric keypad (numpad) frequency inputs. Currently only top-row numbers can change frequency. (For the longest time I thought that keyboard frequency entry was broken since I always use the numpad).

* **Push-and-Hold Frequency increase/decrease scrolling**
Allow holding down mouse on frequency digit up/down to continue rolling digit after moment delay.  Repeat rate of 0.1s or 10 digits per second should suffice. (I wrote a rapid-fire mouse macro to simulate this behavior and it's a wonderful interface enhancement!)

* **Zero-out Frequency Digits to the right**
Some sort of keyboard, mouse, or keyboard-mouse combo to zero out frequency digits on-and-to-the-right of the currently hovered or clicked frequency digit position.  For example, Alt+LeftClick or MiddleClick or the number zero 0.

* **DEL key delete bookmarks / active frequencies**
When the Bookmarks panel has keyboard focus, the Delete key should delete the selected item from Active, Bookmarks, Recents, etc.  This panel could use more keyboard love in general (plus a remove-all button for parent groups).

cjcliffe, 2018-03-14:
@a-raccoon thanks for the suggestions.

- I thought Numpad inputs were working but we will have to investigate.

- Push and hold frequency increase/decrease is already kinda implemented; though you have to click and drag the digit LEFT/RIGHT instead of holding UP/DOWN and dragging further increases the speed.   I'm not sure I could logically keep the left/right and add the up/down as well without causing conflicts but it was an early feature from before individual digits were clickable that's been carried forward -- so it might be up for review.

- Zeroing the digits to the right might be useful; I think we should experiment with that.

- DEL key on the bookmarks might be a good option; I think i would still have the pop-up dialog to confirm groups/bookmarks though.   Note that there's already a "Remove Group" button if you select the group that will remove the group and all the bookmarks within it.

a-raccoon, 2018-03-15:
Numpad inputs would probably have to use `case 'WXK_NUMPAD0':` to `case 'WXK_NUMPAD9':` per reference [wxpython](https://wxpython.org/Phoenix/docs/html/wx.KeyCode.enumeration.html) at this code block, and handle the string-to-num conversion by grabbing the right-most string character.

https://github.com/cjcliffe/CubicSDR/blob/7fb66b69981a5b81734616982dfe44735e3813f5/src/AppFrame.cpp#L2766-L2778