cjcliffe/CubicSDR #628: Improve Snap Frequency Feature

a-raccoon, 2018-03-06:
Currently: You can right-click to set a Snap digit frequency stepping resolution for a Tuned Frequency, only.

Request:  Allow setting Snap digits, as well as user-defined Snap resolutions, for the Tuned Frequency, as well as for the Center Frequency and the Bandwidth size.  Interface-wise, you would be able to use right-click to set a stepping resolution at that digit marker, and that digit would be outlined in red.  Additionally, you can specify an arbitrary value for a stepping resolution which will outline multiple digits in red which correspond with the arbitrary value's size.  An arbitrary value might be something like "12500" or "5kHz" as typed by the user.

Allow each of the MODES to have its own unique Bandwidth snap / stepping resolution.

Feature creep:  Allow for creating custom user-defined modes with their own Bandwidth size and snapping.  (IE: WFM vs NFM).

cjcliffe, 2018-03-14:
@a-raccoon this seems like a good enhancement to the snapping options; thanks for the suggestions.

Note: for when you bump the bandwidth/frequency for now -- you can hover any digits and press SPACE (or just start typing numbers) to directly set the value.  Also helpful is holding SHIFT while adjusting the digits prevents adjacent ones from 'carrying' (so 0->9 and 9->0 doesn't change other digits)

a-raccoon, 2018-03-14:
I just discovered that SDR# does is zero-to-the-right on a digit that is right-clicked.  I just wanted to add that to this thread of knowledge.