cjcliffe/CubicSDR #629: Bug: Delta Lock doesn't work with [, and ], keyboard scrolling.

a-raccoon, 2018-03-06:
Toggling on the Delta Lock (V) prevents usage of ([) and (]) keyboard frequency up/down scrolling.  The frequency still increases and decreases, but the center frequency does not move with it, creating a display desynchronization and confusion.  Up/down keyboard arrows still work, but do not respect the snapping frequency resolution.

cjcliffe, 2018-03-14:
@a-raccoon thanks for the report; this does appear to be a problem using '[' and ']' with some quick testing here.   Making it respect the frequency snap might be more logically difficult than it seems but we can investigate.

vsonnier, 2018-03-14:
@a-raccoon @cjcliffe I should add that on my French AZERTY keyboard [ ] are on the same place as 5 and °. Pressing the [ indeed trigger the direct numeric frequency entry. 
I think we should move the direct frequency entry on the numeric keypad to prevent this kind of issue.

cjcliffe, 2018-03-14:
@vsonnier hmm, that might work for full keyboards but my Macbook here and iMac at work don't have a numeric keypad :)   I suppose space still works to activate though.

There's likely a better way to handle those keys than using the raw keycodes in the global handler.