cjcliffe/CubicSDR #631: Possible using the wifi receiver?

opabert, 2018-03-10:
For RPi-3 we have CUBICsdr using the RTLSDR they go up to 1700MHz max. 
And just thinking loud,
RPi-3 having a build in WiFi receiver that be possible CUBICsdr use these receiver for extended frequency range to cover from 2000 - 2500 MHz or 5000 - 6000 MHz 
Is this be possible?

parkerlreed, 2018-03-11:
No.

vsonnier, 2018-03-11:
@parkerlreed Has pretty settled the matter, in quite a concise way. More precisely, you would need to develop a SoapySDR module driver able to control the WiFi part in such a way to generate I/Q data, and being able to adjust the frequency at will which would mean a custom firmware. And even then, the WiFi hardware may be adapted to certain frequency ranges.
Well, No.

closed 2018-03-11 by vsonnier

vsonnier, 2018-03-16:
@opabert @parkerlreed FYI, here is something similar to this that has just pop-up on www.rtl-sdr.com : 

[NEXMON SDR: TURNING A BROADCOM 802.11AC WIFI CHIP INTO A TX CAPABLE SOFTWARE DEFINED RADIO](https://www.rtl-sdr.com/nexmon-sdr-turning-a-broadcom-802-11ac-wifi-chip-into-a-tx-capable-software-defined-radio/)

opabert, 2018-03-16:
sorry close the topic, because I know this whas coming and hoped inspire developers  :)

parkerlreed, 2018-03-16:
At that point it's only transmit which CubicSDR doesn't do to begin with. And even if RX was implemented, you'd still be limited to the 2.4 - 5GHz range. What else other than WiFi/WiFi-direct would you be able to pick up?

vsonnier, 2018-03-16:
@parkerlreed @opabert I was merely mentioning this in passing, but the above arguments still stands, and nobody here is going to waste time implementing drivers a such particular, closed-source hardware to finally have a less than capable SDR on a smartphone.

The fact that WiFi / Bluethooth chips are SDR are no news, and the "discovery" to make it generate I/Q data is exactly the same story as the I/Q reverse engineering of the DVB-T capable RTL-SDR dongles back in the day. 
Then it took years to have a complete driver and supporting SDR software. 
The difference with this particular Broadcom chip is that the RTL-SDR hardware has almost not changed from that day, and is cheap as dirt.
On the other hand this Broadcom chip will likely be obsolete fast given the smartphone market and disappear, so it would be a waste of time to support it for SDR software.

To sum up: by a 200 bucks phone instead of a 300 one (it does the same thing anyway) and buy a SDRplay RSP1A with the difference :)

opabert, 2018-03-16:
First quest,  are you both hardware engineers or have a Hamradio license or build or repair yourself any hardware receiver hardware?
Like to remember, the RTL from origin used as DVB-T receiver and the 820T2 receiver chip be obsolete in summer 2018.  There will be fallow up new hardware who have dual band receivers chips to cover wider extended range (like used in video receivers who cover 2000 - 5000).
Frequency monitoring 2000MHz to 5000MHz there is plenty more  than only Wi-Fi or Bluetooth 
But back to the point,  look the Broadcom datasheet so lets wait what future new development bring.
73's

ve1jot, 2018-03-17:
We have ham bands in and near both 2.4 and 5ghz...

opabert, 2018-03-17:
Not only hamradio but many other users on these frequencies, Software makers developing software and hamradio know all about the freq spectrum and there users so lets put our ideas together and move forward to find new possibilities ;)