cjcliffe/CubicSDR #632: Feature: Select Audio Device

a-raccoon, 2018-03-12:
CubicSDR lets you specify the Audio Sample Rate for each of the Playback Audio Devices (output) your system supports, but it does not let you actually specify which Playback Audio Device you wish to use.

Suggestion: Let the user specify which audio device to use.

Currently: CubicSDR only uses whichever device you specify as the System Default Playback Device.  This isn't very useful.  In the Windows RTL-SDR world, it's necessary to pipe audio from one program to another program for things like digital decoding (ie, DSD+ for D-STAR, P25, DMR, etc).

vsonnier, 2018-03-12:
@a-raccoon What you are looking for is already there:
[image]

You can choose here an output device for a particular modem, and make another choice for another modem. This choice is saved in a session file per-modem and is properly restored at session reload.

closed 2018-03-12 by vsonnier

a-raccoon, 2018-03-12:
Thank you.  I searched the documentation did not find this with keyword `device`.  But now I see it under `audio out`.  http://cubicsdr.readthedocs.io/en/latest/application-window.html?highlight=audio

dslotter, 2019-11-01:
I see where I should be able to select audio devices under the Modem, but as you can see in this attached screen capture, I have four audio out devices, but only "PulseAudio" shows up. This is under Ubuntu Linux 16.04. Does CubicSDR v0.2.5 under Linux not have the capability to send to a particular audio output device? I read the documentation and couldn't find any help on this. Please advise, and thanks.

[image]

adeptex, 2019-12-23:
You need to install rtaudio from http://github.com/thestk/rtaudio/