cjcliffe/CubicSDR #634: CubicSDR 0.2.3 crashes on strart with USRP B210

leeguang517, 2018-03-17:
OS: gentoo linux

crash stack:

`Register dump:

 RAX: 00005636d7138a20   RBX: 00005636d7137a40   RCX: 0000000000000000
 RDX: 0000000000000141   RSI: 0000000000000050   RDI: 00005636d7138a20
 RBP: 0000000000000141   R8 : 00000000000003d0   R9 : 00005636d71255e0
 R10: 0000000000000000   R11: 00007fab9e3027a0   R12: 00005636d7138a20
 R13: 00005636d6de7630   R14: 00005636d7137a50   R15: 0000000000000003
 RSP: 00007ffe7cfdbf58

 RIP: 00007fab9e2c13d1   EFLAGS: 00010202

 CS: 0033   FS: 0000   GS: 0000

 Trap: 0000000e   Error: 00000004   OldMask: 00000000   CR2: 00000050

 FPUCW: 0000037f   FPUSW: 00000420   TAG: 00005636
 RIP: d2617ffe   RDP: 7cfdc680

 ST(0) 0000 0000000000000000   ST(1) 0000 0000000000000000
 ST(2) ffff 8000000000000000   ST(3) ffff 8000000000000000
 ST(4) ffff a000000000000000   ST(5) 0000 0000000000000000
 ST(6) ffff 8000000000000000   ST(7) 8000 8000000000000000
 mxcsr: 1fa4
 XMM0:  00000000000000000000000000000000 XMM1:  00000000000000000000000000000000
 XMM2:  00000000000000000000000000000000 XMM3:  00000000000000000000000000000000
 XMM4:  00000000000000000000000000000000 XMM5:  00000000000000000000000000000000
 XMM6:  00000000000000000000000000000000 XMM7:  00000000000000000000000000000000
 XMM8:  00000000000000000000000000000000 XMM9:  00000000000000000000000000000000
 XMM10: 00000000000000000000000000000000 XMM11: 00000000000000000000000000000000
 XMM12: 00000000000000000000000000000000 XMM13: 00000000000000000000000000000000
 XMM14: 00000000000000000000000000000000 XMM15: 00000000000000000000000000000000

Backtrace:
/lib64/libc.so.6(+0x13e3d1)[0x7fab9e2c13d1]
/usr/lib/gcc/x86_64-pc-linux-gnu/6.4.0/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_+0xcb)[0x7fab9edbefdb]
CubicSDR(+0x14d953)[0x5636d2632953]
/usr/lib64/libwx_baseu-3.0.so.0(_ZNK16wxAppConsoleBase16CallEventHandlerEP12wxEvtHandlerR14wxEventFunctorR7wxEvent+0x3e)[0x7fab9fbdd3be]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent+0x55)[0x7fab9fd5fe95]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent+0x6d)[0x7fab9fd6028d]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler11TryHereOnlyER7wxEvent+0x1f)[0x7fab9fd6030f]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler19ProcessEventLocallyER7wxEvent+0x33)[0x7fab9fd603c3]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler12ProcessEventER7wxEvent+0x45)[0x7fab9fd60425]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler18SafelyProcessEventER7wxEvent+0x7)[0x7fab9fd60197]
/usr/lib64/libwx_gtk2u_core-3.0.so.0(+0x2a2e33)[0x7faba0594e33]
/usr/lib64/libgtk-x11-2.0.so.0(+0x13050c)[0x7fab9d77b50c]
/usr/lib64/libgobject-2.0.so.0(g_closure_invoke+0x145)[0x7fab9d152f15]
/usr/lib64/libgobject-2.0.so.0(+0x21f47)[0x7fab9d164f47]
/usr/lib64/libgobject-2.0.so.0(g_signal_emit_valist+0x903)[0x7fab9d16d6b3]
/usr/lib64/libgobject-2.0.so.0(g_signal_emit+0x97)[0x7fab9d16dff7]
/usr/lib64/libgtk-x11-2.0.so.0(+0x2486ac)[0x7fab9d8936ac]
/usr/lib64/libgtk-x11-2.0.so.0(gtk_propagate_event+0xc4)[0x7fab9d779cd4]
/usr/lib64/libgtk-x11-2.0.so.0(gtk_main_do_event+0x2cb)[0x7fab9d77a08b]
/usr/lib64/libgdk-x11-2.0.so.0(+0x5b36c)[0x7fab9d3f136c]
/usr/lib64/libglib-2.0.so.0(g_main_context_dispatch+0x2a7)[0x7fab9c196517]
/usr/lib64/libglib-2.0.so.0(+0x4a780)[0x7fab9c196780]
/usr/lib64/libglib-2.0.so.0(g_main_loop_run+0xc2)[0x7fab9c196aa2]
/usr/lib64/libgtk-x11-2.0.so.0(gtk_main+0xb7)[0x7fab9d779107]
/usr/lib64/libwx_gtk2u_core-3.0.so.0(_ZN14wxGUIEventLoop5DoRunEv+0x25)[0x7faba05678b5]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN15wxEventLoopBase3RunEv+0x8b)[0x7fab9fc1de2b]
/usr/lib64/libwx_baseu-3.0.so.0(_ZN16wxAppConsoleBase8MainLoopEv+0x56)[0x7fab9fbe3c96]
/usr/lib64/libwx_baseu-3.0.so.0(_Z7wxEntryRiPPw+0x70)[0x7fab9fc6ef40]
CubicSDR(+0x5c7f8)[0x5636d25417f8]
/lib64/libc.so.6(__libc_start_main+0xf1)[0x7fab9e1a3541]
CubicSDR(_start+0x2a)[0x5636d254546a]

Memory map:
[... 709 lines cut ...]
7faba1386000-7faba13e6000 rw-s 00000000 00:05 1605642                    /SYSV00000000 (deleted)
7faba13e6000-7faba13ee000 rw-s 00000000 00:05 36647                      /drm mm object (deleted)
7faba13ee000-7faba13ef000 rw-s 00000000 00:05 36643                      /drm mm object (deleted)
7faba13ef000-7faba13f7000 rw-s 00000000 00:05 36640                      /drm mm object (deleted)
7faba13f7000-7faba13f8000 rw-s 00000000 00:05 36636                      /drm mm object (deleted)
7faba13f8000-7faba1400000 rw-s 00000000 00:05 36633                      /drm mm object (deleted)
7faba1400000-7faba1401000 rw-s 00000000 00:05 36629                      /drm mm object (deleted)
7faba1401000-7faba1409000 rw-s 00000000 00:05 36626                      /drm mm object (deleted)
7faba1409000-7faba1414000 r--s 00000000 08:13 395387                     /var/cache/fontconfig/acc285bc1956c3c4bc7afb41d537a85a-le64.cache-7
7faba1414000-7faba1417000 r--s 00000000 08:13 395398                     /var/cache/fontconfig/131ab5cc1583381c4f7ce0194912c56d-le64.cache-7
7faba1417000-7faba1442000 rw-p 00000000 00:00 0 
7faba1442000-7faba1454000 r--s 00000000 08:13 395397                     /var/cache/fontconfig/f2e43df3287529323e79e7929dc44cc9-le64.cache-7
7faba1454000-7faba145b000 r--s 00000000 08:13 395394                     /var/cache/fontconfig/ccb2d0786bada64fd4b4887a4949240d-le64.cache-7
7faba145b000-7faba1460000 r--s 00000000 08:13 395392                     /var/cache/fontconfig/a595ca43be94766bbfe429f07fba82d0-le64.cache-7
7faba1460000-7faba146b000 r--s 00000000 08:13 395386                     /var/cache/fontconfig/12b26b760a24f8b4feb03ad48a333a72-le64.cache-7
7faba146b000-7faba1472000 r--s 00000000 08:13 791711                     /usr/lib64/gconv/gconv-modules.cache
7faba1472000-7faba1474000 rw-p 00000000 00:00 0 
7faba1474000-7faba1475000 r--p 00022000 08:13 183256                     /lib64/ld-2.25.so
7faba1475000-7faba1476000 rw-p 00023000 08:13 183256                     /lib64/ld-2.25.so
7faba1476000-7faba1477000 rw-p 00000000 00:00 0 
7ffe7cfbe000-7ffe7cfdf000 rw-p 00000000 00:00 0                          [stack]
7ffe7cfeb000-7ffe7cfed000 r--p 00000000 00:00 0                          [vvar]
7ffe7cfed000-7ffe7cfef000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
`

vsonnier, 2018-03-17:
@leeguang517 Sorry, this is absolutely unhelpful. 
Please provide context (pre-built-binary ? pre-built-AppImage ? Compiled from master ? with which external libraries ? Compiling options ? ....etc. 

- Rant mode on:
I'm starting to get quite pissed by Linux users having, by definition, a different configuration than every other Linux user, dumping minimal info and expecting a solution. You made the choice of using Linux distribution, so try to investigate by yourself first, and participate as a knowledgeable developer would by providing the most complete bug report you can. This is free software, on our free time, even for us.
Linux is not a toy that can be handled by everyone, because "meh, Window$ 10", sorry. 
- Rant mode off.

Aaah, I feel better now. I'm still interested by more information on the matter, though.

leeguang517, 2018-03-19:
My CubicSDR is compiled from v0.2.3 source. with wxGTK-3.0.3.
I did some debugging and found that the crash happened here：
        `SDRDevicesDialog::OnUseSelected() at SDRDevices.cpp:381 0x55555575f6cb`
The source code here is like this：
        `streamArgs[arg.key] = arg.options[prop->GetChoiceSelection()];`
 The value of “prop->GetChoiceSelection()” is 2, but the array “arg.options”   has only two elements, so the program crashed.

vsonnier, 2018-03-21:
@leeguang517  Aahh, __THAT__ is a useful bug report, thank you :). Turns out it is the same as #534, but your explanation is much clearer on the matter.

vsonnier, 2018-03-21:
@leeguang517 I've committed a change in master to assure correct bound checking.

vsonnier, 2018-04-05:
@leeguang517 No news means good news, so I'm closing this issue now.

commit 4cd87350 (2018-03-21): Fix #634 plus #534: bound checking in SDRDevices dialog selection
closed 2018-04-05 by vsonnier