cjcliffe/CubicSDR #635: Issues with wxWidgets v3.11

vsonnier, 2018-03-20:
This issue gathers the various problems encountered during the migration to wxWidgets v3.11:
- [ ] @Magalex-2x14 : OS X 10.13.3, wxWidgets 3.1.1 - missing the labels on the buttons below the bookmarks. Same commit but with wxWidgets 3.1.0 - all OK.
- [ ] @parkerlreed : Syntax incompatible with wxWidgets v3.04

vsonnier, 2018-03-20:
Solutions  / Suggestions:
- @parkerlreed : The minimum requirement is wxWidgets v3.1, so you have to compile it from scratch, see the Wiki. The previous wxWidgets v3.0x syntax compatibility was pure luck, and had to end one day.
- @Magalex-2x14 : Same story here on Windows 10. The custom text button color given from the current theme doesn't work well. I see if I can fix it, or else leave the default color would do.

Magalex-2x14, 2018-03-20:
@vsonnier I also noticed that with 3.1.1 when new modem is added, corresponding buttons do not appear (bookmark, start recording, remove active). They appears only after "refocusing" -  I have to change focus to the bookmarks or ranges, then back to the modems.

parkerlreed, 2018-03-20:
(And by a long compile time it took nearly 8 minutes on a 4 core/10 thread machine running make at -j10. A bit lengthy and would only be longer for somebody running lower hardware)

vsonnier, 2018-03-20:
@parkerlreed It is quite long indeed, but you have only to do it every 2 years given the super-slow wxWidgets release schedule.

I've push 1 commit on the master branch, which reverts most of the `BookmarkView` modifications... The control refresh is really slow, but at least it is correct.

MicroJoe, 2018-03-21:
v3.1.x is shipped as binary release for Debian, Ubuntu and Fedora : http://codelite.org/LiteEditor/WxWidgets31Binaries

parkerlreed, 2018-03-21:
Ahh thanks. Wonder why all the official repos seem to be behind (I guess due to compat with older software)

vsonnier, 2018-03-24:
Hello @Magalex-2x14, @parkerlreed and all. I've commited some changes on the master, which I believe fixes the following points:
- Invisible button labels (Windows, OS-X). I forced them to white instead of using the actual theme color. 
- Slow `BookmarkView` refresh (Windows). Turns out the UI events were calling themselves back and forth.
- (Maybe ?) Improper control refresh (OS X). I've restored most of the old `BookmarkView` control logic, plus added other fixes.  

Please tell me how it goes for you.

Magalex-2x14, 2018-03-24:
OS X. Invisible button labels are still invisible (maybe labels must be forced to black? White labels becomes visible when the button is pressed...) In the rest everything seems OK.
[image]

vsonnier, 2018-03-24:
Thanks @Magalex-2x14. Funny thing, colors are exactly inverted on Windows 10:
[image]

So finally in the last commit I removed the buttons custom colors. Less sexy, but should be OK on all platforms.
What about your UI refreshing problems ?

vsonnier, 2018-03-24:
Great ! Thanks for your input, I think we can close this issue then.

commit b6b0f857 (2018-03-21): Fix #635 plus #534: bound checking in SDRDevices dialog selection
commit 00a864f7 (2018-03-24): Fix #635 Round 2 for Bookmarkview on Windows, should be OK now
commit a25253e7 (2018-03-24): Fix #635 round 3, BookmarkView: clear the buttons when 'Remove active'
commit c9d38ab3 (2018-03-24): Fixes #635, round 4, Bookmarkview : always have Start/Stop Recording button irrespective of the actual demod active state
closed 2018-03-24 by vsonnier