cjcliffe/CubicSDR #639: Cubic can't run on Raspian Stretch release

rrobinett, 2018-03-24:
Hi 
I'm trying to create a standalone WSPR receive station by attaching RTL-SDR and USB audio adapters to a RP3b and running CubicSDR feeding WSJT-x.  I set the USB adapter as my default audio device and have no problem installing the WSJT-x 1.8.0 package from the KiJT site and it recognizes the USB audio device.  
Since there is no CubicSDR package for the Raspberry Pi, I compiled the v0.2.3 sources.  But when I start Cubic I have two problems:
1) At start I get a dialog box with many "Couldn't create OpenGL context: messages and many sidebar controls are missing
2) When I select the rtl-sdr it starts decoding and displaying a meaningful waveform, but I am offered only "Pulse Audio" as the output and I hear nothing on the HDMI audio or the pi analog audio.    Using omxcplayer I can play audio through both the builtin pi analog audio and the USB audio adaptor, but the existence of the USB audio seems invisible to Cubic.
I've spent the day trolling and experimenting, so I've come here as a last resort.  Any pointers would be welcome.

vsonnier, 2018-03-25:
Hello again @rrobinett ! 
I assume this time you want to really run CubicSDR on the Pi. Beware though, it will be extremely slow, and you would rather run it in remote like your SDRPlay attempt.

> 1) At start I get a dialog box with many "Couldn't create OpenGL context: messages and many sidebar controls are missing

Well, you can stop dead here : CubicSDR is an OpenGL application, and you need to have it set on the Pi to run it. 
I can't help you here, but I'm sure your favorite search engine have plenty of answers.

For the building part, @loughkb [scripts](https://github.com/loughkb/SDRplaylinux) can be used as a guide. He installed and run CubicSDR successfully with those scripts on the Pi, as he demonstrated here:
[Ham Radio - SDRPlay running with CubicSDR on a raspberry Pi 3](https://www.youtube.com/watch?v=dx_iEpcZsDs). 
You only have to adapt this procedure for RTL-SDR:
This is similar to the SDRPlay setup, except you have to build
- `librtlsdr` : the RTL-SDR driver library,
- The SoapyRTLSDR module,
- The SoapySDR main library, but you already know how to do that.

If we assume you indeed have available packages for `librtlsdr` already, simply follow the [SoapyRTLSDR build instructions.](https://github.com/pothosware/SoapyRTLSDR/wiki#building-soapy-rtl-sdr)

Else, you have to additionally build `librtlsdr` yourself, from the GitHub repo using something like:
````
git clone https://github.com/librtlsdr/librtlsdr.git
cd librtlsdr
mkdir build
cd build
cmake ..
make
sudo make install
````

rrobinett, 2018-03-25:
Hi,

Thanks for the pointer to  @loughkb <https://github.com/loughkb> excellent
scripts.  While I get a clean compile, I get the same 'No OpenGL' error
upon starting Cubic.  I have enabled OpenGL in Raspian and rebooted the Pi,
and opening  https://get.webgl.org/ tells me that the Pi's chromium browser
supports WebGL and shows the spinning cube.

So what should I try next?

thanks,

rob

vsonnier, 2018-03-25:
Did you executed @loughkb scripts in full ? Their OpenGL settings look good, so that should be OK. Maybe you need "full hardware OpenGL" acceleration enabled like [this ?](http://www.raspberryconnect.com/gamessoftware/item/314-trying_out_opengl_on_raspberry_pi_3)

Beside this, did you check with `SoapySDRUtil` that the SoapyRTLSDR setup is right ?

rrobinett, 2018-03-26:
HI Vincent,

Yes, I ran the scripts in full in the order specified and piped all the
stdout and stderr build output to log files.  I see nothing in the logs to
indicate problems with OpenGL.
On my first attempt the Pi locked up, which I infer was due to running out
of memory when executing through a VNC session.  So I restarted the whole
scripts process directly from the console, and the second time the
compilation completed without problems.
Nevertheless, Cubic reports the same " Couldn't Create OpenGL context"
error.

I have followed the OpenGL setup instructions on the page you reference and
the glxgears program he mentions as a test of the OpenGL functionality does
run fine, although there are some command line errors reported which
suggest some incompatibilities with the Strech OS:

pi@pi1-berkeley:~ $ glxgears
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
299 frames in 5.0 seconds = 59.765 FPS
300 frames in 5.0 seconds = 59.792 FPS
301 frames in 5.0 seconds = 60.180 FPS
299 frames in 5.0 seconds = 59.608 FPS
^[c300 frames in 5.0 seconds = 59.892 FPS
301 frames in 5.0 seconds = 59.965 FPS
301 frames in 5.0 seconds = 60.123 FPS
^C
pi@pi1-berkeley:~ $

Soapy seems happy:

pi@pi1-berkeley:~ $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.0-g824d83e8
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libremoteSupport.so
(0.5.0-3f9666f)
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so
(0.2.5-b77a9fc)
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so
(0.1.0-b592d54)
Available factories... remote, rtlsdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
pi@pi1-berkeley:~ $

rrobinett, 2018-03-26:
​Upon further testing, it seems that Cubic is fully functional as I now see
all the items in the left hand config tree, although I still get the '"
Couldn't Create OpenGL context" error.  However that error seems to be only
associated with the 'select SDR device' dialog box presented upon startup.
If I dismiss the error dialog box, I can then open Cubic's File->SDR
Devices and select the RTL-SDR.  After than Cubic runs normally.

I am building Cubic on the Pi in order to implement a solar powered WSPR
receive station in a remote location on Maui.  While there will be internet
connectivity to the station, I would like to monitor multiple bands
simultaneously using multiple PI+RTL-SDR, and even at the lowest 250 K
sample rate, SoapyServer generates 10 Mbps per RTL-SDR.  While running
SopayServer on the Pi and Cubic on a PC is the best architecture for my
other 'flying antenna' project, for the remote WSPR system running Cubic on
the Pi is the best, most stable and least expensive solution.  On the Pi 3b
Cubic + WSJT-x consume < 70% of the CPU, and I hope to gain even more
headroom on the Pi 3b+ which have just arrived.

My final challenge is to get Cubic communicating with WSJT-x.  Since the Pi
has no built in audio input port, I am waiting for some USB audio dongles
to arrive.  It would be far more elegant if Cubic could output its audio to
a named pipe with WSJT-x listening to the pipe, but for now I'm just going
to connect a loopback cable between audio out and audio in on the dongle.

Thanks for creating Cubic,.  I is really perfectly architected for my
applications.

vsonnier, 2018-03-26:
> Thanks for creating Cubic,.  I is really perfectly architected for my
applications.

Thanks, but I'm merely a interested third party, you have to thank @cjcliffe for creating CubicSDR.

> While there will be internet
connectivity to the station, I would like to monitor multiple bands
simultaneously using multiple PI+RTL-SDR,

Be aware that CubicSDR only manage 1 device at a time, though, so you'll need as many Pi as you monitor 2.4Mz-large (max) bands.

> My final challenge is to get Cubic communicating with WSJT-x.  Since the Pi
has no built in audio input port,

You have to install something like [VAC](http://software.muzychenko.net/eng/vac.htm) to be able to pipe audio from each other "virtual audio devices" that will show up in Cubic audio devices. 
You don't have to *physically* plug different audio devices :)
 
Alternatively to VAC, you may use tricks like  @loughkb did on this video:
[Ham Radio - Linux - Decoding signals from your SDR using FLDigi and Pulse Audio](https://www.youtube.com/watch?v=cwYuispWn08)

rrobinett, 2018-03-30:
Hi Vincent,

I am back to trying to get Cubic to work on my new Pi 3b+ Stretch with my RTL-SDR.
While I had no problem building Soapy and Cubic and Cubic no detects and decodes a local FM broadcast station, the audio output is faint and somewhat garbled.   When run from a terminal window Cubic is continuously printing error messagees like:

SDRThread::readStream(): 2. SoapySDR read failed with code: -1
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
SDRThread::readStream(): 2. SoapySDR read failed with code: -1
.......
SDRThread::readStream(): 2. SoapySDR read failed with code: -1

Further, even when I File->Stop device the cpu stays 75% bush and vmstat reports lots of interrupts and context switches:

procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 4  0  50612  91172  27312 637664    0    0     0     0 3689 2257 79  1 20  0  0
 4  0  50612  90924  27320 637664    0    0     0     2 3559 1980 75  1 24  0  0

Do you have any suggestions on how to proceed?
Thanks

vsonnier, 2018-03-30:
Hello @rrobinett ,

> While I had no problem building Soapy and Cubic and Cubic __no__ detects and decodes a local FM broadcast station, the audio output is faint and somewhat garbled.

I suppose you maint __now__, given the rest of the phrase :)
The garbled audio and traces meant your Pi is not able to keep up, due to too much CPU load.
There are several things you can keep trying to reduce CPU load:
- Minimize the window, so that the display is not refreshed and therefore eat less CPU,
- Reduce to minimum bandwidth , something like 250 Khz, also try the "Manual Entry" on the Sample Rate menu,
- Settings => Show Hover Tips : un-checked
- Settings => Reduce CPU usage: checked

You can also try the following experiement:
- Run a SoapySDRServer on localhost, something like 127.0.0.1:61234,
- Run CubicSDR using the SoapyRemote client an connects to this localhost instance using the most compact format CS8 and setting appropriate buffer sizes.

In such configuration, the decoding runs on a process, and Cubic on another, buffered in a socket link.

Given @loughkb video, you should be able to obtain normal audio stream, even with slow UI animation... so I still think your OpenGL may be not hardware-accelerated. I such a case, it will eat a ton of CPU trying to run in software and totally kill the user experience.

> the audio output is faint

Remember that RTL-SDR Gain is "normal" contrary to SDRplay (Reducing) Gain, which means __bigger__ gain value really means __higher__ gain.

rrobinett, 2018-03-30:
Hi Vincent,

Thanks for your quick reply.

Yes, I meant now.

The bursty problem was present when using SoapyServer and OSX or PC clients
even though the Pi cpu was only 5% so I have found the bursty audio appears
at the 250 ksps rate.  When I change to 1.024 Ksps Cubic runs fine on the
Raspberry, OSX and Win10 when the RTL_SDR is local or remote..

On the Pi and using pulseaudio I am able to pipe the audio output of Cubic
to the audio input of WSJTx and thus have a single channel stand alone WSPR
receiver.  On the Pi 3++ cpu is 95% but appears to be fully functional
without any receive glitches.    I will run it long term to see if there
are any memory leaks or other stability problems.​

While not necessary for my application, it would be great if WSJT-x could
CAT control the receive frequency of Cubic.  WSJT-x offers me two CAT
control serial ports on the PI:  /dev/ttyAMA0 and USB.  I saw mention that
Cubic can be controlled by CAT but I see no Rig Control configuration
option on the Cubic Pi.  Do you know why?

Thanks so much for the help.  It has taken me months to get this functional
and I am a member of a group of WSPR hams who want to implement this
standalone receiver

vsonnier, 2018-03-30:
>  I saw mention that
Cubic can be controlled by CAT but I see no Rig Control configuration
option on the Cubic Pi.  Do you know why?

You need to (re) build CubicSDR with Hamlib support I think, like in #606. 
You should be able to do it this way
````
sudo apt-get install hamlib-utils libhamlib-dev
````
to install the Hamlib packages, then re-do the Cmake command for CubicSDR including `-DUSE_HAMLIB=1` at the end.

rrobinett, 2018-03-31:
OK, I now have Rig Control in Cubic, but it isn't clear how WSJT-x would communicate with Cubic.
The Rig Control -> Control Port dialog box says 'Select a port or enter your own' but there is nothing listed nor can I enter /dev/ttyAMA0, which is the only serial port offered by WSJT-x.  I see an open issue #444 which hasn't been answered.

rrobinett, 2018-04-19:
I was able to compile in Rig Control on the Pi, but I am still unable to get WSJT-x to communicate CAT commands to Cubic.  I am also trying to use Rig Control in the same way on the Mac without success.  the serial devices offered by WSJT-x and Cubic are remote bluetooth ports which don't work and also seem unlikely to be useful.  
In addition some of Cubic's Rig Control  selections are unclear to me.  I want Cubic only to be controlled by WSJT-x, so which should be selected?

vsonnier, 2019-03-31:
The remaining problems has nothing to do with the original issue, this is Spring time, so Cleanup this issue by letting it die.

closed 2019-03-31 by vsonnier