cjcliffe/CubicSDR #640: Problem installing CubicSDR on Mac OSX

maggiebasta, 2018-04-05:
Hi!

I am having the following issue installing CubicSDR. This comes after the running: DwxWidgets_CONFIG_EXECUTABLE=/Users/mabasta/Dev/wxWidgets-staticlib/bin/wx-config -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1

`64 bit compiler detected
CMake Error at /usr/local/Cellar/cmake/3.11.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.11.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.11.0/share/cmake/Modules/FindwxWidgets.cmake:953 (find_package_handle_standard_args)
  CMakeLists.txt:175 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/mabasta/Dev/SoapySDR/CubicSDR/build/CMakeFiles/CMakeOutput.log"`

Any pointers would be much appreciated!

maggiebasta, 2018-04-06:
Thanks for the quick response!

I realized after looking closer that the issue is actually coming from these two commands: 

wxWidgets-3.1.0$ ./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --with-mac --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --with-macosx-version-min=10.9 --prefix=/Users/mabasta/Dev/wxWidgets-staticlib CXXFLAGS="-std=c++0x" --with-libiconv=/usr

wxWidgets-3.1.0$ make -j4 && make install

After running the wxWidgets-3.1.0$ make -j4 && make install I get an error that 
'QTKit/QTKit.h' file not found. 

I looked at a few different threads and found related fixes. I downloaded wxWidgets-3.0.4 instead of 3.1.0 from the wxWidgets github, which initially seemed to fix the problem. I was able to run the prior DwxWidgets_CONFIG_EXECUTABLE command, but when I tried to run $cpack agterwards, I started having errors with that. Specifically that the preinstall command was not found.

righthalfplane, 2018-04-06:
I have cpack 3.10.2 installed - it worked for me.

[dir:~] dir% /opt/local/bin/cpack --version
cpack version 3.10.2

righthalfplane, 2018-04-16:
Try rebuilding CubicSDR from the start or at least delete everything in the build directory and do a new cmake -

git clone https://github.com/cjcliffe/CubicSDR.git
cd CubicSDR
mkdir build
cd build
cmake ../ -DwxWidgets_CONFIG_EXECUTABLE=/usr/local/bin/wx-config -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1
cpack

closed 2019-04-11 by cjcliffe