cjcliffe/CubicSDR #644: USRP B200 mini didn't work on mac

cn0xroot, 2018-04-20:
Hi ,i use b200 mini on mac,but cubicsdr didn't work
```
SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.0-release
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libHackRFSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libRedPitaya.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libbladeRFSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libmiriSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libosmosdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/librfspaceSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/librtlsdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libuhdSupport.so
Loading modules... Mac OS; Clang version 9.0.0 (clang-900.0.38); Boost_106501; UHD_003.010.002.000-3-g122bfae1

done
Available factories...bladerf, hackrf, lime, miri, null, osmosdr, redpitaya, remote, rfspace, rtlsdr, uhd,

 SoapySDRUtil --info |grep uhd
Module found: /usr/local/lib/SoapySDR/modules0.6/libuhdSupport.so
Available factories...bladerf, hackrf, lime, miri, null, osmosdr, redpitaya, remote, rfspace, rtlsdr, uhd,
```

righthalfplane, 2018-04-24:
does 

SoapySDRUtil --find

find your device ?

righthalfplane, 2018-04-27:
Have you gone into the package and started CubicSDR from the terminal - that gives a lot of information and tells you if it found your device - this is what I get -

[dir:~] dir% cd /Users/dir/Desktop/CubicSDR.app 
[dir:~/Desktop/CubicSDR.app] dir% ls
Contents
[dir:~/Desktop/CubicSDR.app] dir% cd Contents
[dir:~/Desktop/CubicSDR.app/Contents] dir% ls
Info.plist	MacOS		Resources
[dir:~/Desktop/CubicSDR.app/Contents] dir% cd macOS
[dir:CubicSDR.app/Contents/macOS] dir% ls
CubicSDR
libSoapySDR.0.7.0.dylib
libSoapySDR.0.7.dylib
libliquid.dylib
libwx_baseu-3.1.0.0.0.dylib
libwx_baseu-3.1.dylib
libwx_osx_cocoau_adv-3.1.0.0.0.dylib
libwx_osx_cocoau_adv-3.1.dylib
libwx_osx_cocoau_core-3.1.0.0.0.dylib
libwx_osx_cocoau_core-3.1.dylib
libwx_osx_cocoau_gl-3.1.0.0.0.dylib
libwx_osx_cocoau_gl-3.1.dylib
libwx_osx_cocoau_propgrid-3.1.0.0.0.dylib
libwx_osx_cocoau_propgrid-3.1.dylib
[dir:CubicSDR.app/Contents/macOS] dir% CubicSDR
Loading:: configuration file '/Users/dir/Library/Application Support/CubicSDR/config.xml'

Audio Device #0 Apple Inc.: Built-in Output
	Default Output? Yes
	Default Input? No
	Input channels: 0
	Output channels: 2
	Duplex channels: 0
	Native formats:
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		44100hz
		48000hz
		88200hz
		96000hz

Audio Device #1 Apple Inc.: HDMI
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 2
	Duplex channels: 0
	Native formats:
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		32000hz
		44100hz
		48000hz

Audio Device #2 ma++ ingalls for Cycling '74: Soundflower (2ch)
	Default Output? No
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

[... 69 lines cut ...]
[ERROR] SoapySDR::ConverterRegistry(CU8, CF32, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CS16, CU16, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CU16, CS16, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CS16, CS8, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CS8, CS16, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CS16, CU8, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CU8, CS16, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CU16, CS8, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CS8, CU16, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CS8, CU8, 0) duplicate registration
[ERROR] SoapySDR::ConverterRegistry(CU8, CS8, 0) duplicate registration
	Available factories...hackrf, null, rfspace, rtlsdr, sdrplay
Loaded font 'Bitstream Vera Sans Mono' from '/Users/dir/Desktop/CubicSDR.app/Contents/Resources/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/Users/dir/Desktop/CubicSDR.app/Contents/Resources/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/Users/dir/Desktop/CubicSDR.app/Contents/Resources/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/Users/dir/Desktop/CubicSDR.app/Contents/Resources/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/Users/dir/Desktop/CubicSDR.app/Contents/Resources/fonts/vera_sans_mono27_0.png', parsed 255 characters.
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1707039B20
Make device 0
  mir_sdr_api_version=2.110000
  mir_sdr_hw_version=2

Reporting enumeration complete.
SDR enumerator done.

cjcliffe, 2018-05-08:
@cn0xroot The 0.2.4 release at https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.4 might be worth a try; some of the dependent libs were updated on my build system including UHD -- don't have one here to test with unfortunately.

cn0xroot, 2018-05-08:
I try the 0.2.4 release (on windows and osx) with usrp ,it still doesn't work