cjcliffe/CubicSDR #645: Mismatch between the program and library build

bdn76, 2018-04-20:
$ ./CubicSDR
Fatal Error: Mismatch between the program and library build versions detected.
The library used 3.1.1 (wchar_t,compiler with C++ ABI 1002,wx containers,compatible with 3.0),
and your program used 3.1.1 (wchar_t,compiler with C++ ABI 1011,wx containers,compatible with 3.0).

$ dpkg -l | grep wx
ii  libgnuradio-wxgui3.7.12:amd64           3.7.12.0-2                              amd64        gnuradio wxgui functions
ii  libwxbase3.0-0v5:amd64                  3.0.4+dfsg-3                            amd64        wxBase library (runtime) - non-GUI support classes of wxWidgets toolkit
ii  libwxbase3.1-0-unofficial:amd64         3.1.1-1.stretch                         amd64        wxBase library (runtime) - non-GUI support classes of wxWidgets toolkit
ii  libwxbase3.1-0-unofficial3:amd64        3.1.1-1.stretch                         amd64        wxBase library (runtime) - non-GUI support classes of wxWidgets toolkit
ii  libwxbase3.1-dev                        3.1.1-1.stretch                         amd64        wxBase library (development) - non-GUI support classes of wxWidgets toolkit
ii  libwxbase3.1unofficial3-dev             3.1.1-1.stretch                         amd64        wxBase library (development) - non-GUI support classes of wxWidgets toolkit
ii  libwxgtk3.0-0v5:amd64                   3.0.4+dfsg-3                            amd64        wxWidgets Cross-platform C++ GUI toolkit (GTK+ runtime)
ii  libwxgtk3.0-gtk3-0v5:amd64              3.0.4+dfsg-3                            amd64        wxWidgets Cross-platform C++ GUI toolkit (GTK+ 3 runtime)
ii  libwxgtk3.1-0-unofficial:amd64          3.1.1-1.stretch                         amd64        wxWidgets Cross-platform C++ GUI toolkit (GTK+ runtime)
ii  libwxgtk3.1-0-unofficial3:amd64         3.1.1-1.stretch                         amd64        wxWidgets Cross-platform C++ GUI toolkit (GTK+ runtime)
ii  libwxgtk3.1-dev                         3.1.1-1.stretch                         amd64        wxWidgets Cross-platform C++ GUI toolkit (GTK+ development)
ii  libwxgtk3.1unofficial3-dev              3.1.1-1.stretch                         amd64        wxWidgets Cross-platform C++ GUI toolkit (GTK+ development)
ii  python-wxgtk3.0                         3.0.2.0+dfsg-7                          amd64        Python interface to the wxWidgets Cross-platform C++ GUI toolkit
ii  python-wxversion                        3.0.2.0+dfsg-7                          all          API for selecting the wxPython version to use
ii  wx-common                               3.0.4+dfsg-3                            amd64        wxWidgets Cross-platform C++ GUI toolkit (common support files)
ii  wx3.1-headers                           3.1.1-1.stretch                         amd64        wxWidgets Cross-platform C++ GUI toolkit (header files)
ii  wx3.1-i18n                              3.1.1-1.stretch                         all          wxWidgets Cross-platform C++ GUI toolkit (i18n support)

$ cat /etc/apt/sources.list.d/wx.list 
deb http://repos.codelite.org/wx3.1.1/debian/ stretch libs

$ cat /etc/debian_version 
buster/sid

vsonnier, 2018-04-20:
You are mixing v3.0.4 prebuilt libs and the v3.11 version, which are incompatible. You need to rebuild wxWidgets v3.11 in full. This is a duplicate of #635.

bdn76, 2018-04-21:
Download wx and make in to specific dir directly.

closed 2018-04-21 by bdn76