cjcliffe/CubicSDR #650: CubicSDR not show fine through remote connection

Tnix76, 2018-05-07:
I'm using CubicSDR 0.2.3 x64 in appimage format with SoapySDR and SDRPlay. In a normal X session it works fine showing all the buttons and the waterfall. But I usually connect to with an X2go client that's a X11 with a SSH session tunnel. I've tried with several desktop environments (XFCE, Plasma, ...) but the problem is CubicSDR not show well (it hasn't several buttons and there's no waterfall). When I run it in a command line it shows the repeated error:

(CubicSDR-0.2.3-x86_64.AppImage:27635): GLib-CRITICAL **: 12:00:05.182: g_error_free: assertion 'error != NULL' failed

Is there a way to use CubicSDR in an x2go remote client??

Thanks.

cjcliffe, 2018-05-07:
@Tnix76 If x2go can fully support OpenGL applications then it should work -- my guess is that it may not be fully supported and is unable to show the OpenGL portions of the interface.

Try some other applications that use OpenGL such as glxgears or some of the free 3D games like tuxracer or something to see if they're also missing graphics.  If x2go can't fully support OpenGL it's unlikely that CubicSDR will be able to work through it.

Tnix76, 2018-05-09:
Ok, thanks for your help. I'll try to find any way to show through x2go OpenGL applications.

Regards.

closed 2018-05-15 by cjcliffe