cjcliffe/CubicSDR #652: Signal strength readout request

mpvano, 2018-05-21:
I'm sorry if I appear a little dense, but I don't really understand how one measures the level of a signal in CubicSDR and never have seen any description in the documentation.

It's not clear to me what the calibration lines actually measure (largest peak, minimum noise level, ?), and they never stop moving and jumping around between different scaling factors and numbers of lines - even if AGC is off. I understand that using the mouse, one can change the scaling, but it instantly starts going back to doing whatever it randomly appears to be doing once you have done that!

It would be nice to have a single place to read a value out for the currently signal so one could use CubicSDR for measurements, even if the value was only relative.

thanks,

M

vsonnier, 2018-05-21:
Hello @mpvano,
Please have a look at #621 for some explanation on how the dB display works, and #531 for some enhancement suggestions.

Long story short: the dB display is dynamic to offer a maximum displayed range at all times, which prevent the user to be compelled to adapt the display with zooming controls like the others SDR software do. This is intended, but could be bothering in some contexts, so workarounds exist to stabilize the range display as described in #621.

mpvano, 2018-05-21:
I've been sort of doing that, but it still seems a little awkward - for example: when trying to measure the sensitivity of two different preamps using a step attenuator.

Visual gain does NOT freeze anything here - it only briefly resets it - usually not for long enough to take a reading after clicking, it immediately heads for a completely different scale. Especially since the autoscaling sometimes gives a display with only one fiduciary line!

On any real signals, the scale lines are constantly jumping and moving around, and because there's no calibration readout on them, it's purely guesswork to tell what they mean.

anyway, you said this was also a wish list, so I'd still wish to see a simple way to measure the strength of the signal I'm currently listening to - - like all my other SDR receiver software programs have.

M

closed 2018-05-21 by mpvano