cjcliffe/CubicSDR #654: Command line support?

dmitryelj, 2018-05-24:
Hi. 

I want to use CubicSDR and SDRPlay on Raspberry Pi for autonomous WSPR receiver. 
Is it possible to run CubicSDR from command line without UI, just specifying receiver/frequency/modulation/etc? 

Something like ./CubicSDR --device=sdrplay --modulation=usb --frequency=7037000 ...  ?

Thanks.

vsonnier, 2018-05-24:
Hello @dmitryelj , this is a duplicate of #577 and #316,  also related to #485, #438, and  . As of today, there is indeed no way to start GUI-less, nor ways to start a device automatically by the command line.

closed 2018-05-24 by vsonnier

cjcliffe, 2019-01-04:
@dmitryelj if you haven't found it already there's a project at https://github.com/rxseger/rx_tools that works with SoapySDR/SDRPlay and provides command line tools that basically do exactly what you're looking for.

aolszowka, 2021-01-01:
@cjcliffe / @vsonnier 

First thank you for CubicSDR.

I realize that this is an old thread (and a duplicate), however I am attempting to do the exact same thing. I have tried many times over to use `rx_fm` but I simply cannot get it to work (I constantly get static no matter the arguments I use, CubicSDR on the other hand has no issue).

Furthermore my use case is I would like to record as many distinct streams as possible at once (There are usually three radio stations within the tuning band per SDR allowing for more capture) and as far as I can tell `rx_fm` would lock the tuner to just record a single frequency per tuner. Compare this to CubicSDR which allows you to record as many distinct frequencies within the tunable band "Window", something like 3Mhz? (this feature is absolutely amazing by the way)

Ideally I would like to invoke CubicSDR via CLI, but to go even further I would want to pipe the output of the recording (via pipe `|`) into either `flac` or `ffmpeg` to compress them down to flac/mp3 as appropriate for post production work.

Alternatively if there was some way to see the settings CubicSDR is using to pass to `rx_fm` I would settle for that as well (even if I could only tune a single station at a time, I've got multiple NESDR SMArts); here's what I was trying without luck (simply static):

`C:\Program Files\PothosSDR\bin>rx_fm.exe -f 105.9M -M wbfm - | C:\DevApps\System\ffmpeg\bin\ffmpeg.exe -f s16le -ac 1 -ar 32k -i pipe:0 -acodec libmp3lame -ab 32k -f mp3 -y -t 00:00:10 "R:\Sample.mp3"`

Others online have had similar issues, and messing with the gain (`-g37.2`) or upping the sampling rate (`-s25000`) as was suggested somewhere seems to give slightly less static (to the point where you can maybe make out a word or two here or there), but again nothing seems to work as well as CubicSDR's FM tuning.

What would be really killer is if there was some API that I could just invoke from C#/PowerShell (I am comfortable writing the p/invoke wrappers if pointed in the right direction).

rrobinett, 2021-01-02:
I want to reinforce @aolszowka 's appreciation for CubicSDR.
I too want to use CubicSDR for testing my receive system and finding signal, and then run rx_fm to record those signals to wav files for further processing by a system service.
Since CubicSDR is already communicating with my RTL-SDR, I don't think it would be best if I could build rx_tools using the SoapySDR libraries already installed by CublicSDR.
Is that possible?