cjcliffe/CubicSDR #655: CubicSDR-0.2.3-x86_64.AppImag from SDRPlay

hoppelpoppel, 2018-05-25:
Hello,
When I try to launch the CubicSDR-0.2.3 image provided by SDRPlay, I get that message:
/lib/x86_64-linux-gnu/libthai.so.0: version `LIBTHAI_0.1.25' not found (required by /usr/lib64/libpango-1.0.so.0)
but, when I do a
ldd /usr/lib64/libpango-1.0.so.0
linux-vdso.so.1 (0x00007ffc10153000)
libgobject-2.0.so.0 => /usr/lib64/libgobject-2.0.so.0 (0x00007fbb70ddc000)
libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007fbb70ac4000)
libm.so.6 => /lib64/libm.so.6 (0x00007fbb70731000)
libthai.so.0 => /usr/local/lib64/libthai.so.0 (0x00007fbb70527000)
libfribidi.so.0 => /usr/lib64/libfribidi.so.0 (0x00007fbb7030b000)
libc.so.6 => /lib64/libc.so.6 (0x00007fbb6ff4d000)
libffi.so.7 => /usr/lib64/libffi.so.7 (0x00007fbb6fd43000)
libpcre.so.1 => /usr/lib64/libpcre.so.1 (0x00007fbb6fab6000)
libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fbb6f897000)
/lib64/ld-linux-x86-64.so.2 (0x00007fbb71277000)
libdatrie.so.1 => /usr/lib64/libdatrie.so.1 (0x00007fbb6f68f000)
as you can see, libpango find libthai.
Any suggestions ?
Thanks
DA

vsonnier, 2018-05-25:
@hoppelpoppel Hello ! Cubic's AppImage from SDRPlay is outdated, please use the one from the [offical release page](https://github.com/cjcliffe/CubicSDR/releases), thanks.

vsonnier, 2018-06-02:
@hoppelpoppel Ping ! Any news on your issue ?

hoppelpoppel, 2018-06-02:
Yes indeed !
I compiled both libthai first, then libpango. The problem was in libpango
from my distribution. After compilation, everything works as it should.with
version 0.2.4 of CubicSDR that I also compiled.
Thanks
DA

vsonnier, 2018-06-02:
Great ! Indeed the problem is AppImage is linked with the dependencies of the Linux box it was built on, and may work or not at all on the Linux flavors out there.
The best chance is to rebuild CubicSDR but not everybody is able to to this. Glad it worked for you.

closed 2018-06-02 by vsonnier