cjcliffe/CubicSDR #658: crash on start up

AndrewLag, 2018-06-02:
crash on start up  it worked thr first time https://www.youtube.com/watch?v=dsM00K1Pz8E&feature=youtu.be

vsonnier, 2018-06-02:
Hello @AndrewLag. I just tried to re-install my RTL-SDR.com v3 dongle, together with Zadig v2.3 on Windows 10 1803 x64, as described everywhere. 

At CubicSDR startup, I have 2 RTL-SDR devices listed:
- `Generic RTL2832U OEM ::`
-  `Generic RTL2832U OEM :: 00000001`

Only the second one is a valid device and operates properly.
 
Make sure you run the latest CubicSDR v0.24 from [here](https://github.com/cjcliffe/CubicSDR/releases), and try to run it in command line:
- Press 'Windows' start button,
- Type `cmd.exe` and run it (Command line prompt),
-  Navigate to the CubicSDR installation directory, by typing `cd c:\Program Files\CubicSDR`,
- Type the command `CubicSDR.exe > start.log`
- Start as the device as you did before,
- Copy and paste the `start.log` file contents here.

Good  luck :)

AndrewLag, 2018-06-04:
Loading:: configuration file 'C:\Users\Andrew\AppData\Roaming\CubicSDR\config.xml'
Loaded 220 rig models via hamlib.

Audio Device #0 Default Device
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		8-bit signed integer.
		16-bit signed integer.
	Supported sample rates:
		4000hz
		5512hz
		8000hz
		9600hz
		11025hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #1 LG TV (NVIDIA High Definition Audio)
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 2
	Duplex channels: 0
	Native formats:
		8-bit signed integer.
		16-bit signed integer.
	Supported sample rates:
		4000hz
		5512hz
		8000hz
		9600hz
		11025hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #2 Stereo Mix (Realtek High Definition Audio)
	Default Output? No
	Default Input? No
	Input channels: 2
	Output channels: 0
	Duplex channels: 0
	Native formats:
		8-bit signed integer.
		16-bit signed integer.
	Supported sample rates:
		11025hz
		22050hz
		44100hz
		96000hz

SDR enumerator starting.
SoapySDR init..
	API Version: v0.7.0
	ABI Version: v0.7
	Install root: c:\Program Files
	Loading modules... 
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//airspySupport.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//audioSupport.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//bladeRFSupport.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//HackRFSupport.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//LMS7Support.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//RedPitaya.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//remoteSupport.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//rtlsdrSupport.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//sdrPlaySupport.dll..
Loading bundled SoapySDR module c:\Program Files\CubicSDR/modules//uhdSupport.dll..
Initializing WGL swap control extensions..
Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.003-0-unknown

	Available factories...airspy, audio, bladerf, hackrf, lime, redpitaya, remote, rtlsdr, uhd
Loaded font 'Bitstream Vera Sans Mono' from 'c:\Program Files\CubicSDR\fonts\vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'c:\Program Files\CubicSDR\fonts\vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'c:\Program Files\CubicSDR\fonts\vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'c:\Program Files\CubicSDR\fonts\vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'c:\Program Files\CubicSDR\fonts\vera_sans_mono27_0.png', parsed 255 characters.
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = Stereo Mix (Realtek High Definition Audio)
Make device 1
  device_id=2
  origin=https://github.com/pothosware/SoapyAudio

  available = No
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 
  manufacturer = 
  product = 
  rtl = 0
  serial = 
  tuner = 
Make device 2
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 1
  serial = 00000001
  tuner = Rafael Micro R820T
Make device 3
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=1

  driver = redpitaya
Make device 4

  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device
Make device 5
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = Stereo Mix (Realtek High Definition Audio)

vsonnier, 2018-06-06:
In your log, the important part is:
````
available = No
driver = rtlsdr
label = Generic RTL2832U OEM ::
manufacturer =
product =
rtl = 0
serial =
tuner =
Make device 2
origin=https://github.com/pothosware/SoapyRTLSDR
rtl=0

available = Yes
driver = rtlsdr
label = Generic RTL2832U OEM :: 00000001
manufacturer = Realtek
product = RTL2838UHIDIR
rtl = 1
serial = 00000001
tuner = Rafael Micro R820T
Make device 3
````

which looks like mine:
````
  available = No
  driver = rtlsdr
  label = Generic RTL2832U OEM ::
  manufacturer =
  product =
  rtl = 0
  serial =
  tuner =
Make device 6
usb_open error -12
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 1
  serial = 00000001
  tuner = Rafael Micro R820T
Make device 7
Found Rafael Micro R820T tuner
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=1
````

Except you don't go to the point of selecting the device because it crashes. Well...
You can try several things to see if your RTL-SDR is faulty of not:
- Does it run with SDRSharp ?
- Does it run with the CubicSDR part of  the [PothosSDR distribution](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D)?
- Does it run with [CubicSDR_win64_20180606_2113_VS2017.zip](https://github.com/cjcliffe/CubicSDR/files/2077834/CubicSDR_win64_20180606_2113_VS2017.zip) ?, A fresh executable from the master branch, compiled with VS2017. Replace the `CubicSDR.exe` in the official CubicSDR v0.24 installation by this one.

AndrewLag, 2018-06-07:
hey i tried all of those things an they didn't work.... but what I did is I uninstalled the CubicSDR an deleted the app data now it works like 10/10.

vsonnier, 2018-06-07:
Great ! Have fun with CubicSDR !

closed 2018-06-07 by vsonnier

LightlySalted, 2021-03-23:
Sorry for being a grave digger...

I had the same issue, 
instead of uninstalling and re-installing I just deleted my C:\Users\$username\AppData\Roaming\CubicSDR\ folder and copied back in my bookmarks.xml file.  that seemed to do the trick for now.