cjcliffe/CubicSDR #659: Manual edit of bookmarks XML

sstandeford, 2018-06-06:
Hello and thank you for this great SDR application. I am using MacOS v2.4 and I have tried to "generate" my own bookmarks XML file from a frequency list. The XML looks correct, I basically added a new group "DFW AppDep" and added multiple modems to the original XML then renamed it. When I try to load the new bookmarks XML file the application crashes? Not sure if an element or its content is causing this? I have a feeling it is the "user_label", does this element data need to be encoded? Below is the generated XML that crashes on load:

```
<?xml version="1.0" ?>
<cubicsdr_bookmarks>
    <header>
        <version>%30%2e%32%2e%34</version>
    </header>
    <branches>
        <active>0</active>
        <range>0</range>
        <bookmark>1</bookmark>
        <recent>0</recent>
    </branches>
    <ranges>
        <range>
            <label>%31%36%30%20%4d%65%74%65%72%73</label>
            <freq>1900000</freq>
            <start>1800000</start>
            <end>2000000</end>
        </range>
        <range>
            <label>%38%30%20%4d%65%74%65%72%73</label>
            <freq>3750000</freq>
            <start>3500000</start>
            <end>4000000</end>
        </range>
        <range>
            <label>%36%30%20%4d%65%74%65%72%73</label>
            <freq>5368500</freq>
            <start>5332000</start>
            <end>5405000</end>
        </range>
        <range>
            <label>%34%30%20%4d%65%74%65%72%73</label>
            <freq>7150000</freq>
            <start>7000000</start>
            <end>7300000</end>
        </range>
        <range>
            <label>%33%30%20%4d%65%74%65%72%73</label>
            <freq>10125000</freq>
            <start>10100000</start>
            <end>10150000</end>
        </range>
        <range>
            <label>%32%30%20%4d%65%74%65%72%73</label>
            <freq>14175000</freq>
            <start>14000000</start>
            <end>14350000</end>
        </range>
        <range>
            <label>%31%37%20%4d%65%74%65%72%73</label>
            <freq>18068180</freq>
            <start>17044180</start>
            <end>19092180</end>
        </range>
        <range>
            <label>%31%35%20%4d%65%74%65%72%73</label>
            <freq>21225000</freq>
            <start>21000000</start>
            <end>21450000</end>
        </range>
        <range>
            <label>%31%32%20%4d%65%74%65%72%73</label>
            <freq>24940000</freq>
            <start>24890000</start>
            <end>24990000</end>
        </range>
        <range>
[... 313 lines cut ...]
                <type>FM</type>
                <user_label />
                <squelch_level>-100</squelch_level>
                <squelch_enabled>1</squelch_enabled>
                <output_device>Apple Inc.: Built-in Output</output_device>
                <gain>1.25000</gain>
                <muted>0</muted>
            </modem>
        </group>
    </modems>
    <recent_modems>
        <modem>
            <bandwidth>8330</bandwidth>
            <frequency>135000000</frequency>
            <type>AM</type>
            <user_label></user_label>
            <squelch_level>-18.0000</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>Apple Inc.: Built-in Output</output_device>
            <gain>0.0950000</gain>
            <muted>0</muted>
        </modem>
    </recent_modems>
</cubicsdr_bookmarks>
```

vsonnier, 2018-06-06:
Hi @sstandeford !
`user_label` and `label` are encoded, in a specific format destined to support `std::wstring` with > ASCII characters for labels.
This is the origin of the crash.
This is not practical for manual editing, so I commited in  https://github.com/cjcliffe/CubicSDR/commit/3e4aadfaef05cba3b9232479913bdd82377c8b4d
 a change so that you can edit the bookmarks / sessions labels in normal 'text' ASCII and it will be loaded as normal.
Note that the labels will be in the encoded form again once they are saved.
With that change, your example load successfully.

closed 2018-06-11 by vsonnier

sstandeford, 2018-06-13:
Great thank you!

I can modify my import script to encode the user_label and label. It appears to be

edarnoldnet, 2022-06-13:
Noticed that I also get the crash with same circumstance,  but the crash doesn't happen (though labels still don't import) if you eliminate the <user_label  /> (in your code it's found under your NOAA group name item).

commit 3e4aadfa (2018-06-06): Fix #659 : Allow manual editing of user_labels in normal ASCII