cjcliffe/CubicSDR #661: Unable to change frequency using waterfall on Mac OS 10.13

DWetherilt, 2018-06-18:
I'm using CubicSDR 0.2.4 on Mac OS 10.13.5 with an SDR Play RSP 1.

Double clicking on the waterfall when the app starts creates a demodulator and the receiver starts its audio.  However I am unable to drag the demodulator across the waterfall or click on a specific spot on the waterfall to change frequency.  It seems like my clicks only take me to the nearest whole Mhz - for example, if I click around 14.2Mhz, it will take me to 14.0Mhz and if I click on 14.7Mhz, it will take me to 15.0Mhz.  

I am only able to accurately set frequency by using the up/down controls on the frequency display which is a bit of a fiddle.

Is this a software bug or user error?? ;-)

Magalex2x14, 2018-06-18:
Hello! I also use CubicSDR 0.2.4 on Mac OS 10.13.5. I do not have such problems, and I can not reproduce this behavior ... Single click is enough to create a demodulator (shift+click to add another one), and I'm completely free in its placement on the waterfall. Do you use prebuild version from [releases](https://github.com/cjcliffe/CubicSDR/releases) or own build with precompiled dependencies?

vsonnier, 2018-06-18:
> Double clicking on the waterfall when the app starts creates a demodulator...
 
First thing, single click is enough here...

> However I am unable to drag the demodulator across the waterfall or click on a specific spot on the waterfall to change frequency. It seems like my clicks only take me to the nearest whole Mhz...

Once created you can move the demod by putting the mouse over it and pressing left button (cursor changes to a cross) and then you can __drag__ it left of right to change its frequency, meaning you hold left mouse button and moves left or right (well, like all the usual drag-n-drop moves really)

DWetherilt, 2018-06-18:
Yes, as a software engineer, I'm pretty familiar with drag-n-drop etc ;-)  

I'm using the pre-built release 0.2.4 as found from this link: https://github.com/cjcliffe/CubicSDR/releases/download/0.2.4/CubicSDR-0.2.4-Darwin.dmg

OK, I have tested a bit more and found that if I zoom out so I can see a whole Mhz of bandwidth, then I can drag the demodulator to the next whole 1Mhz step, I don't seem to be able to drag it to anything between 1Mhz boundaries.  This seems like a setting somewhere, but nothing it leaping out at me.

vsonnier, 2018-06-18:
OK I think I understand finally. 
Drag-n-drop is limited to the current visible bandwidth and it is expected. By zooming out, you'll be able to show more bandwith, limited to your sample rate.
Now what you want to do is moving the central frequency (arrows, left - right) and your demodulator to follow when doing that. 
There is no 'standard' shortcut to move simultaneously the demod AND the central frequency.

Except if you Delta-Lock (V, or through the main screen) your demod, which means its frequency will be at a fixed distance ('delta') from the central frequency and so will keep moving with the central frequency changes.

vsonnier, 2018-06-24:
@DWetherilt Ping ! Did it answer your questions ?

closed 2018-06-28 by vsonnier

cjcliffe, 2018-08-03:
@DWetherilt it sounds like you may have the frequency snap enabled; check if one of the frequency digits is bordered in red; just right-click the digit or right-click the far-right digit to clear the snap.

DWetherilt, 2018-09-03:
Yes - that was it.  Definitely something stupid on my part then.  Thank you so much