cjcliffe/CubicSDR #662: CubicSDR not starting after upgrade Arch Linux

Terminalator, 2018-06-27:
After updating I am now getting this error:
[code][john@jupiter64 ~]$ cubicsdr 
Fatal Error: Mismatch between the program and library build versions detected.
The library used 3.1.1 (wchar_t,compiler with C++ ABI 1011,wx containers,compatible with 3.0),
and your program used 3.1.1 (wchar_t,compiler with C++ ABI 1012,wx containers,compatible with 3.0).
Aborted (core dumped)[/code]

Any help? Thanks

Terminalator, 2018-07-03:
Any idea how I can downgrade CubicSDR to a previous version?

Terminalator, 2018-07-03:
After trying several times I got this to build OK.

vsonnier, 2018-07-04:
Great ! Let's continue discussion on #663 then :)

closed 2018-07-04 by vsonnier

cyberdavev, 2018-07-05:
I have exactly the same problem. How did you manage to fix this?

Terminalator, 2018-07-05:
I don't honestly know @cyberdavev . I just rebuilt everything (using yaourt) including:

aur/wxgtk-common-dev 3.1.1-1 [installed] (3) (0.31)
aur/wxgtk2-dev 3.1.1-1 [installed] (3) (0.31)
aur/wxgtk2.8 2.8.12.1-6 [installed] (Out of Date) (27) (0.46)
aur/wxgtk3-dev 3.1.1-1 [installed] (3) (0.31)