cjcliffe/CubicSDR #663: CubicSDR freezes whole computer

Terminalator, 2018-07-03:
Using SDRPlay RSP2

When I try to change the Sample Rate from 2.048MHz to anything else, my whole computer freezes. I have to press the reset button to recover it.

Using v 0.2.4 built from latest git packages from Arch Linux AUR. Is there any info I can capture to see why this is happening?

```
[john@jupiter64 ~]$ cubicsdr 
Loaded 229 rig models via hamlib.

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: Yes
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: Yes
Using glxSwapIntervalEXT.

Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono27_0.png', parsed 255 characters.
SDR enumerator starting.
SoapySDR init..
	API Version: v0.7.0
	ABI Version: v0.7
	Install root: /usr
	Loading modules... 
	Available factories...null, sdrplay
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1707043D20
Make device 0
  mir_sdr_api_version=2.110000
  mir_sdr_hw_version=2

Reporting enumeration complete.
SDR enumerator done.
SDR thread starting.
device init()
[INFO] Using format CF32.
Device Stream MTU: 65536
starting readLoop()
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono32_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono36_0.png', parsed 255 characters.
```

SDRplay, 2018-07-03:
If you have built from the latest SoapySDRPlay github source, then you will need 2.13 version of the API from our website (https://www.sdrplay.com/downloads)

vsonnier, 2018-07-04:
Ah, it looks like a duplicate of #586 too. Try to re-do the steps described there and following @SDRplay advice.

Terminalator, 2018-07-04:
Thank you all for your feedback. I'll give your suggestions a go!

Terminalator, 2018-07-04:
@vsonnier I don't think it's a duplicate (as that referenced bug was mine also). The first one just didn't start up the application, this one freezes the computer solid. I have to press the reset button to recover it.

vsonnier, 2018-07-04:
It is the same story again: CubicSDR don't work on your Arch. Same cure, recompile all CubicSDR and SoapySDRPlay from scratch from the master(s), install the  v2.13 ... and so on. The fact that it freeze is irrelevant here.

Terminalator, 2018-07-04:
Yesterday I reinstalled all the wxgtk files:

```
[john@jupiter64 ~]$ yaourt -Ss  wxgtk|grep installed
aur/wxgtk-common-dev 3.1.1-1 [installed] (3) (0.32)
aur/wxgtk2-dev 3.1.1-1 [installed] (3) (0.32)
aur/wxgtk2.8 2.8.12.1-6 [installed] (Out of Date) (27) (0.47)
aur/wxgtk3-dev 3.1.1-1 [installed] (3) (0.32)
```

Today I installed the 2.13 version of the API

```
[john@jupiter64 ~]$ yaourt -Q libsdrplay
local/libsdrplay 2.13.1-1
[john@jupiter64 ~]$ yaourt -Ql libsdrplay
libsdrplay /etc/
libsdrplay /etc/udev/
libsdrplay /etc/udev/rules.d/
libsdrplay /etc/udev/rules.d/66-mirics.rules
libsdrplay /usr/
libsdrplay /usr/include/
libsdrplay /usr/include/mirsdrapi-rsp.h
libsdrplay /usr/lib/
libsdrplay /usr/lib/libmirsdrapi-rsp.so
libsdrplay /usr/lib/libmirsdrapi-rsp.so.2
libsdrplay /usr/lib/libmirsdrapi-rsp.so.2.13
libsdrplay /usr/share/
libsdrplay /usr/share/licenses/
libsdrplay /usr/share/licenses/libsdrplay/
libsdrplay /usr/share/licenses/libsdrplay/LICENSE
```

I reinstalled liquid-dsp

```
yaourt -Ss liquid-dsp|grep installed
aur/liquid-dsp-git v1.3.1.r134.gdd3facf8-1 [installed: v1.3.1.r136.g9658d811-1] (10) (0.66)
```

I reinstalled all the soapy files:

```
[john@jupiter64 ~]$ yaourt -Ss soapy|grep installed
aur/simplesoapy 1.5.1-1 [installed] (4) (0.22)
aur/soapy_power 1.6.1-1 [installed] (5) (0.22)
aur/soapyremote-git 0.2.0.r0.g71f7183-1 [installed: 0.4.3.r43.gf2e51b3-1] (5) (0.02)
aur/soapysdr-git 3:0.5.4.r32.ga412ad4-1 [installed: 3:0.6.1.r76.g937131b-1] (18) (0.48)
aur/soapysdrplay-git r66.3890517-1 [installed: r111.12c3db6-1] (4) (0.00)
```

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.0-g937131be
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr
Search path:  /usr/lib/SoapySDR/modules0.7
Search path:  /usr/local/lib/SoapySDR/modules0.7                (missing)
Module found: /usr/lib/SoapySDR/modules0.7/libremoteSupport.so  (0.5.0-f2e51b3)
Module found: /usr/lib/SoapySDR/modules0.7/libsdrPlaySupport.so (0.1.0-12c3db6)
Available factories... remote, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

To test, plug in SDRplay and run 'SoapySDRUtil --probe sdrplay'
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
error: missing 'poppler-qt' dependency for 'kfilemetadata4'
[john@jupiter64 ~]$ SoapySDRUtil --probe sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[ERROR] avahi_service_browser_new() failed: Bad state

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=1707043D20
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=2

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```

I still get the total freeze when increasing Sample Rate above 2.048MHz!

SDRplay, 2018-07-04:
Can you list the settings you are using for the RSP control please?

vsonnier, 2018-07-05:
@Terminalator Well I was talking about _recompiling_ all, not merely reinstalling from a package manager...
I was asking not for the pleasure of bothering you, but to be be sure to have a common starting point.

Anyway, let's assume it is good enough to have the right CubicSDR. After all, the software itself can't possibly freeze your PC so it's something deeper.

We can also rule out an OpenGL / Graphic card driver, because your problem seems very correlated to to the change of sample rate above `2.078 MHz`, which doesn't change the OpenGL drawing in significant ways.

That leaves a USB problem. 
@SDRplay, does the driver has additional dependencies, like `libusb`, that @Terminalator  could upgrade ?

The problem can also lies in the kernel drivers, where > 2.048Mhz rates and the unusual transmission mode SDRPlay uses (isochornous, with USB band pass reservation, if I remember correctly) can be buggy.

Indeed the 2.048Mhz problem has  already occurred in the past like in #529. There is also our recent issue https://github.com/pothosware/SoapyRemote/issues/26 affecting ARM devices about the erroneous implementation the isochronous mode. Could this issue show up on PCs as well ?

@Terminalator  Have you tried other USB ports ? Ex, changing from/to USB 2.0 from/to USB 3.0 port ? Is your device directly connected to a PC usb port or through a switch ? ...etc.

Terminalator, 2018-07-05:
Surely this isn't right?

```
[john@jupiter64 pkg]$ SoapySDRUtil --probe=sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device sdrplay
[ERROR] avahi_service_browser_new() failed: Bad state
libusb: error [udev_hotplug_event] ignoring udev action bind

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=1707043D20
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=2
```

Terminalator, 2018-07-05:
I also found that changing the IF mode to 2048kHz causes everything to lock up also :(

If this issue is not related to CubicSDR, should I take this discussion to another forum? Or is it it still relevant here?

SDRplay, 2018-07-05:
Couple of things... firstly the avahi error is related to SoapyRemote I believe (@vsonnier may be able to comment)

Secondly, there is an issue trying to use versions of Libusb greater than 1.0.19 or 1.0.20 (I will need to check the specific version) with higher sample rates. The x86_64 API should have the right version of libusb, but if you are using the i686 version, this might be the issue. It's worth checking the API dependency and see if that's the case.

I will check out the > 3 MHz sample rate and 2.048 MHz IF mode tomorrow on Ubuntu - I don't have Arch Linux to hand I'm afraid.

Terminalator, 2019-01-04:
Sorry for the long delay in updating this. 

I have just figured out how to go about downgrading my libusb on Arch Linux. I'm currently using 1.0.19 and this has fixed the freezing issue when changing the bandwidth. 

My question is (probably should be directed at @SDRplay ) - what can I do about this? You seem to know about this issue already looking at your comment above. Thanks for all your help BTW.

closed 2020-07-13 by cjcliffe